terraref/reference-data #2: Determine meta-data format for raw data from Lemnatec

dlebauer, 2015-08-05:
[ ] Review and comment on and propose changes to filesystem and meta-data

## Metadata 

The current proposal is to use key-value pair json files for metadata, each data file and will be in a separate directory, alongside an `info.json` meta-data file. The meta-data will be comprehensive and redundant in order to be portable.

We will later extract meta-data from the metadata files into a searchable database (BETYdb).

Meta-data should contain anything that is not 'default' about the sensor, e.g. anything that can be set by the user (including position, optics, etc.). @robkooper suggested using url to point to some resources (Rob please clarify).

There has been some discussion of using json vs. xml. Many developers preferr json, and notably, geojson is a postgres data type. 

The scope of information that would be provided by an RGB camera is in [this gist](https://gist.github.com/dlebauer/75429d046badb100483b) that contains the original proposal (`info.xml`, and other structures (`info2.xml`, `info2.json`)

## Directory Structure

The directory structure will be nested by date then sensor, and divided into 'location independent' data like met:

* `/Date/Met/met.csv`
* `/Date/Time/Sensors/data.raw`

We need to see if this is feasible (Markus?) as the sensors may be moving and not taking images at the same rate (perhaps the laser scanner will be continuous). But we could at least combine by date and then some common interval imposed by the equipment (e.g. at end of each pass across the field.

The original proposal is below. It divides sensors into location independent (like meteorology) and location dependent (sensors) and then by `date/time/sensor`

### Markus' original proposal:

separating data between 'location independent' and location dependent'

_The rest of this is the description of this alternative approach...

This is based on a file system organization of all data. 

A folder structure has the advantage, that it can be accessed easily and does not need much explanation. The data within the folder structure is organized in a way, that each sensor system can work and store its data on its own. That way the different sensor system stay independent from each other. The folder structure may be provided through an ftp server, a simple and well performing protocol. 

The two topmost folders in the proposed folder structure are separating the two kinds of sensors used in the system "Location Independent" data such as a single meteorology sensor and "Location Dependent", including data from moving sensors.

#### Location Dependent (sensors)

def: Sensors like cameras that fetches data only during an imaging job of the gantry system.

In the folder "locationDependent" you find a sub folder for each camera/sensor unit. The next sub folder structure divides the measurements by date (see folder "VisCamera"). For each measurement within a day an new folder is created that carries the timestamp. Within that folder you find a info.xml file containing all meta data and a data.XXX file that contains the raw binary data, as provided by the different sensors. The info.xml file is supposed to contain all (really *ALL*) meta data, that is relevant for that specific measurement like camera type, camera settings, timestamp, location... (e.g. info.xml, here: https://gist.github.com/dlebauer/75429d046badb100483b) . All meta data xml files will have the same layout, the are essentially a key value dictionary. The binary data format depends on the different sensors/cameras and contains the data as provided by the sensor. In case of the Hyperspec cameras these files will be hypercubes, in case of the 3D scanner these files will be ".ply" point clouds.

The original [proposal from Lemnatec]([Proposed filesystem](https://github.com/terraref/lemnatec-data/tree/84ccc82f34ea9b775ffdbb4982db4415cc0b351f/sample-data)) (details below) is a filesystem organized as `Sensor/YYY-MM-DD/YYY-MM-DD__HH-MM-SS/`, with each directory containing a raw file and a text-based metadata file. 

#### Location Dependent (e.g. Meteorology)

def: Sensors with a constant data flow, independent of the current location of the gantry, like wind or temperature sensors. These sensors will deliver data all the time, 24/7 in short intervals.

The folder "locationIndependent" stores the constant stream of gantry location independent sensor data. It is divided in sub folders by date and then by timestamp. Each timestamp folder contains a single info.xml file, that contains the data and meta data of all sensors (an example is here: locationIndependent\2015-11-22\2015-11-22__00-00-15\info.xml).

* Can this be written out to a standard csv file (e.g. the met files used by Fluxnet/Ameriflux)?

gsrohde, 2015-09-04:
See my comment on the Gist.

gsrohde, 2015-09-04:
Note on Summary item 3: Will there be enough files generated during one second to warrant the level of granularity implied by the directory name `HH-MM-SStz/`.  Also, I question the wisdom of a naming schema for directories where there exist many possible names for the same moment of time: A directory called `21-36-07CDT` could equally well be called `20-36-07CST`.  Even the parent directory might change depending on the time zone value used: `2015-09-04/00-30-00CDT` corresponds to `2015-09-03/23-30-00CST`.  Perhaps this isn't a real problem if we always agree either to always use UTC as the time zone or always agree to use the current local time zone (alternatively, always the current local standard time or always the current local daylight time) where the measurement was made as the time zone so there is always one canonical name for a given sensor.

gsrohde, 2015-09-04:
### Metadata

#### geojson:

As far as I can tell, geojson is _not_ a native Postgres data type, nor is it one of the data types added by the PostGIS extension.  (Perhaps there is some other extension that adds it.)  PostGIS _does_ have `geometry` constructor and accessor functions that take and return geojson representations of a `geometry`.  But it has these for GML representations as well.

[See also my comments on the gist.]

dlebauer, 2015-09-04:
The hstore extension provides a hstore data type and hstore_to_json and
hstore_to_jsonb functions according to the text "PostGIS in Action" 2nd
edition

dlebauer, 2015-09-04:
Will reference all time stamps to UTC with an offset, a subset of ISO 8601.

It is important to note that AZ -where most data will come from - switches
time zones twice per year.

This is also discussed in PecanProject/pecan#602.

yanliu-chn, 2015-10-20:
I'd like to provide some feedback on geo metadata:
- for point cloud or weather data, i assume each record has x and y info. the metadata for each data file needs the projection information so that we data from different places are to be processed later in the pipeline we know how to reproject.
- the projection info can be implicit (e.g., always use lat lon), but may need to be explicit (e.g., if data is raster)
- you can find all projection definitions at http://spatialreference.org, but as part of the standard, we may require the definition to have at least one common format (epsg number or the proj4 format).

dlebauer, 2015-11-05:
@max-zilla to geojson format (this is a standard)? how to define projection? this is not necessarily part of the standard.

max-zilla, 2015-11-05:
GeoJSON format is outlined [here](http://geojson.org/geojson-spec.html). Four key members of a GeoJSON object:
* **type** can be:
  * Point, MultiPoint
  * LineString, MultiLineString
  * Polygon, MultiPolygon
  * GeometryCollection
  * Feature, Feature Collection

* **geometry** must have a member called "coordinates" with an array of positions - Point types will of course only have one position in the array.
  * a position is an array of numbers, in [x,y,z] order

* **crs** contains the coordinate reference system object. this is optional.
  * should be on the top-level GeoJSON object in a hierarchy (in feature collection, feature, geometry order) and should not be repeated or overridden on children or grandchildren of the object.
  * two mandatory members: **type** and **properties**
  * for a named CRS, "type":"name" and "properties": { "name": "urn:ogc:def:crs:EPSG::26912" }
  * Additional discussion about formatting the CRS object [here](http://gis.stackexchange.com/questions/15953/whats-up-with-the-geojson-spec-and-crs-as-a-irm)

* **properties** can be any arbitrary object or null. LemnaTec's "metadatadictionary" could essentially go into this entirely.

---

So, I think this should be able to account for arbitrary CRS in addition to basic lat/lon if no CRS is specified. A sample GeoJSON blob based on [LemnaTech's example JSON object](https://gist.github.com/dlebauer/4a20f0d4512bbe2d1cdb) might look like:

```
{
  "type": "Point",
  "geometry": {
    "type": "Point",
    "coordinates": [
      -115.809165, 37.251581, 15
    ]},
  "crs": {
      "type": "name",
      "properties": { 
        "name": "urn:ogc:def:crs:EPSG::26912" 
      }
    },
  "properties": {
    "camera": {
      "fixed": {
        "type": "stereo vis",
        "cameraA model": "allied vision GT3300"
        "cameraB model": "allied vision GT3300"
        "cameraA_serial_number": "1234567",
        "cameraB_serial_number": "2345678",
        "location_on_gantry": "side at higher y values"
        "image_width": "3296",
        "image_height": "2472",
        "rgb_bayer_pattern": "BayerGR8",
        "another_data_field": "... "
      },
      "variable": {
        "exposure": "123",
        "gain": "456",
        "whitebalanced": "30"
      },
      "optic": {
        "attribute1": "value2",
        "attribute2": "value2"
      }
    }
  }
}
```

dlebauer, 2015-12-10:
@max-zilla and @rjstrand please discuss. I think we still need details of what data will be writing from @rjstrand and then Max can document and provide examples for Lemnatec to implement

max-zilla, 2016-01-04:
For initial organization, LemnaTec could write the sensor outputs to a directory structure like this, as outlined above:
* SensorName
  * CaptureDate
    * CaptureTime
      * Image & Metadata

Then, a script would iterate through the images for each sensor and use the API to push the images to our Clowder system, sending the contents of the JSON metadata file as metadata without requiring the JSON metadata file itself to be sent. Optionally, subsets of the metadata can also be sent to a BETYdb instance with proper parameters.

Some particulars about the metadata:
* The date/time/sensor information **should be included in the JSON metadata**, rather than only be represented in the folder paths. The folder structure at LemnaTec can be whatever is convenient for LemnaTec (unless @dlebauer or others have particular requirements) because Clowder will provide the search and filter capabilities for meta fields.

* All times should be UTC with an [offset integer](http://wwp.greenwichmeantime.com/info/timezone.htm) for clarity (e.g. US Central Standard is GMT-6).

* Geometric metadata (e.g. location of a captured image) should be stored in GeoJSON format as I outline above. We can define a field name, e.g. "file_location" that we should expect to be geo metadata. **Unlike my GeoJSON example above, we should not include other metadata in the "properties" of the location object** - those metadata should live alongside the geo data, not within it.

* To the extent we can standardize fields across sensors/cameras we should do so. @dlebauer's example addresses this with sub-objects for "fixed location attributes", "common sensor variables", "specific optic settings" and the like:
```
"fixed": {
        "type": "stereo vis",
        "cameraA model": "allied vision GT3300"
        "cameraB model": "allied vision GT3300"
        "cameraA_serial_number": "1234567",
        "cameraB_serial_number": "2345678",
        "location_on_gantry": "side at higher y values"
        "image_width": "3296",
        "image_height": "2472",
        "rgb_bayer_pattern": "BayerGR8",
        "another_data_field": "... "
      },
      "variable": {
        "exposure": "123",
        "gain": "456",
        "whitebalanced": "30"
      },
      "optic": {
        "attribute1": "value2",
        "attribute2": "value2"
      }
```
This standardization could be somewhat straightforward once we look at typical outputs for all the various sensors.

* Clowder supports [JSON-LD format](https://en.wikipedia.org/wiki/JSON-LD), which allows one to attribute reference information or data links for fields in a JSON object. In effect, we could have a metadata field for Camera Model with an associated link to the manufacturer specs for that camera model (for instance), to avoid needing to replicate those specs for each image from that model. This is just one example, but we can point to authoritative sources for certain fields/measurements/concepts that help 3rd parties understand the data.

dlebauer, 2016-01-22:
Here are comments from Markus Radermacher at Lemnatec :

---

A first implementation of writing out meta data as json has been finished in our software now and is attached to this mail.
 
At the current state of development six different kinds of meta data are available. Hence you find six different object sections:

*  user_given_data
  * an arbitrary number of key value pairs, provided by the user within the measurement.
  * The key value pairs here are fully flexible. They can even be changed between any to sensor measurements.
*  gantry_fixed_data
  * all information about the gantry system that will not change during a measurement run. Once we have figured out the final set of key value pairs the entries here are fixed.
*  gantry_variable_data
 * all the current settings of the gantry hardware during a measurement like current location of all axis and state of the lights. Once we have figured out the final set of key value pairs the entries here are fixed.
*  sensor_fixed_data
  * fixed dataset of the specific sensor. The key value pairs will be different from sensor to sensor but will be fixed for the same sensor.
* sensor_variable_data
  * all current settings of the specific sensor. The key value pairs will be different from sensor to sensor but will be fixed for the same sensor.
*   measurement_additional_data
  *  all additional information that is available for a specific sensor measurement like start and end time of the measurement.

example:

```json
{
  "lemnatec_measurement_metadata": {
    "user_given_data": {
      "treamtment": "water stress",
      "plant id": "plantName",
      "experiment": "name"
    },
    "gantry_fixed_data": {
      "dimension_x [m]": "200",
      "dimension_y [m]": "20",
      "dimension_z [m]": "6"
    },
    "gantry_variable_data": {
      "location x [m]": "20",
      "location y [m]": "10",
      "location z [m]": "3",
      "vel x [m/s]": "0.1",
      "vel y [m/s]": "0.1",
      "vel z [m/s]": "0.0",
      "light1 is on": "true",
      "light2 is on": "true",
      "light3 is on": "true",
      "light4 is on": "false"
    },
    "sensor_fixed_data": {
      "sensor type": "stereo camera",
      "cameraA model": "allied vision GT3300",
      "cameraB model": "allied vision GT3300",
      "cameraA_serial_number": "1234567",
      "cameraB_serial_number": "2345678",
      "location_on_gantry": "side at higher y values",
      "image_width": "3296",
      "image_height": "2472",
      "rgb_bayer_pattern": "BayerGR8",
      "another_data_field": "..."
    },
    "sensor_variable_data": {
      "gain": "2.4",
      "exporsure [ms]": "12",
      "red balance ratio": "34",
      "blue balance ratio": "45"
    },
    "measurement_additional_data": {
      "measurement_starts_at": "2015.12.16 11:11:22:801",
      "measurement_ends_at": "2015.12.16 11:11:56:808"
    }
  }
}
```

czender, 2016-01-25:
My understanding is that there will be one file as above for each camera image. Each camera image will actually be a pair of files of the data/data.hdr and whiteReference/whiteReference.hdr variety posted at https://github.com/terraref/computing-pipeline/issues/38
We will combine the JSON with the image data into a netCDF4 file for compression and analysis. @dlebauer please correct any mis-impressions...

My inclination is to parse all this JSON metadata into an attribute tree in the netCDF4/HDF file.
The file level-0 (root) group would contain a level-1 group called "lemnatec_measurement_metadata", which would contain six level-2 groups "user_given_data"..."measurement_additional_data" and each of those groups would contain group attributes for the fields listed above. We will use the appropriate atomic data type for each of the values encountered, e.g., String for most text, float for 32-bit reals, unsigned byte for boolean,... Some of the "gantry variable data" (like x,y,z location) will need to be variables not (or as well as) attributes, so that their time-varying values can be easily manipulated by data processing tools. They may become record variables with time as the unlimited dimension. 

Any thoughts/concerns/questions about this strategy before we proceed?

dlebauer, 2016-01-26:
@czender 

Just to avert confusion - this issue refers generically to all sensors on the gantry. So perhaps we should move the conversation to the implementation of a hyperspectral data pipeline in terraref/computing-pipeline#38, where the field names are located (and for the record, I believe CF does a fair job of providing these). I'll paste the rest of the comments there ...

FlyingWithJerome, 2016-02-13:
Hi everyone, I'm Jerome and right now an assistant working with Prof. Zender in converting json to netCDF. 
@dlebauer Thank you for your sample jsons: https://gist.github.com/dlebauer/4a20f0d4512bbe2d1cdb
 But I couldn't successfully parse them, and I found that some commas have gone missing?

dlebauer, 2016-02-13:
@FlyingWithJerome Sorry for the confusion - those aren't properly formatted json because of the crude loop I wrote to print the file name and file contents. 

I've created a new gist with the current draft of the metadata for the two hyperspectral sensors: https://gist.github.com/dlebauer/78bbb0ed2da9172f31fb (these are valid json according to jsonlint.com

For reference, these files along with raw files and metadata from the field sensors are in this folder: https://uofi.box.com/terraref-samples-20160210

It contains both a .zip file and all of the files unzipped, including the files called metadata.json 

I've updated the output at https://gist.github.com/dlebauer/4a20f0d4512bbe2d1cdb but its still not valid json.

czender, 2016-02-18:
The first results from the JSON->netCDF attributes parser are in. We will re-organize the structure and turn into attributes.
http://dust.ess.uci.edu/tmp/test.json
http://dust.ess.uci.edu/tmp/test.nc

FlyingWithJerome, 2016-02-20:
We tuned up the JSON->netCDF parser and now it can export a better organized netCDF file: 
https://drive.google.com/a/uci.edu/file/d/0B7JWv6BOtHUzZWd6U3FSbzdYRGc/view?usp=sharing

FlyingWithJerome, 2016-02-25:
And here is the newest version of netCDF:
https://drive.google.com/a/uci.edu/file/d/0B7JWv6BOtHUzOHB5b1RMNUFNd2s/view?usp=sharing
@dlebauer Sorry for the previous one, I had not assigned value for it and there were still some flaws.
This one would be better.

dlebauer, 2016-02-25:
here is the ncdump: https://gist.github.com/dlebauer/3afb356094bd89e95b00

czender, 2016-03-03:
ncap2 script to combine 2D TR image data into single 3D variable working. Example:
/* Purpose: Combing 2D TR image data into single 3D variable
   Usage:
   ncap2 -4 -v -O -S terraref.nco in.nc out.nc */

*sz=926;

defdim("band",sz); 

allbands[band,x,y]=0.0f; 

for(*idx=1;idx<=sz;idx++)
{
 {@loop=snprint(idx,"Band%d");}
   allbands((idx-1),:,:)=*@loop;   
}

// Sanity check: Make sure Band10 minus tenth band is zero
err_chk=(Band10-allbands(9,:,:)).total();

FlyingWithJerome, 2016-03-10:
the cdl dump for the merged netCDF file (data generated by Prof. Zender and metadata generated by me) is available now on gist:
https://gist.github.com/e397c63918bdfb3cbe45

czender, 2016-03-11:
You are right. Jerome inadvertently used the old version that had 926 variables, because that was the only data file he had. Workflow is the same with the one 3D variable version. Will update the gist soon.

closed 2016-03-17 by dlebauer