terraref/reference-data #3: Comment on format for meteorological variables exported from Lemnatec platform

dlebauer, 2015-08-07:
PEcAn uses [Climate Forecasting 'standard names' and 'canonical units'](http://cfconventions.org/Data/cf-standard-names/29/build/cf-standard-name-table.html) conventions (widely used in climate / met community) for meteorological and ecosystem-level mass and energy balance variables.

Here are some examples (note that we can change from canonical units to match the appropriate scale, e.g. "C" instead of "K".

|CF standard-name | units |
|:------------------------------------------|:------|
| time | days since 1700-01-01 00:00:00 UTC|
| longitude | degrees_east|
| latitude |degrees_north|

## The variable names should be `standard_name`

| CF standard-name                          | units | bety         | isimip       | cruncep | narr  | ameriflux |
|:------------------------------------------|:------|:-------------|:-------------|:--------|:------|:----------|
| air_temperature                       | K     | 
| air_pressure                         | Pa    | 
| mole_fraction_of_carbon_dioxide_in_air    | mol/mol |
| moisture_content_of_soil_layer            | kg m-2 |        
| soil_temperature                          | K     | 
| relative_humidity                         | % | 
| specific_humidity                     | 1 | 
| water_vapor_saturation_deficit            | Pa    |  | surface_downwelling_longwave_flux_in_air | W m-2 |  
| surface_downwelling_shortwave_flux_in_air| W m-2 |
| surface_downwelling_photosynthetic_photon_flux_in_air | mol m-2 s-1 |
| precipitation_flux                    |  kg m-2 s-1 |          
| wind_speed                                | m/s   | 
| eastward_wind                         | m/s   |
| northward_wind                        | m/s   |

* standard_name is CF-convention standard names
* units can be converted by udunits, so these can vary (e.g. the time denominator may change with time frequency of inputs)

rjstrand, 2016-02-02:
Hello all, 

The gantry is powered up and we are beginning to bring sensors online.  We need to think about the handling of the meteorological data.  The list of instruments is as follows:

1) Wind speed
2) Wind direction
3) Air temperature
4) Humiidity
5) Air pressure
6) Solar intensity*
7) Solar angle*
8) Precipitation

I am checking on the solar values... I'm looking at a simple translation from German and want to get more detail.

First, Markus Radermacher asked for a proper name... At USDA we generally referred to these as MicroMet.  

Not only do we need to decide on a format, but also whether we want instantaneous values for some these values, or whether averages would be sufficient.  We will need to discuss averaging times and data accumulation time; i.e. do we want to a file to contain 5 minute averages over an hour.

- Bob

dlebauer, 2016-02-02:
to be consistent with how we treat other data Id say start by keeping anything meaningful. We can trim later. I suspect especially keep the high frequency sonic anemometer and downwelling light sensors. Isn't there also CO2?

If you have variables not mentioned above in this thread please see the table of CF standard names http://cfconventions.org/Data/cf-standard-names/30/build/cf-standard-name-table.html

dlebauer, 2016-03-11:
@czender and @terraref/standards-committee  this the first draft of the environmental sensor data from Arizona. It is a 4MB json file; it appears data is collected every two seconds and a new file is written every two minutes. 

[2016-02-15_21-20-08_enviromentlogger.json.txt](https://github.com/terraref/reference-data/files/170133/2016-02-15_21-20-08_enviromentlogger.json.txt)

```json
  "environment_sensor_set_reading": {
    "timestamp": "2016.02.17-00:39:18",
    "add metadata about weather station and par sensor here": "...",
    "weatherStationSunDirection": {
      "value": "241.8823816645",
      "unit": "degrees",
      "rawValue": "6.71895504623554"
    },
    "weatherStationAirPressure": {
      "value": "1011.1294900357",
      "unit": "hPa",
      "rawValue": "8.12036500137333"
    },
    "weatherStationBrightness": {
      "value": "122.9677419355",
      "unit": "kilo Lux",
      "rawValue": "6.12964262825404"
    },
    "weatherStationRelHumidity": {
      "value": "14.2765587329",
      "unit": "relHumPerCent",
      "rawValue": "1.42765587328715"
    },
    "weatherStationTemperature": {
      "value": "28.8100833155",
      "unit": "DegCelsius",
      "rawValue": "6.88100833155309"
    },
    "weatherStationWindDirection": {
      "value": "351.5402691733",
      "unit": "degrees",
      "rawValue": "9.76500747703482"
    },
    "weatherStationPrecipitation": {
      "value": "0.1108655354",
      "unit": "mm/h",
      "rawValue": "0.010986663411359"
    },
    "weatherStationWindVelocity": {
      "value": "2.5873592334",
      "unit": "m/s",
      "rawValue": "0.43122653889584"
    },
    "sensor par": {
      "value": "594.8226101819",
      "unit": "?mol/(m^2*s)",
      "rawValue": "7.81296544988402"
    },
    "spectrometer": {
      "maxFixedIntensity": "16383",
      "integration time in ?s": "5000",
      "band": {
        "wavelength": 337.70483,
        "spectrum": 1496.0
      },
      "band": {
        "wavelength": 338.16013791719934,
        "spectrum": 1500.0
      },
      "band": {
      "wavelength":"repeated ~ every 1/2 nm "
      },
      "band": {
        "wavelength": 823.98401644529815,
        "spectrum": 1513.0
      }
    }
  }
}{
  "environment_sensor_set_reading": {
    "timestamp": "2016.02.17-00:39:21",

```

dlebauer, 2016-03-17:
I have moved the data product description to the [documentation](https://github.com/terraref/documentation/blob/master/environmental_data.md) in https://github.com/terraref/documentation/commit/ae09deab200b8415af35df19dbd83ba2343eab15 and the remaining changes in the format output by the Lemnatec sensors to #26. 

And will now close this issue.

closed 2016-03-17 by dlebauer