terraref/reference-data #5: Identify existing standards, conventions, and ontologies to use, integrate with.

dlebauer, 2015-08-14:
There are a few classes of data:

1. Crop physiological traits
2. Agronomic meta-data
3. Sensor Output
4. Derived metrics from image analysis (spectral indices like NDVI, geometric statistics like height, convex hull)
4. Genomics

For each, 

* what are the key existing standards / conventions? 
* What are the 'unique identifiers'? 
* What are the use cases (queries that we should support)?

dlebauer, 2015-08-14:
Date-Time: `YYYY-MM-DD hh:mm:ssZ`: 
* based on [ISO 8601](https://en.wikipedia.org/wiki/ISO_8601) . Optional offset for local time; precision determined by data (e.g. could be `YYYY-MM-DD` and decimals specified by a period.

rachelshekar, 2016-07-07:
@gsrohde - have you started on this yet?

dlebauer, 2016-07-07:
@craig-willis please discuss with @gsrohde

craig-willis, 2016-07-14:
Per discussion with @dlebauer, the remaining work on this task is to ensure that the data standards documentation is up-to-date based on discussions in the subissues.

craig-willis, 2016-07-15:
@dlebauer is there a list of known/expected data products similar to the NEON [catalog](http://www.neonscience.org/sites/default/files/basic-page-files/NEON_DOC_002652.pdf)? If not, would it be worth me trying to put something like this together? I think it would be helpful to ensure full coverage of existing standards for this task.

dlebauer, 2016-07-15:
@craig-willis 

>  is there a list of known/expected data products similar to the NEON catalog?

not beyond what is already in the documentation, and not summarized in a table you linked, but this is a good idea.

> map from the Field Scanner Data to the current locations in the Globus Endpoint.

That is a good idea, although I think the data products will either be in as-yet non-existent directories (not under raw_data/ua_mac/moving_sensors), and at this point I am not sure of the directory structure that Clowder users, and I don't think it is organized by sensor. So we should discuss 1) if it makes sense to provide access via Globus (beyond just to the terraref team) and 2) if we should ask Clowder to organize the files (or perhaps make a human readable directory of soft-links)

craig-willis, 2016-07-18:
@dlebauer perhaps I could just maintain this as internal documentation for now.  Is there a place we put internal docs, e.g., Wiki? I'm using the NCSA Confluence for my personal notes now.

Another question: I'm trying to map different processes (e.g., extractors) to the different data levels/products.  Max walked me through 3 extractors: Geospatial, PlantCV, and Hyperspectral.  Is there a list of current/planned extractors for the initial data product release?

dlebauer, 2016-07-19:
@craig-willis You can use the documentation repository (github.com/terraref/documentation) or google drive for drafting this.

>  Is there a list of current/planned extractors for the initial data product release?

Not that I can point to right now - many exist as open issues in the computing pipeline repository but I can start writing one up.

craig-willis, 2016-08-01:
@dlebauer I've been writing up notes from my analysis if this and #18 and #31 in a Google Doc 

https://docs.google.com/document/d/13gXD_OVLffm0hqahDZ3tUvru8IV1fRfM6DiuOcfjr3s/edit?usp=sharing

and accompanying spreadsheet:

https://docs.google.com/spreadsheets/d/1qu3LYomGIiC4Lmzler-bH8jvyfR-GA6lDl5Go0NhO2c/edit?usp=sharing

They are still rough, but I wonder if any of it might be useful for the upcoming meetings.

craig-willis, 2016-11-15:
@rachelshekar 
It's not clear to me whether I can call this "done"  given the earlier work on 
https://github.com/terraref/documentation/blob/master/existing_data_standards.md

craig-willis, 2016-12-15:
Closing this issue. Updating this documentation will likely be ongoing but can be incorporated into future tickets as needed.

closed 2016-12-15 by craig-willis