terraref/reference-data #9: End-user API to query data

dlebauer, 2015-09-01:
* [ ] define use cases (@dlebauer) 
* [ ] what features should an API have?
  * query by genotype? query by location? time?
  * what features will make it easy to document and integrate with other API's? 
  * What is the value of an API Framework such as [Swagger](http://swagger.io)?
* [ ] what framework does the current BETYdb API use? What features does it have? What would make it easier to use / extend?   
  
## references

* [BETYdb API documentation](https://www.authorea.com/users/5574/articles/7062/_show_article)
* [Zapier Intro to API design]https://zapier.com/learn/apis/chapter-6-api-design/
* [Wickham API design best practices](https://cran.r-project.org/web/packages/httr/vignettes/api-packages.html)

max-zilla, 2015-11-19:
Here are some examples, Not all of these will be valid but perhaps it will engender discussion... these examples might be scattershot across the various components.

**End-users working with existing data in Clowder**
1. Browse a gallery of daily images of the same plant across 2 weeks.
2. Search for images by metadata fields (e.g. one particular sensor with a value exceeding some threshold), and download the results
3. Search for images with ranges/fuzzy-matching (e.g. all images of plant with height +-X cm or X% of nominal value)
4. Read about the specification of the sensors that gathered these data to understand what the value represents.

**end-users with a bit more ambition...**
5. Show a histogram of some metadata field/trait values for all images in a Dataset (this one is getting out there...)
6. Search for all images with some value outside 2 standard deviations of the average value for that day and field
7. Load a selection of images from one of the use cases above into an analysis VM with PlantCV, etc available for study.
8. Export some tabular structuring of selected metadata fields from a batch of images, e.g. a CSV with identifier, trait1, trait2, trait3, date

**developer-oriented**
9. augment the Clowder image extractor to ingest a new data format that is proprietary/internal
10. write SQL queries against BETYdb directly to bulk-upload data into e.g. the Traits table
11. Stand up new instance of BETYdb for my own use

dlebauer, 2016-05-06:
1. Find all plots where [accession X] was planted
  1. Find all points and dates where seeds of [accession X] were planted
  2. find bounding boxes associated with these plots
  3. use bounding boxes to query data from [sensor X] (e.g. raster image) from [date range] (in Clowder) 
2. find all trait values (in BETYdb traits table, summarized to plant or plot level) associated with [accession X] collected using [method X] 
3. after sensor data have been processed geospatially orthorectified and aligned with overlap and artifacts removed, use bounding boxes to clip / select data from some set of sensors

1. return experimental design 'plot plan' from PostGIS database (BETYdb)
2. return information about accessions from BETYdb cultivars table and also in BMS, 
3. return information about lineage, seed packets, experimental design from BMS, 
4. Load field measurements collected in FieldBook APP into BMS
5. Import traits from BMS to BETYdb and vice-versa
The reason we are using both BETYdb and BMS despite substantial overlap is that BETYdb has better support for geospatial data, numeric traits, and large external raster files; BMS does a better job at tracking experimental design, lineage, genomics, and can (apparently now or in the near future) import directly from the FieldBook app.

max-zilla, 2017-01-12:
https://github.com/terraref/computing-pipeline/issues/231
Created a discussion issue to continue this discussion in context and linked here. Closing this issue to consolidate that discussion.

closed 2017-01-12 by max-zilla