terraref/reference-data #15: Develop talking points for project

rachelshekar, 2015-10-08:
- [ ] Ask Nick for Claire's time
- [ ] Compile information for Claire 
- [ ] Schedule time to meet with Claire
- [ ] Claire finishes draft 1
- [ ] Feedback from @dlebauer on Draft 1
- [ ] Claire finishes draft 2
- [ ] Feedback from @dlebauer on Draft 2

rachelshekar, 2015-10-08:
Oct 16	Draft 1 ready
Oct 20	Draft 1 feedback
Oct 30	Draft 2 ready
Nov 3	Draft 2 feedback
Nov 10 	Ready

rachelshekar, 2015-10-08:
[TERRAref Background.docx](https://github.com/terraref/reference-data/files/10117/TERRAref.Background.docx)

closed 2015-11-10 by dlebauer

doctorroboto, 2015-11-10:
@dlebauer and team,

Is this something we can share with the LemnaTec software team, to explain the role of their system in greater detail and to expedite their delivery of new sensor data / metadata products?

rjstrand, 2015-11-11:
I have been talking with Justin Manzo over the past week about LemnaTec's contributions.  Some of the delay on the part of my company stems from the effort to simply meet production, testing, and installation deadlines.  Even with the modified schedule, ARPA-E has pushed a very tight timeline to get the system built.  Granted, the end-date is really set by the growing season in Arizona, this is still an aggressive schedule.  Things are very busy over in Germany as well as here in the U.S.  I am currently split between the Arizona field system and the expansion of the greenhouse system at the University of Nebraska.  The bulk of my work at the UNL system should end before Thanksgiving.

The document above is a good start.  However, it would be very helpful for me to have a few things to show my German colleagues.  I imagine that much of this is in the project proposal, but Justin doesn't think that document is available for distribution.

1) An overview of the facets of the Cat 5 project. (Gantry installation and testing, testing and implementing data transfer, data storage and processing, data distribution, efforts on the genetic side, etc. 

2) Timelines of each facet, possibly showing dependencies or critical points shared between those timelines,  More importantly, showing milestones, and noting those milestones that are currently at risk without more robust documentation at this point.

In the end, I think this a clear picture of would be helpful for many involved in managing this project.

Maybe a good start would be to focus on the gantry/data storage & processing timelines... 

Thanks,

- Bob

doctorroboto, 2015-11-11:
Perfect. And that's right Bob, I'm highly confident that we can't share the Cat5 proposal can't be shared outside of ARPA-e. Joe and team are checking on what can be shared by the gov't folks, but if there's something Dr. LeBauer or the Danforth folks could share publicly, that would streamline the request.

We can talk about this in greater detail tomorrow, and see what there is to be done about getting everybody more data, formats, etc.

dlebauer, 2015-11-11:
I spoke with Todd and the majority of the proposal is not proprietary. But I've tried to summarize bits and pieces of it in the about page http://terraref.ncsa.illinois.edu/about/. 

I've also attached a summary that provides more detail about the proposed timeline, though it hasn't been revised following negotiations about specific milestones. [arpa-e-terra-timeline-summary.pdf](https://github.com/terraref/reference-data/files/32203/arpa-e-terra-timeline-summary.pdf)