terraref/reference-data #17: Define expertise of current committeemembers

dlebauer, 2015-10-09:
This will help understand 
* [ ] create a google form to let people check boxes (or select expertise levels 1-5) as well as provide free-form text responses
* [ ] send to committee members 

* categories: remote sensing, image analysis, physiology, breeding, genomics, informatics, ... 
* levels: 0: never heard of it, 1: have heard of it but don't know, 2: have read the wikipedia page 3: could contribute to the wikipedia page (grad level) 4: use it as a tool actively in research 5: leader in field / expert in developing new methods

rachelshekar, 2015-10-19:
See https://github.com/terraref/admin/issues/3

I suggest that you send this list to the committee before the meeting and ask for changes during the meeting

davidlee2, 2015-10-19:
At the moment I’m trying to pin down a couple more TERRA individuals about having the next call the afternoon of the 28th or 30th.

See terraref/admin#3<https://github.com/terraref/admin/issues/3>

dlebauer, 2015-11-03:
The table of participant expertise is here: https://docs.google.com/spreadsheets/d/14Z-Y2MmVy94X561VJdecBSu577VQ4H6G65gh8p_pkD4/edit?usp=sharing and has been shared with the group

closed 2015-12-03 by dlebauer