terraref/reference-data #21: Give Roger access to beta testers

rachelshekar, 2016-01-07:
Noah Fahlgren, Rob West at WUSTL and Charlie Zender

Include sample data

Private owncloud server?

yanliu-chn, 2016-01-07:
the ldap groups they should be assigned to: prj_cg_users and prj_cg_arpae

yanliu-chn, 2016-01-07:
If they don't have ncsa account, please let me know their email addresses. i can contact them to create sponsored guest forms.

dlebauer, 2016-01-07:
@yanliu-chn email addresses and github usernames are

| name | email | github |
|----|----|----|
| Charlie Zender | zender_at_uci.edu | @czender |
| Noah Fahlgren | nfahlgren_at_danforthcenter.org | @nfahlgren |
| Robert Pless | pless_at_wustl.edu |  |

dlebauer, 2016-03-10:
This issue was moved to terraref/computing-pipeline#71

closed 2016-03-10 by dlebauer