terraref/reference-data #24: Define data products, repositories, publication venues

dlebauer, 2016-03-11:
We will be producing diverse data products, protocols, and software (hereafter ‘products’).  These are intended for distribution and reuse with attribution Products will be continuously revised, updated, and expanded. New versions released annually, with opportunity for community feedback. Other teams will potentially contribute protocols, similar types of data. Data sets from different teams could be merged for analysis (ie more than one attribution)

#### Key Products:

* raw data and data products 
  * fifteen + sensors, 
  * trait data: direct measurements and sensor derived
  * genomics
* software  
  * pipeline components like Clowder
  * databases (schemas) 
  * interfaces
* protocols 
  * field measurements
  * sensor calibration

#### Approach

* Define products defined by 'level' of processing (sensu NASA)
* Clearly mark data with conditions of reuse (CC-By by default) and how to cite
* Versioning of products by year of release
  * each year a new version of data products will be released
  * new version will have increased number of data products, some data products will be re-analyzed
  * define what data can be 'retired', based on information content or reproducible pipeline from raw data   
* Data repositories for different data products
  * Illinois data bank (UI Library) for large sensor data
  * Genomics data: phytozome?
  * Others? BMS hosted by Cyverse.
* Submit letter of inquiry to Nature Scientific Data or similar venue regarding one or more data papers to define scope of data.

See also meeting notes: https://goo.gl/QhpwcH

dlebauer, 2016-03-18:
@rachelshekar please review [submission instructions](http://www.nature.com/sdata/for-authors/submission-guidelines) and draft a letter of inquiry to Nature Scientific Data, regarding the timing of submission with respect to our timeline for developing data products and division of data sets into multiple submissions. http://www.nature.com/sdata/

Note Nature's endorsement of this effort: http://www.nature.com/ng/journal/v47/n2/pdf/ng.3213.pdf?platform=hootsuite

rachelshekar, 2016-03-18:
@dlebauer :I have created a draft at: https://docs.google.com/document/d/1lJHsUXdOooCoDAh0Gz8rGDIZb8K3-ujpdYBgL5ZqrbM/edit

dlebauer, 2016-03-23:
@rachelshekar I've revised your draft - could you please review once more, edit for clarity, and add anything necessary, send to Heidi for her feedback / additions?

rachelshekar, 2016-03-29:
@dlebauer  - I've made additional edits

dlebauer, 2016-08-09:
@craig-willis please review - and if it would be easier, open new issues that separately address how to define protocols and data product descriptions (unless they already exist).

rachelshekar, 2017-01-26:
@craig-willis and @dlebauer please review

rachelshekar, 2017-02-02:
URL should not be dependent on github or clowder.   Should we use Zenodo for persistent documentation? @craig-willis

craig-willis, 2017-02-23:
I've had a few discussions about DOIs as they relate to TERRA-REF data products and will have more in the coming weeks.  One idea that seems to make the most sense at the moment is to create a metadata package (ORE or otherwise) that references all of the files that make up a particular data product with their locations on ROGER/Globus/Clowder with checksums. This package should be pretty small and can be published to an external repository with instructions on how to access via TERRA-REF system.  Database data like BETY should follow the RDA recommendations for dynamic data, if not producing some archivable dump of the full database at a point in time.

dlebauer, 2017-02-23:
BETYdb does produce an archivable dump  https://terraref.ncsa.illinois.edu/bety/dump/bety.tar.gz

max-zilla, 2017-03-15:
meet w @dlebauer @craig-willis 

What should have DOIs for citations? And a persistent URL.
- sensor collection in Clowder ala IMLCZO? "TerraRef V0 hyperspectral data product"
- one DOI for entire TerraRef V0 data release?
- genetics/genomics vs. traits vs. sensors, but minimal number of citation options.
- ability for user to say "here are the files I used in my research, create a DOI to cite for my product but also cite the overall product"

Motivations:
- citability for publications
- do we need to specify an exact subset, or entire reference dataset
- don't want to fragment DOI more than necessary so we can have that cited many times instead of many DOIs cited a few times

craig-willis, 2017-03-15:
Starting to put together a design doc  https://docs.google.com/document/d/1bwx9uWRMfft7uR_IBGkJja9dZO-Uan_5fbikz5N6Eyw/edit#

craig-willis, 2017-03-30:
From https://github.com/terraref/reference-data/issues/104, regarding the genomics data:   "For publication, the raw data should be deposited in the NCBI SRA minimally. We can definitely get a DOI for the Data Commons collection as well. I opted out of that for now because we will still be adding derived data products to the collection (and possibly more raw data from related projects?). We may want to assign DOIs to sub-collections within the larger terraref collection in the event we want to keep adding more data here?"

@nfahlgren  David, Max  and I on 3/15 to discuss how we will be packaging the TERRA-REF data for publication.  The meeting centered mainly on a Nature Scientific Data paper (some meeting notes are in the Google doc linked in the above comment) and what we do will in part be driven by Nature's requirements.  One open question is whether there will be one DOI for the TERRA-REF v1 dataset or multiple subsets (traits, sensors, genomics).  Some of this will depend on how we want the data to be cited. Based on your above comment, it sounds like you've thought through this somewhat for the genomics side.

I'm tasked with putting together a preliminary "data package" that conforms to the Nature requirements, mainly to drive more discussions/feedback.  I'd welcome any input you have about how you see the genomics data fitting in.

dlebauer, 2017-03-30:
@craig-willis should we meet to convert this to an epic and define subtasks?

craig-willis, 2017-03-30:
@dlebauer   Sure -- I've already created https://github.com/terraref/reference-data/issues/106 for the preliminary data package with an April milestone, but happy to discuss further.

dlebauer, 2017-03-31:
@craig-willis then can this be closed in favor of #106? And open a new epic for protocolswith children 

 Where to publish / archive / reference protocols
 Sensor calibration #34
 Field measurements #45
 general approach to protocols #80

closed 2017-04-04 by craig-willis