terraref/reference-data #25: Further updates and revisions to Lemnatec Field metadata

dlebauer, 2016-03-17:
Here is a recent meta-data being generated by the Lemnatec Field system is here: https://gist.github.com/dlebauer/ccc1940fefbacaa60296

We will also need the following information:

* the position of each sensor within the box
* additional time information 
  * for sensors like VNIR/SWIR that are operated in line-scan mode, presumably we need the start time as well as the end time (currently it is not clear if the time stamp is the time that the file was finished writing, or otherwise)
  * for PSII sensor, we will need the time of the flash as well as the time that the fluorescence response was captured
* Time and location of any calibration / correction data (e.g. dark and white reference, etc)

* Calibration / correction data
   * 3x3 K matrix for geometric correction / transformation from sensor view to reality
  * Bandwidth, spectral resolution, FWHM, and / or spectral response
  * Downwelling solar radiation.
  * temperature (for FLIR)
  * Field of view (bounding box on a plane at 2m below sensor bay, which we assume is top of canopy) 

I'll note that any data that is stable in time (e.g. especially calibration data, but also position and structure of each camera) _could_ be stored separately. 

Anything else?

dlebauer, 2016-04-03:
@markus-radermacher-lemnatec

markus-radermacher-lemnatec, 2016-04-20:
Position of sensors in the box->
position of remaining sensors in the camera box will be added to meta data files from my colleagues Tino and Bob next week.

the position of the 0,0,0 -> 
we have one fixed coordinate system of the gantry, that can not be changed. But of course there should be a well defined transformation of this coordinate system to either a customer coordinate system (in which the seeding locations are defined) with a fixed point on the field or GPS coordinates. 
Furthermore once this transformation is available we are planning to include GPS coordinates in all meta data.

additional time information -> 
For the hyperspec cameras you have to calculate the time on your own, based on the gantry scan speed or frame period.
In the final setup there will be two modes for hyperspec cameas:
* free running, then you have to derive the point in time from the start timestamp and the frame period
* position trigger (still under construction), each 1mm of scan movement a hardware trigger leads to a hyperspec measurement, then you have to use the speed

For PS2 Sensor->
There are 102 bin files. The first (index 0) is an image taken right before the LED are switched on (dark reference). Frame 1 to 100 are the 100 images taken, with the LEDs on. There Frame 1 is tiggered by the LED on event. In binary file 102 (index 101) is a list with the timestamps of each frame of the 100 frames.
Right now the LED on timespan is 1s thus the first 50frames are taken with LEDs on the latter 50 frames with LED off. But its fully customizable by customer.
If there are any open questions, please let me know and we add this information to the meta data.

Time and location of any calibration ->
Right now there is no calibration.
Tino will work on a calibration setup, that will perform a special measurement (white target/ dark reference...) and add this information to the meta data.
But the whole calibration work flow needs more discussion.

Calibration/correction data->
Needs further discussion. 
Right now we are only planning to include the FoV info for each sensor.
Tino will work on Flir calibration.

TinoDornbusch, 2016-04-27:
@dlebauer by Bounding box of sensors you mean the theoretic coordinates in m on a plane 2m away?

IF so which format? as array? [xmin,ymin,xmax ymax]?

dlebauer, 2016-04-27:
@TinoDornbusch as an array or set of key:value pairs is fine. A standard format like geojson (see https://github.com/terraref/reference-data/issues/2#issuecomment-168700121) would be ideal, but transferring the information is most important. 

But if your interested, [geojson would define a bounding box](http://geojson.org/geojson-spec.html#id4) as:

```
  { "type": "Polygon",
    "coordinates": [
      [ [xmin, ymin], [xmin, ymax], [xmax, ymax], [xmax, ymin], [xmin, ymin] ]
      ]
   }
```

TinoDornbusch, 2016-04-28:
OK, will add the bounding box in this notation. Please note that it will be as text. Not sure how to write the text 100% like your example in XML.

Position of all cameras/sensors in box added to Metadata.

Downdwelling solar radiation is given in EnvironmentalSensors (PAR and Spectrometer).

Please check and Lets make an update on missing meta data tomorrow.

dlebauer, 2016-04-28:
> Downwelling solar radiation is given in EnvironmentalSensors (PAR and Spectrometer).

this is discussed further under 'changes to environmental sensors': terraref/reference-data#26

but for the spectrometer, I see this:

```
    "spectrometer": {
      "maxFixedIntensity": "16383",
      "integration time in ?s": "5000",
      "band": {
        "wavelength": 337.70483,
        "spectrum": 1504.0
      },...
```

that doesn't provide a measure of energy flux. I might expect a third field 'radiance' or other measure of energy flux

TinoDornbusch, 2016-04-28:
I added the field of view as Metadata. From the postion of camera in box & FoV you can compute bounding box.

TinoDornbusch, 2016-05-05:
I have added information on userGiven.xml to get first information on experiment Metadata in.

Important are two entries: "campaign" and "mission or scan".

"campaign" somewhat denotes a series of different measurements grouped together. Currently we run "seedling emergence and early vigor" experiments.

"mission or scan" denotes the a scan done with a specific set of sensors.

At the moment I have setup the following scans on regular / daily basis:

"3D Scan Field 0.33MperS" - 3D scan on 30m north side. Note that this data will appear once the calibration is made.

"3D Scan 4m" - 3D scan (multiple times per day) at a given location. Smaller data set to test

"HyperspecWhiteTarget-LightOn" - stereoVis, SWIR,VNIR, FLIR over a piece of lawn, white target, color chart and flowers. Also done at different times of the day. Halogen lights are switched on. 
 
"HyperspecWhiteTarget-LightOn" -  same as above but Halogen lights are switched off.

"HyperspecWhiteTarget_FullLine" -  stereoVis, SWIR,VNIR, FLIR, white target over a full scan width of 21.3m.

"emergence full plot 0.1m" - stereoVis, FLIR on 30m north side. Daily scan to monitor emergence.

Due to some error in XML syntax (comments not allowed), user given meta data on these experiments is unavailable for some of the previous measurements. If these information are of interest, I have scan log files denoting the start of each scan.

We really need to discuss the experimental Metadata with both, data analysts and experimentalists.

dlebauer, 2016-05-12:
@TinoDornbusch please use [ISO-8601 standard](https://en.wikipedia.org/wiki/ISO_8601) standard for all dates and times. i.e. "YYYY-MM-DD".

dlebauer, 2016-05-12:
@TinoDornbusch I see that you have added the bounding box information - which is great - but it is not clear what information the bounding box and field of view provides. We will need to calculate the coordinates in [x, y, z] for the bounding box in the focal plane for each image.

Here is a sample from the file /projects/arpae/terraref/raw_data/ua-mac/MovingSensor/stereoTop/2016-05-10/2016-05-10__14-10-32-010/a3012717-3e46-4f3b-bf3b-8606d457e3e4_metadata.json  

```
      "optics focus setting (both)": "2m",
      "optics apperture setting (both)": "4",
      "Output data format": "Bayer GR8 3296x2472",
      "sensor purpose": "capture RGB images",
      "location in gantry system": "camera box, facing ground",
      "location in camera box X [m]": "0.877",
      "location in camera box Y [m]": "2.276",
      "location in camera box Z [m]": "0.578",
      "Field of view at 2m in X- Y- direction [m]": "[1.857 1.246]",
      "Bounding Box [m]": "[1.857     1.246]",
```

Questions:

For the 'Bounding Box and Field of view fields: 
* Are these the width (x) and height (y)? 
* If so, how can these be converted to coordinates? 
* will 'field of view' be fixed (assuming 2m) and 'bounding box' vary with focal plane?

dlebauer, 2016-05-12:
@TinoDornbusch I also noticed that the file above does not contain the location of the sensor box, only the location of the sensor within the box. Earlier metadata files had this information under 'gantry_system_variable_metadata'

e.g. from VNIR/2016-05-05/2016-05-05__08-11-33-268/2a071022-6c45-459d-839a-73f0ab69d73f_metadata.json

```json
    "gantry_system_variable_metadata": {
      "Time": "05/05/2016 08:11:33",
      "Position x [m]": "203.6",
      "Position y [m]": "0",
      "Position z [m]": "0.67",
      "Velocity x [m/s]": "0",
      "Velocity y [m/s]": "0",
      "Velocity z [m/s]": "0", 
```

TinoDornbusch, 2016-05-13:
@dlebauer  will adjust times in Metadata....Changes in source code forwarded to IT

I thought bounding box at 2m distance was of interest. Else please specify what you need.

Gantry_system_variable_metadata contains the position of the reference point in the camera box, to which all sensors are referenced to.

dlebauer, 2016-05-13:
* I don't see the gantry position in the file referenced above. Can you
confirm that these are being generated?
* I don't understand the data provided in the bounding box. There are two
values provided. Are these coordinates or dimensions? Preferably
coordinated referenced to the gantry [0,0,0] point.

TinoDornbusch, 2016-05-13:
@dlebauer I do not get it gantry position was X=203.6m Y=0m Z=0.67m as specified in gantry_system_variable_metadata. Sorry

Field of view means a box with dimension in X and Y given with the camera X,Y as center of mass. That was a value Jeff White was interested in.

dlebauer, 2016-05-16:
@TinoDornbusch The file mentioned in https://github.com/terraref/reference-data/issues/25#issuecomment-218888032 did not have the sensor box position. [stereoTop/2016-05-10/2016-05-10__14-10-32-010/a3012717-3e46-4f3b-bf3b-8606d457e3e4_metadata.json](https://gist.github.com/dlebauer/8da83f500a6003dcf736e1890f3d6131#file-a3012717-3e46-4f3b-bf3b-8606d457e3e4_metadata-json). But now I see that is because the data was in 'testing' mode and the metadata was `"Only small set of meta data available": "measurement was done during sensor setup"`. 

I can see in the file [stereoTop/2016-05-13/2016-05-13__11-59-18-631/335fa0c7-d549-43d1-96c7-7d4f40d2907e_metadata.json](https://gist.github.com/dlebauer/8da83f500a6003dcf736e1890f3d6131) that it does have the required information. 

The key bit of information was:

> Field of view means a box with dimension in X and Y given with the camera X,Y as center of mass. 

But what is 'center of mass'? Does that have anything to do with where the sensor actually is, or is it just the center of mass of the sensor itself?

TinoDornbusch, 2016-05-17:
@dlebauer  yes my apologies I/Bob did an XML spelling mistake and some user-defined metadata was not written.

The center of the bounding box is the camera position.

dlebauer, 2016-07-11:
closing this issue.  existing components of the meta-data format for raw data from the scanalyzer should not be changed. Additional fields can be added. If changes to existing fields are necessary, these can be described as a github issue and scheduled for implementation prior to 2017 field season

closed 2016-07-11 by dlebauer