terraref/reference-data #29: Fix invalid environmental_logger files

dlebauer, 2016-05-05:
Need to re-process existing environmental logger files per terraref/reference-data#26

* [ ] rename to fix typo in name
* [ ] convert to valid json

something like:

```sh
for file in /projects/arpae/terraref/raw_data/ua-mac/EnvironmentLogger/*/*_enviromentlogger.json; 
do 
  cp $file $file.backup
  echo "[`cat $file` ]" | sed 's/}{/},{/g' > $file;
  rename enviroment environment $file 
done
```

When everything checks out, we can `rm *.backup`

dlebauer, 2016-05-05:
I've copied files to /gpfs_scratch/terra-rename/ and have submitted a job ([this script](https://gist.github.com/dlebauer/3eb6910b687209486b88f7c12110a370)) to process these files

dlebauer, 2016-05-06:
Here is the code so far:

```sh
mkdir /gpfs_scratch/terra-rename
cp -R /projects/arpae/terraref/raw_data/ua-mac/EnvironmentLogger /gpfs_scratch/terra-rename/
for file in /gpfs_scratch/terra-rename/EnvironmentLogger/*/*_enviromentlogger.json
do    
    cp $file $file.backup
    echo "[$(cat $file)]"  > $file
    sed -i 's/}{/},{/g' $file
    rename enviroment environment $file
done
```

dlebauer, 2016-05-06:
@yanliu-chn when PID 35213 is finished, could you please run the following (requires write access to `raw_data` folder)?

```sh
# move valid files as *.json
rsync -routi /gpfs_scratch/terra-rename/EnvironmentLogger/*/*environmentlogger.json \ 
     /projects/arpae/terraref/raw_data/ua-mac/
```

after everything is okay we can delete the scratch folder

```sh
# delete folder from scratch
rm -rf /gpfs_scratch/terra-rename
```

TinoDornbusch, 2016-05-06:
Thanks for this workaround. We will fix this as soon as Markus is back on track from his sick leave.

yanliu-chn, 2016-05-06:
Done except that I didn't remove the scratch data. Could anyone check if rsync was successful?

dlebauer, 2016-05-12:
@yanliu-chn sorry, I gave you the wrong Rsync code ... 

1. [ ]  delete the previously moved *environmentlogger.json from /projects/arpae/terraref/raw_data/ua-mac/ (`rm /projects/arpae/terraref/raw_data/ua-mac/*.environmentlogger.json`)
2.  [ ] delete the misspelled original files `rm /projects/arpae/terraref/raw_data/ua-mac/EnvironmentLogger/*/*.enviromentlogger.json`
2.  [ ] move files from /gpfs_scratch/terra-rename/EnvironmentLogger/* to /projects/arpae/terraref/raw_data/ua-mac/EnvironmentLogger/

yanliu-chn, 2016-05-12:
ok. i have rerun the rsync. but the `*/*environmentlogger.json` does not work because there are too many files to fit in a linux command. instead, i did the following. Please check to see if it worked as expected. It should work now.

<pre>
for f in /gpfs_scratch/terra-rename/EnvironmentLogger/*; do [ -d $f ] && echo "rsync -routi $f/*environmentlogger.json /projects/arpae/terraref/raw_data/ua-mac/EnvironmentLogger/`basename $f`/" >>/gpfs_scratch/arpae/logs/envjson-rsyn2.sh; done

chmod +x /gpfs_scratch/arpae/logs/envjson-rsyn2.sh

# on cg-hm02, do
/gpfs_scratch/arpae/logs/envjson-rsyn2.sh

</pre>

rachelshekar, 2016-09-23:
@yanliu-chn - can this be closed?

yanliu-chn, 2016-09-28:
if @dlebauer verified the results, it can be closed.

closed 2016-09-28 by dlebauer