terraref/reference-data #31: Vocabularies / ontologies to support, and framework for linking synonyms

dlebauer, 2016-05-24:
Vocabularies to support and framework for relating synonyms across vocabularies, enable to search datasets with translation across vocabularies

Vocabularies to support:

* CGIAR sorghum ontology:
 * http://www.cropontology.org/ontology/CO_324/Sorghum%20ontology
 * http://www.cropontology.org/ontology/CO_715/Crop%20Research
* CF Standard name XML table http://cfconventions.org/Data/cf-standard-names/33/src/cf-standard-name-table.xml
* BETYdb variables https://www.betydb.org/variables.json
* ICASA Master Variable list

Related issues:

* terraref/reference-data#18
* terraref/reference-data#14

@lmarini

craig-willis, 2016-07-14:
As per discussion with @dlebauer, for this task I'll focus on a system to support vocabulary mapping, initially focusing on the listed vocabularies/ontologies. I'll review existing solutions, particularly the EC Geosemantics server, and flesh out requirements. The basic requirement here appears to be a software-based solution and process for mapping these vocabularies where the resulting mapping can be leveraged by existing systems (e.g., BETYdb, cross-database search).

craig-willis, 2016-08-03:
Spoke with IN-CORE project team at NCSA -- they have evaluated a number of vocabulary servers and seem to have closely aligned needs.  Will try to setup a meeting with ECGS and IN-CORE team in the coming week or so to share information and requirements.

craig-willis, 2016-08-11:
I've been reviewing several services considered by the IN-CORE team for vocabulary management -- iQvoc, TemaTres, LOV, and Skosmos. I've also reached out to the Marine Metadata Interoperability (MMI) project about their ontology registry (http://mmisw.org/orr/). The MMI ORR was selected by the Earth Science Information Partners (ESIP) federation.  While MMI ORR is ontology-centric, it is designed specifically for the development of cross vocabulary/ontology mappings.

craig-willis, 2016-08-15:
@JeffWhiteAZ @chporter @dlebauer 

We discussed one goal of this task as mapping the TERRAref output data and BETYdb to the ICASA vocabulary, possibly hosting a mapping service. Most of the services I've explored so far  (ECGS-KIS, MMI ORR, etc) rely on vocabularies or ontologies being represented in standard formats such as OWL or SKOS to take advantage of semantic mapping operations such as owl:sameAs, skos:exactMatch, etc. 

From my understanding, the ICASA master variable list is published as a Google spreadsheet. Do you know if there's been any discussion of publishing or rendering this in a more linked-data suitable format? If not, do you see any problems with us doing so for TERRA -- for example, writing an interface to the spreadsheet that publishes the variables via permanent URLs.

JeffWhiteAZ, 2016-08-15:
Craig,
We have had several discussions about "ontologizing" the ICASA master variable list and as a first step, have considered publishing in a more readily viewed and searched format. That work hasn't moved ahead due to lack of funding. I expect Cheryl would support me in saying "yes" to any effort by TERRA to help us move in that direction. Thanks for the offer.
What sort of effort would this require? What approach would you suggest beyond what you outline above?

craig-willis, 2016-08-15:
Thank you, Jeff. Now that I know you're open to it, I'll put together a few options for discussion this week.

nfahlgren, 2016-08-23:
@craig-willis @dlebauer I saw the note about a standard/interchange format for PlantCV, do we want to have that discussion here or a separate issue?

craig-willis, 2016-08-24:
@nfahlgren I think #18 is the right forum for this.  I've added a Google Doc with a rough sketch of my thinking and welcome your feedback.

craig-willis, 2016-08-24:
@JeffWhiteAZ @chporter @dlebauer 

I've added a (rough) document to the computing pipeline folder with some options for creating a linked-data suitable version of the ICASA model.

https://docs.google.com/document/d/1fTcTpKxcanPfRKPC-ZG8KjufGHscx5OCCB-5Ti58ARg/edit?usp=drive_web

Any feedback welcome.

chporter, 2016-08-24:
@craig-willis 
This is a really nice framing of the issues involved in using the ICASA vocabulary for linked data. 
For what we are doing in AgMIP and DSSAT currently, the flat data dictionary is sufficient, but we understand the need for linked data and we want to support the efforts of groups that need that capability. We don't currently have resources to put a lot of effort into making that transition, but we welcome the opportunity to guide those efforts. Can we set up a call between Jeff, Cheryl, and Craig (and anyone else interested) to discuss where to go next?
For a while we had regular Skype calls with Celine Aubert, Elizabeth Arnaud, Medha Devare and others regarding the mapping of ICASA variables to the Crop Ontology and Agronomy Ontology. We may want to consider restarting that effort, including Craig Willis in the conversation.

craig-willis, 2016-08-25:
Thank you, @chporter.  I'll bring this up in our weekly meeting tomorrow and report back.

craig-willis, 2016-08-26:
@chporter @JeffWhiteAZ 
We discussed this at the meeting yesterday and do want to move forward with the effort to put together a RDF or similar version of the ICASA data dictionary and model -- the task will be assigned to me.  I'll setup a call to discuss. Thanks for your feedback.

chporter, 2016-08-26:
@craig-willis @JeffWhiteAZ 
Glad to hear it, Craig.

craig-willis, 2016-09-23:
Why is this closed? The related work is incomplete.

max-zilla, 2017-03-08:
https://docs.google.com/spreadsheets/d/1FKSLTTIywQygRDv3nU6Xk6Z9D0xKxnoR_Mw1a8Kbjj0/edit?usp=sharing

dlebauer, 2017-05-03:
Should we turn this into an epic, close, or ?

There are many subtasks associated with this.

craig-willis, 2017-06-08:
Late response, but this should probably be an epic.

dlebauer, 2017-06-22:
@craig-willis could you please update the milestone on this? turn it into an epic if necessary?

closed 2020-02-13 by dlebauer