terraref/reference-data #33: Upload plot center coordinates to clowder

dlebauer, 2016-06-08:
From @LTBen: The researchers marked the measurement “center” of each range (plot).  Stuart and Tino collected the gantry coordinates of these positions 

@BuffMyRadius and @TinoDornbusch could you please send these coordinates to me or upload them to google drive?

TinoDornbusch, 2016-06-09:
It is uploaded to Google Docs.

BuffMyRadius, 2016-06-09:
[https://docs.google.com/spreadsheets/d/1eQSeVMPfrWS9Li4XlJf3qs2F8txmddbwZhjOfMGAvt8/edit#gid=883764630](url)

dlebauer, 2016-06-09:
The goal here is to develop a map of the field that includes the bounding boxes of each plot (as shown in [this field map](https://docs.google.com/spreadsheets/d/1QQaWc0UaQQKfEtnSO1G2za8tKU2huC0_VYMBqm5CKAo/edit#gid=1066629008).

This worksheet provides the northern and southern boundaries and the centerline for each range. I believe that we also need the eastern and western boundaries and centerline for each column. Will it be possible for you to provide these as well?

Can we assume that each plot is aligned with the gantry x,y coordinate system?

If you can give the coordinates of the field corners, and if the assumption that rows are evenly spaced, it should be possible to generate these plot boundaries and estimate their centers.

What approach do you suggest?

BuffMyRadius, 2016-06-10:
It would be hard for me to define east-west boundaries of the plots. Plants will grow into each other over time; its entirely plausible that a plot of smaller plants could get completely out-competed by neighboring plots and essentially cease to exist within just the next few weeks. The x-axis division of plots won't have this issue because the isles between ranges will be trimmed and maintained.

I think that I should measure the y-axis center of every plot, but I'm not willing to assign any dimensions to them yet. I'll take measurements at the north and south end of the field tomorrow to check if the rows are indeed planted parallel to the gantry coordinates and get back to you.

smarshall-bmr, 2016-06-10:
I took a few y-axis measurements today; its looking like the plots are not 100% square with the gantry coordinates. Storms are popping up some I'm parked for the time being. How high would you say priority is on this?

TinoDornbusch, 2016-06-12:
@smarshall-bmr can you specify how much plots are not square? Note that this is a field experiment. 100% square and resolution is an illusion. The plots should be roughly 4m long. To me the idea is to cut out a region of interest roughly 3m long (in X) and 0.4m wide (in Y) leaving out edge effects in X- and Y- direction. In this area, one should find predominantly plant structures of the genotype of interest for statistical analysis. To me it need to be sure that this area is covered by the scan.

dlebauer, 2016-06-13:
@smarshall-bmr 

To further clarify: I am looking for the boundaries of each plot based on the experimental design; this measurement will be **_independent_** of how the plants end up growing.

Any division according to how the plants actually grow will be handled separately. As @TinoDornbusch points out, we will also provide a smaller bounding box within each plot that would be of more interest to researchers. 

So to be clear, I am not asking for a bounding box that will enclose all of the plant material of a particular genotype or exclude plants from neighboring plots. That is a separate problem. I am only asking for the experimental layout of the plots.

smarshall-bmr, 2016-06-13:
Ah! Now that I can do! I'll try and get help doing this tomorrow then.

dlebauer, 2016-06-16:
@smarshall-bmr any updates? Have you been able to measure the column coordinates here, and what about the position of the [0,0] for standard coordinates starting in the SW corner?

smarshall-bmr, 2016-06-17:
With Tino's help I was able to check the column coordinates under control from the PLC. The skew that I thought was present turned out to not be there. I suspect that the first time I was confirming measurements under manual control that something went amiss with the gantry's movement. We have measured column 4 =  22.04m and column 32 = 0.632m. Columns 1-3 aren't accessible to the zero point on the gantry but we are able to extrapolate where they are. I'll update the google doc soon.

dlebauer, 2016-06-17:
@smarshall-bmr where on column 4 is this? Did you make a permanent mark for this point? Shall we set this as [0,0,0]? what is the height of this point relative to the gantry coordinate system?

smarshall-bmr, 2016-06-18:
These values were measured at 3.8, 101.75, and 207.3. We drove the gantry from 101.75 to 3.8 on PLC control and the column runs true to a tolerance that's certainly smaller than I am capable of measuring.

I didn't measure the z-axis coordinates of this point or make any permanent mark at it's location.

If we are to set a point of origin at the south end of the field I would suggest [3.8,0,-0.5] because this is the minimum point accessible by the PLC without overrides (except in the z-axis), isn't coexistant with a planting row, and we could set a permanent marker there with relatively little risk of someone tripping over it. This does carry the disadvantage that the origin point will fall outside the gantry's regular travel when it gets cold out due to thermal expansion.

I would still lobby for an origin that coincides with the anchor point of the rail on the x-axis. The "scar" in the paint at the floating anchors at the end of the rail indicates a +-4.6cm error from thermal expansion at the north end of the field and +-4.1cm at the south end.

dlebauer, 2016-06-18:
I think I agree - but your last two paragraphs seem to contradict each
other. Could you draw a quick diagram?

smarshall-bmr, 2016-06-19:
The x-axis rail anchor point is in the middle of the field, somewhere around 102M. What I was trying to say is that we can set the origin at the south end of the field at the minimum (x,y) for the gantry system but it might save some headache later to set the origin at the midpoint anchor on the x axis where the gantry coordinates and real world field coordinates will be the same, regardless of temperature.

Either way works really, it's just a choice between an unstable origin point that we'll have to correct for later, or the headache of working with a zero point with positive and negative numbers on the axis.

rachelshekar, 2016-08-17:
@smarshall-bmr - has this task been completed?

smarshall-bmr, 2016-08-18:
Range and row coordinates are available here @rachelshekar  https://docs.google.com/spreadsheets/d/1eQSeVMPfrWS9Li4XlJf3qs2F8txmddbwZhjOfMGAvt8/edit#gid=883764630

closed 2016-09-23 by rachelshekar