terraref/reference-data #34: Document sensor calibration protocols

dlebauer, 2016-06-08:
From 2016-06-03 meeting, @LTBen offered to draft sensor calibration protocols, and would do this with help from @BuffMyRadius, @jeffwhiteAZ, @TinoDornbusch

JeffWhiteAZ, 2016-06-08:
From the Tuesday 8 June meeting at ALARC, field calibration of hyperspectral and FLIR are challenging and will require testing different options. We expect to pursue this during June-July and will hold meetings at ALARC to discuss progress.
I'd suggest Ben focus initially on protocols for:
RGB color component of StereoVIS
Stereo component of StereoVIS
3D Laser

JeffWhiteAZ, 2016-06-08:
> @JeffWhiteAZ<https://github.com/JeffWhiteAZ> Definitely there are a lot of issues to address, but I'd like to get a first draft for the hyperspectral sensors (terraref/computing-pipeline#88<https://github.com/terraref/computing-pipeline/issues/88>) that will allow my team to move forward. Once we have test datasets we can get feedback from the team and standards committee. This approach will allow quick iteration and improvement.
>
>
> This electronic message contains information generated by the USDA solely for the intended recipients. Any unauthorized interception of this message or the use or disclosure of the information it contains may violate the law and subject the violator to civil or criminal penalties. If you believe you have received this message in error, please notify the sender and delete the email immediately.

If there is pressure for an initial document, I would just ask Lemnatec (Ben) to provide their basic protocol.

JeffWhiteAZ, 2016-07-22:
Now that  Lemantec has applied the calibration matrix to the 3D scanners, how can we best test how well the system is working? A first step would seem to be scan the aluminum target and determine how well the  measurements with calibrations applied compare to the measured dimensions of the target. 
This could easily be done while the field is clear. It would be interesting to try:
1. Different orientations of the target
2. Different scanning speeds
3. Night vs. day
I would expect to see X,Y,Z dimension accuracy expressed as RMSE or an equivalent indicator.

solmazhajmohammadi, 2016-07-22:
@JeffWhiteAZ I am working on this. Staurt is going to do couple of test scans for me next week. Currently west 3D scanner is working perfectly, but East scanner needs some modification in the calibration file.

rachelshekar, 2016-09-23:
See https://docs.google.com/spreadsheets/d/1xkEHQd_5x3Yzv3f61bfns3ZFoP7xXq8AUNK4fmS3kXg/edit#gid=194274492

closed 2016-09-23 by rachelshekar