terraref/reference-data #35: Quick inventory of data available

JeffWhiteAZ, 2016-06-08:
Just had a query for wheat data. Do we know if any usable data were obtained from the Scanalyzer?
Is Clowder the best place to monitor data availability? I'd guess I'd like a data overview that tells the status of anything in the pipeline - raw, initial QC ... final product

dlebauer, 2016-06-08:
@JeffWhiteAZ wheat data should be available, though if I recall correctly there is a lot of missing information in the meta-data. You can currently search by date in Clowder (e.g. https://terraref.ncsa.illinois.edu/clowder/search?query=2016-03) for data collected before the first week in April. You could also look in Globus ([see documentation](https://terraref.gitbooks.io/terraref-documentation/content/data_access.html)), which has a more familiar GUI for file system navigation. 

@max-zilla will your log monitor provide information about what has been collected?

JeffWhiteAZ, 2016-06-08:
But looking at Clowder, there is no easy way to know:
- if these are just random tests or are useful whole field scans?
- if the system performed as expected?
- what level of processing they received
A search for "wheat" == no results.
How do we clean up metadata from these initial runs?
Possibly, there are no useful Scanalyzer data for 2016 wheat, so maybe not worth pursuing.

JeffWhiteAZ, 2016-06-08:
For the 2016-03 search, the displayed results show two "collections" identified as:
Name:
2016-04-03
Description:
All datasets on this date.

One seems to be VNIR and one is FLIR. Can we improve the information displayed here or again, is this because metadata are incomplete?

JeffWhiteAZ, 2016-06-08:
A search for "2016-04" identifies one collection from 2016-04-04. 
A search for "2016-03" identifies one collection from 2016-03-05 and the two mentioned above from 2016-04-03.
I would have expected multiple datasets between 2016-03-05 and 2016-04-03. Am I misunderstanding the search procedure?

max-zilla, 2016-06-08:
@dlebauer yes, the logs are collecting information about what is being collected. I'm working with Rob to integrate them with LogStash so we get a digest summary - the logs are large and unwieldy right now. Hope to have this ready soon.

@JeffWhiteAZ the metadata and Clowder search capabilities are still being improved.

The duplicate collections are likely an artifact of the development process - we are going to merge and de-duplicate these artifacts in the near future as the pipeline has largely stabilized. We can look into the gantry logs between those time periods but you understand the search procedure - I don't recall very much data from March.

JeffWhiteAZ, 2016-07-07:
Any progress on improving search capabilities? I need RGB and FLIR data from sorghum plots showing leaf desiccation for a talk on 4 August.
Thanks!

JeffWhiteAZ, 2016-07-07:
(sorry about "closing" -- I misunderstood "close" as "exit" -- not close the issue discussion)
Is the a quick guide to the search tool syntax. I just tried "2016-07" expecting to get datasets from July, but the response was no data for 2016-04-07. Help icon is not very informative.

dlebauer, 2016-07-07:
once terraref/computing-pipeline#64 is complete, the data will be converted into a useable format. 

Enhanced search functionality is in progress (terraref/computing-pipeline#46) by @max-zilla and @craig-willis. @JeffWhiteAZ - defining specific use cases there would help ensure we implement the functionality you are looking for.

max-zilla, 2016-07-07:
@JeffWhiteAZ in Clowder there is not currently any data from July - just globus. @robkooper and I needed to refactor several aspects of Clowder to support the volume of data coming in because the Clowder uploads were not keeping up with the globus transfers, and the code fixes are currently being tested before rollout. this update should occur soon and allow us to catch up to current data.

but as @dlebauer mentioned, I am in the midst of working on enhancing search to make the types of searches you describe more easy and user-friendly.

closed 2016-09-23 by rachelshekar