terraref/reference-data #36: Document the parts numbers and specifications for calibration panels / objects?

dlebauer, 2016-06-17:
We need to know the part numbers and specs for the calibration panels we are using. For example, SphereOptics says that for each panel 

> Calibration will be performed on a Perkin Elmer Lambda 19, data will be supplied electronically in 1 nm steps, 50 nm step printed documentation with NIST/PTB traceability with certificate for the range from 250 nm-2500 nm.

We need to collect this documentation. Please attach such documents to this issue.  
 
| object| image | manufacturer | part number | 
|:---|:---|:---|:---|
| white reflectance panel|[image]|Spectralon|?|
| munsel color chart||[image]|||
| 3D object|[image]|| custom |

 * are there any others?

smarshall-bmr, 2016-06-17:
The white reflectance panel is made by SphereOptics. It is uncalibrated and carries only an I.D. number SG3156-U and has a 95% reflectance.

The color chart is manufactured by mennon. It is printed almost exclusively in Japanese, but it does have a key of the RGB values of all the color squares.

The 3D Object came with a set of labsphere Spectralon diffuse color standards Art 050-166. All 8 targets are individually labeled.

In addition we have a smaller 10x10cm SphereOptics uncalibrated target. I.D. SG3156-U and 95% reflectance.

I can enter all this information in a preferred location.

dlebauer, 2016-06-17:
@smarshall-bmr 

What are the labels on the 8 targets. Are these calibrated?

It seems odd that we have calibration objects that aren't themselves calibrated - how will we use these if we don't know what their properties are?

For now, please put the information in a new worksheet here: https://docs.google.com/spreadsheets/d/1xkEHQd_5x3Yzv3f61bfns3ZFoP7xXq8AUNK4fmS3kXg/edit#gid=1085009115

And any documentation / calibration documents in a subfolder.

smarshall-bmr, 2016-06-18:
It's common practice to use uncalibrated targets to quantify differences between runs. Calibrated targets are only used for initial calibration because they are both expensive and fragile. The uncalibrated target is measured on the calibration run and then further runs can be compared using the uncalibrated target without putting the calibrated standard target at risk through daily use.

Both the mennon color chart and 8 Spectralon diffuse color standards are calibrated. I am putting their full info in the google sheet that you linked.

dlebauer, 2016-06-18:
Sounds like in the end we can calibrate our uncalibrated panels with sufficient precision, it just won't be certified by NIST.

smarshall-bmr, 2016-06-19:
Using the calibrated color panels is more than enough to figure out the necessary correction curves for the colors we will be measuring, yes. The white panels that we have are still good for run to run color corrections too.

dlebauer, 2016-09-23:
@craig-willis please make sure to capture this information in the sensor metadata

smarshall-bmr, 2016-10-31:
I have uploaded the calibration certificates and spectral response csv files for the 8 diffuse color targets in Drive under TERRA-REF>Maricopa Expirements>field scanner operations>Sensor datasheets

craig-willis, 2016-10-31:
Thank you,  @smarshall-bmr. 

For the record,  my plan is to organize the calibration target information in the same way as the sensor/device information in the Clowder Devices and Sensors collection:

https://terraref.ncsa.illinois.edu/clowder/collection/58035fa34f0c4a438cbb53dc

Each calibration object will be represented as a dataset with any associated images and documentation attached as files. Values that are needed programmatically -- such as the spectral response data -- will also be added directly to the dataset metadata.

craig-willis, 2016-10-31:
@dlebauer 

What is the origin of https://github.com/terraref/reference-data/blob/master/Zenith%20LiteT%20Diffuse%20Reflectance%20Target%20-%2095%25R.csv?

I see "Zenith Polymer" (not LiteT) mentioned in the LemnaTec  sensor documentation for testing the VNIR. We have a calibration certificate for the Zenith Polymer (SG3051), but these don't match what was mentioned in the June comments above.

dlebauer, 2016-10-31:
@craig-willis I am not sure - maybe @TinoDornbusch or @smarshall-bmr can provide some insight?

rachelshekar, 2016-11-30:
@craig-willis  - can this be closed?

craig-willis, 2016-11-30:
Yes. I think anything related should be opened as a new issue.

closed 2016-11-30 by craig-willis