terraref/reference-data #37: Add level to test object

dlebauer, 2016-06-17:
For consistent interpretation of calibration panels, Larry Biehl suggests making sure that the panel is level. 

Would it be feasible to attach a circular level [similar to this one](http://www.stahl-libellen.com/Circular_spirit_levels/Aluminium_transparent_28.html) to the test objects?

[image]

smarshall-bmr, 2016-06-17:
Before each scan I have been checking the test object for levelness with a bubble level. A circular level would certainly make my job much easier though, as long as it can put up with the hot sun here. The 3D object gets scaldingly hot out in the sun!

dlebauer, 2016-09-23:
@smarshall-bmr should this issue be closed? Can you get a circular level and attach it to the object(s)? Would be great to have these in the images.

rachelshekar, 2016-11-30:
@smarshall-bmr ???

rachelshekar, 2017-01-12:
@nshakoor - please talk to Stewart

smarshall-bmr, 2017-01-12:
@dlebauer I can get a few bubble levels. Should we actually affix them to the calibration panels though?

dlebauer, 2017-01-12:
Your choice- I think key is ability to put in the same place

smarshall-bmr, 2017-01-18:
@dlebauer Would it be beneficial to have a level large enough that it shows up in imaging?

dlebauer, 2017-01-19:
@smarshall-bmr I suspect anything larger than 2 cm will be sufficient if the imagers have mm resolution, correct?

closed 2018-12-13 by dlebauer