terraref/reference-data #39: Mislabeled aperture in metadata

smarshall-bmr, 2016-06-23:
While the VNIR is uninstalled I have taken the opportunity to try out different aperture settings on the StereoVIS system while the RGB cameras are accessible. I have just discovered that the cameras have been physically set at f/16 while the metadata has been reporting them as set at f/4.

I have taken the liberty of changing the fixed metadata file so it will write correct values moving forward. This does mean that all past metadata files should be edited to be correct.

Our plans for settings on the cameras in the future are now much less limited. My initial "by eyeball" look at the output of the cameras, even at f/8 with gain and exposure both significantly reduced, is that image quality will be much improved. If the cameras were indeed intended to be set at f/4 and the DoF is not an issue I anticipate no further issue with noise from high gain, as well as greatly reduced if not eliminated motion blur.

dlebauer, 2016-06-23:
@smarshall-bmr 

To be clear do you mean "for all SteroVIS metadata files collected before June 23 change aperature from f/16 to f/4"?

dlebauer, 2016-06-23:
@pless could you please comment on the appropriate settings?

smarshall-bmr, 2016-06-23:
@dlebauer I'm assuming that the settings on the cameras have never been changed since installation. I would have to confirm that.

@pless I'm going to do some optimizing to get us "in the ballpark" of the best settings. I'll keep you updated.

rjstrand, 2016-06-23:
@smarshall-bmr - To my knowledge, the settings have not been changed since they were set by Hans Georg Luigs back in January/early February.  There only a few that would have made changes and I will contact them to verify.

smarshall-bmr, 2016-06-23:
The only thing that concerns me is that the f/4 that was being written to metadata had to come from somewhere. Either the setting was just misread/misquoted at some point, or it was actually set at f/4 at some point in the past but the metadata wasn't updated when the cameras were changed to f/16.

At any rate it's looking like the final setting will be somewhere between f/8 and f/5.6. This puts gain in the 1500-1800 range and exposure between 8000 and 11000 ms. Under f/5.6 chromatic aberration starts to become apparent, but if we really want to cut down on motion blur we might consider cranking the aperture as much as we can get away with. This is a drastic improvement on the gain of 2700 and exposure of 27000 ms at f/16 already.

smarshall-bmr, 2016-06-27:
@dlebauer @rjstrand @pless @TinoDornbusch I have done a set of motion blur tests and haven't had any issues, even at 0.33m/s (full speed). I'm about to crank up the scan speed on the VIS/IR scan unless someone has concerns about me doing so. My primary concern is that the FLIR won't keep up.

pless, 2016-06-27:
Awesome!!

Robert

JeffWhiteAZ, 2016-06-27:
Good to hear.
What metrics were used to quantify "haven't had any issues"?
Keep in mind that the scanner is to produce "reference datasets." End-users will expect to know about tradeoffs among scan speed, pixel blur, depth of field, and gain, all as affected by illumination.

smarshall-bmr, 2016-06-27:
@JeffWhiteAZ @pless  I did a pixel luminescence decay test in FIJI. All of the decay curves are virtually identical (within 5%) to a static image. I was slightly baffled by this and actually manually triggered a measurement while running 1m/s on the x-axis and found a 27% increase in decay distance. I suspect that the set of images that I took are not exactly at the focal point but I did a lot of circle of confusion testing on Saturday and I know that the distance that I'm testing at (1.8m from target) is within the depth of field. I would expect a 9% increase in intensity decay distance at 0.33m/s, but I might be getting just enough "help" from a softer focus at short distance that the change in decay at 0.33m/s gets buried. At any rate the pixel decay is happening over the real world equivalent of <0.07mm (this is a "by eye" estimate, I was using the uncalibrated checkerboard so I had a big target to hit.)

In the future I will do a more formal test but at some point we have to admit that there is no quantitative measure of how good an image is. I am very aware that we have picked up some issues with depth of field, chromatic aberration, and possibly even vignetteing, which are all quantifiable, in order to gain performance in the wind-blown image blur on the plants which, short of painting a calibration grid on a plant and measuring its 5-axis oscillation, is a subjective quality.

closed 2016-09-23 by rachelshekar