terraref/reference-data #40: Camera Position in metadata

solmazhajmohammadi, 2016-06-28:
Recorded camera location in camera box for some of the sensors (including SWIR) is not correct. Stuart is going to fix it.

TinoDornbusch, 2016-06-29:
@smarshall-bmr  I thought that positions have been recorded correctly in Aachen. Well always doublecheck. 
How do you want to measure? I suggest to plumb down the center of view of each camera and measure relative to the plumbed down position of center box (0,0).

solmazhajmohammadi, 2016-06-29:
@TinoDornbusch It might be just a typo, and reporting VNIR position instead of SWIR.

smarshall-bmr, 2016-06-29:
@solmazhajmohammadi @TinoDornbusch I'll have a look this afternoon if the weather doesn't go downhill. Predictions aren't looking optimistic right now.

solmazhajmohammadi, 2016-07-14:
@smarshall-bmr can you please update the issue... From which date you have applied the changes in metadada? Which one of the camera's location has been changed?

dlebauer, 2016-07-14:
@smarshall-bmr I also created #42 which should make it easier to update the fixed sensor meta-data

smarshall-bmr, 2016-07-14:
@solmazhajmohammadi @dlebauer would you prefer that the z-axis be the distance to the camera housing, distance to the lens, or distance to the actual detector? In the case of the SWIR there is a mirror that redirects the path of light so the effective imaging distance is quite a bit further away then the vis cameras even though their housings are relatively close in the z-axis.

dlebauer, 2016-07-14:
The detector seems like the logical location but I will defer to Solmaz, Robert @pless, and others

solmazhajmohammadi, 2016-07-18:
Distance to the detector and lens, both are logical. The principal point usually is inside the lens glass, distance to the lens might be a better option.

dlebauer, 2016-07-18:
What is the "principal point"?

solmazhajmohammadi, 2016-07-18:
It is the point that the focal length is measured from it.

closed 2016-09-23 by rachelshekar