terraref/reference-data #41: Over-exposed Hyperspec data

solmazhajmohammadi, 2016-06-30:
@czender Can you please take a look at the SWIR data from the following dates:
"time": "06/08/2016 10:57:09"
"Time": "05/06/2016 10:51:25"
It seems that  all the bands are overexposed. 
@TinoDornbusch  any changes have been made on camera setting since last month?

smarshall-bmr, 2016-06-30:
It's worth noting that the preview jpeg images still look the same. That should narrow down possible causes.

czender, 2016-07-07:
Yes the jpeg images look good from the June image. I thought jpegs were created from three bands (RGB) of the hyperspectral image, meaning at least those bands should be good. Am I wrong about that?

smarshall-bmr, 2016-07-07:
The three band image from RGB wasn't working, but the three band from HSI was. I just tried opening a newer image and it worked though. Have settings been changed? @solmazhajmohammadi could you check that hypercube is working on your end now as well?

solmazhajmohammadi, 2016-07-07:
I don’t see any data from SWIR  after 6/30 on globus. @smarshall-bmr  Did you do the scan recently?

smarshall-bmr, 2016-07-07:
6/30 was the last SWIR scan run. The 4M calibration scan hasn't been run since then because the plants are too tall to set up the calibration targets.

closed 2016-09-23 by rachelshekar