terraref/reference-data #42: Create a metadata file for fixed sensor / field scanner data

dlebauer, 2016-07-14:
### Description

I propose that we keep a separate file containing 'fixed sensor metadata'. It should have the same fields / schema that are currently in the *_metadata.json files as well as additional information.

We should not change the current _metadata.json files that are generated, just have a new canonical source for the information. 

### Context

There have been a number of times when the 'fixed' information has been changed or updated. This will make it easier to update meta-data without having to rewrite the original metadata files that had incorrect information. 

Here are some of the issues that such a static (or infrequently updated) file would help:

* incorrect camera location #40  
* mislabeled aperature in meta-data #39   
* changes to metadata #25 
* incorrect field of view https://github.com/terraref/computing-pipeline/issues/126

### Further Suggestions / Request for Feedback   

1. What additional fields should it contain? 
 * date range for data that it applies to
 * who is responsible for writing it
 * location of sensor calibration files
 * others?
2. How should it be stored? 
 * start with .json files named sensor_metadata.json
 * put in the gantry data stream near calibration files
 * made accessible for programmatic access via an API (clowder?)

craig-willis, 2016-11-02:
@dlebauer 
As noted in [#54](https://github.com/terraref/reference-data/issues/54), I've created the [sensor-metadata](https://github.com/terraref/sensor-metadata/) repository and attached the current LemnaTec fixed_sensor_metatata as JSON-LD to each sensor dataset in Clowder.   I think this fulfills most of the requirements outlined above, except for the addition of any proposed new fields (the current version does include the RSR data, where available):

In short:
* Metadata and format can be updated and versioned in Github by any team member, using standard PR workflow for review/approval
* Metadata can be attached to sensor and calibration target "datasets" in Clowder and accessible via API
* It's unclear to me whether we need this also in the Gantry data stream.

JeffWhiteAZ, 2016-11-02:
If I click on the sensor-metadaa hot-link, I get a 404 error. Is this a broken link?
- Jeff

This electronic message contains information generated by the USDA solely for the intended recipients. Any unauthorized interception of this message or the use or disclosure of the information it contains may violate the law and subject the violator to civil or criminal penalties. If you believe you have received this message in error, please notify the sender and delete the email immediately.

dlebauer, 2016-11-03:
Jeff - which link?

craig-willis, 2016-11-03:
Sorry, resolved this via separate email thread.  The Github repository was created private by default, I've changed to public.  This caused a 404 on the repo link.  The sensor datasets in Clowder were also by default private (inherited from the space), which I've changed to public as well.

JeffWhiteAZ, 2016-11-03:
Craig resolved a couple of permission issues, so the problem no longer exists. The link was:
https://github.com/terraref/sensor-metadata/
We also found an additional permission as you drill down to metadata for individual sensors—also fixed.

craig-willis, 2016-11-07:
@dlebauer @TinoDornbusch 
Is there someplace I can find definitions for all of the fields provided by LemnaTec in the metadata.json files?

dlebauer, 2016-11-07:
I've never seen this ...

rachelshekar, 2017-01-12:
@dlebauer - please update

smarshall-bmr, 2017-01-12:
@craig-willis I don't know if there's an index of the fields that are provided anywhere, but I should be able to help clarify what the meanings are.

@dlebauer Just to clarify, you don't want me to edit the existing metadata writing? It's relatively straightforward for me to do, but if it causes issues downstream I can certainly refrain from doing so. Perhaps I can add "last edited by/date" fields to the fixed metadata?

dlebauer, 2017-01-13:
@smarshall-bmr there are two issues here. 

1. Please never change the names of fields in the sensor metadata.
  * Have you done so since Aug 2016? If so, there must have been a miscommunication. We have been over this many times, perhaps a few before you joined. But it breaks our workflow 
3. You can add fields but not change them
4. If you are going to change the value of a field in the fixed metadata, it should be done in one place. Perhaps you should be in charge of storing this file, and we can use it in our workflows. 
  * Please discuss with @craig-willis about the centralization of the fixed metadata. @craig-willis is there an issue for centralizing the fixed metadata?

smarshall-bmr, 2017-01-13:
@dlebauer I haven't changed the names of fields, but I know that some of the software updates here did. The only modifications of values have been on the stereovis to correct the aperture and the fixed metadata on a new crop. The VNIR will have some additional fields as of this week that weren't being written previously.

I do think it would make sense for me to store the fixed metadata here so it's written straight from collection. @craig-willis would this work for you?

dlebauer, 2017-01-13:
@craig-willis and @smarshall-bmr Sorry for not being clear ... I wasn't suggesting that you store the file there, only that you be in charge of the files. They should be stored online. Or perhaps they could be stored in the gantry filesystem at a higher level than within the metadata for each dataset. Then it could be parsed and put into clowder and accessible to the extractors.

For the record, we discussed doing this earlier but @markus-radermacher-lemnatec suggested the fixed metadata be in each atomic dataset. Perhaps we can do both, have the current metadata values and then let the scanner write it into each dataset. Our pipeline will use the single instance of the current metadata values, but the raw data will also have a copy.

Could you please open a separate issue that describes the changes to the existing field names that have occurred? The pipeline should handle new fields and modified values. If and only if changes are necessary, developers working downstream should be given ample warning as we agreed in https://github.com/terraref/reference-data/issues/25#issuecomment-231854238 as well as every other time changes have broken the downstream workflow (e.g. https://github.com/terraref/computing-pipeline/issues/116 and other issues linked above).

craig-willis, 2017-01-13:
@smarshall-bmr 

It sounds like we should talk. @rachelshekar is going to schedule a meeting about sensor metadata. 

From my understanding, there are a few reasons for centralizing the fixed metadata:
* To have greater control over errors like the ones listed in the description of this ticket, as well as adding new properties that are useful to downstream processes.
* To enforce a process to review changes (i.e., via git pull requests) to catch potential problems/errors.
* To support standardization, since we will have sensor metadata for devices that don't conform to the Lemnatec-defined format.

Of course, this is only for the fixed metadata, not any of the variable metadata.  We've recently encountered a problem (reported in https://github.com/terraref/computing-pipeline/issues/230) apparently caused by  changes to the gantry_system_variable_metadata.

davidlee2, 2017-01-16:
Hi David,

Just wanted to check how things were going and if there was anything I can help with.

Best,

David

Sent from my mobile, sorry for the short response.

Could you please open a separate issue that describes the changes to the existing field names that have occurred? The pipeline should handle new fields and modified values. If and only if changes are necessary, developers working downstream should be given ample warning as we agreed in #25 (comment)<https://github.com/terraref/reference-data/issues/25#issuecomment-231854238> as well as every other time changes have broken the downstream workflow (e.g. terraref/computing-pipeline#116<https://github.com/terraref/computing-pipeline/issues/116> and other issues linked above).

-

dlebauer, 2017-05-03:
@smarshall-bmr can this be closed?

closed 2017-05-03 by smarshall-bmr