terraref/reference-data #43: Create a list or catalog of data products

craig-willis, 2016-08-03:
Creating a list or catalog of existing and planned data products for the TERRA project -- more than just the individual sensor metadata. It might be worth compiling a catalog similar to the NEON project (http://www.neonscience.org/sites/default/files/basic-page-files/NEON_DOC_002652.pdf).   Based on the NEON data catalog, possible attributes of a data product description (e.g., catalog record) are listed in the "Data product catalog" section of the following Google Doc:

https://docs.google.com/document/d/13gXD_OVLffm0hqahDZ3tUvru8IV1fRfM6DiuOcfjr3s/edit?usp=sharing

The resulting list should be put in the documentation repository.

max-zilla, 2016-09-21:
@craig-willis some pointers where documentation exists for the extractors...
- https://github.com/terraref/extractors-stereo-rgb
- https://github.com/terraref/extractors-lemnatec-indoor (PlantCV)
- https://github.com/danforthcenter/plantcv 
- https://github.com/terraref/extractors-environmental
- https://github.com/terraref/computing-pipeline/tree/master/scripts/hyperspectral

I've fleshed out some of these somewhat and point you upstream for others, but this is probably where some of the best 'clean' descriptions of these extractors will go. Like you said earlier, we should make sure the info in the github repo matches whatever is in our docs if we can't just point there directly.

Reminder to both of us also that we discussed using extractor_info.json file in each extractor directory to stash info that we could later reference as canonical source for some properties.

craig-willis, 2016-09-27:
@dlebauer 

I've been working on a proof-of-concept that will allow us to store the data product information in JSON and generate a Data Products page via Github Pages/Jekyll -- drawing related content from the Clowder API.

You can see an example here: https://craig-willis.github.io/

The data product descriptions are in JSON:
https://github.com/craig-willis/craig-willis.github.io/blob/master/data_products.json

The sensor name is retrieved from Clowder (so we only have to store it once):
https://terraref.ncsa.illinois.edu/clowder-dev/datasets/57ea88dee4b00b25cabf8e9b

Feedback welcome.

dlebauer, 2016-09-27:
@craig-willis this is a nice start. We should review this on Thursday with the group.

craig-willis, 2016-10-16:
@dlebauer @rachelshekar 
Created a pull-request for version1 documentation with a rough draft of a similar data product table, linking to subpages with detailed descriptions of each data product.  Feedback welcome.
 
https://github.com/terraref/documentation/pull/80

craig-willis, 2016-11-15:
Closing this, as I think it's fulfilled by 
https://terraref.gitbooks.io/terraref-documentation/content/user/data-products.html

closed 2016-11-15 by craig-willis