terraref/reference-data #46: Upload 2016 first season field measurements to BETYdb

dlebauer, 2016-08-09:
TERRA-Ref Maricopa Sorghum 2016 first season summary:

* Planting date:  April 19
* Harvest dates:  July 11-14
* Crop duration:  82 days post planting

Summary of manually-collected field measurements (ground truth data collection):

* [ ] Emergence seedling vigor, BAP:  7 dates
* [ ] Emergence final stand counts, all plots before thinning:  1 date
* [ ] Plant height, all plots:  5 dates  (separate measurements for canopy and spike heights)
* [ ] Node counts on 180 marked plants, 18 plots in BAP:  9 dates
* [ ] Tiller counts on 180 marked plants, 18 plots in BAP:  6 dates
* [ ] Phenology growth stage notes, all plots:  10 dates
* [ ] Leaf desiccation ratings, all plots:  3 dates 
* [ ] Lodging and architectural leaning, all plots:  1 date
* [ ] Biomass samples in-season, 9 plots:  2 dates (includes leaf areas and moisture samples)
* [ ] Radiation interception, BAP:  3 dates
* [ ] Porometer readings, 24 plots in BAP:  2 dates
* [ ] Harvest measurements, 597 plots plus 48 plots by row (645 plots, 694 subsamples):  
  * [ ] yield estimates 
  * [ ] moisture samples 
  * [ ] plant counts and row lengths for harvested sections of plots

* Raw data are in google drive (to be archived in Clowder)
* Code to process these data and prepare metadata are in the [`/scripts`](https://github.com/terraref/reference-data/tree/master/scripts) directory.

rachelshekar, 2016-09-23:
@dlebauer - is this a priority for the V0 release?

dlebauer, 2016-09-23:
@rachelshekar yes!

rachelshekar, 2017-01-25:
associated with #61

dlebauer, 2017-02-21:
@NewcombMaria what is the status of this?

dlebauer, 2017-05-03:
@NewcombMaria status?

NewcombMaria, 2017-05-04:
Trying to find time, and will make it high priority to make milestone by end of May 2017. Thanks

rachelshekar, 2017-06-01:
redundant. see #61

closed 2017-06-01 by rachelshekar