terraref/reference-data #47: Add geospatial metadata to hyperspectral metadata

dlebauer, 2016-08-18:
Currently the files only contain the start position. We don't have the width of the line and we don't have the location of each line. 

Presumably the gantry knows the x location where each line is triggered.

1. We need the bounding box for the field of view
2. We also need to know the x position of each line

@solmazhajmohammadi and @FlyingWithJerome please update / clarify / revise this issue

rachelshekar, 2016-09-23:
@dlebauer - can you assign this to someone?

solmazhajmohammadi, 2016-09-28:
This could be combined with [Georeferencing hyperspectral data, issue #144](https://github.com/terraref/computing-pipeline/issues/144)

dlebauer, 2016-09-28:
@solmazhajmohammadi closed as duplicate of #144

closed 2016-09-28 by dlebauer