terraref/reference-data #48: Metadata for new Weather Station

dlebauer, 2016-08-18:
* What sensors are on board?
* How are they calibrated?
* What do column names mean?
   * "TOA5","WeatherStation","CR1000","39656","CR1000.Std.29","CPU:F13WeatherStation.CR1","39725","SecData"
"TIMESTAMP","RECORD","BattV","PTemp_C","AirTC","RH","Pyro","PAR_ref","WindDir","WS_ms","Rain_mm_Tot"
"TS","RN","Volts","Deg C","Deg C","%","W/m^2","umol/s/m^2","degrees","meters/second","mm"
"","","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Tot"

dlebauer, 2016-08-19:
(from John Heun at MAC):

This is what the data would look like, if opened with Excel. The top row can be ignored. The second header row contains column information: 

* Timestamp - date and time the measurement was made.
* Record - the record number/measurement number since the current program was started.
* BattV - battery voltage of the station.
* PTemp_C - panel temperature of the data logger. Internal temperature, in degrees Celcius.
* AirTC - Outside air temperature, Celcius. Vaisala HMP60 humidity and temperature probe.
* RH - relative humidity, percent. Inside the same probe with temperature sensor (HMP60).
* Pyro - LI-COR pyranometer, model LI-200. Output in watts/m^-2. 
* PAR_ref - LI-COR quantum sensor, model LI-190. Output in micro mols per second-meter^2.
* WindDir - Gill Instruments WindSonic 2-Axis anemometer. Wind direction (degrees relative to true north).
* WS_ms - same instrument as above, wind speed in meters per second.
* Rain_mm_Tot - Texas Electronics rain bucket, model TE525. Output is counts, 1 millimeter of rain per count.

Custom calibration coefficients are applied at the time of the measurement to convert raw signals into the physical units in the logged record. They were derived during factory calibration and are listed on the sensors' data sheet.

rachelshekar, 2016-09-23:
FYI @ZongyangLi

closed 2016-09-23 by rachelshekar