terraref/reference-data #49: historical weather data

rmgarnett, 2016-08-23:
It would be great to have daily historical weather data (to include t_min, t_max, and precipitation) for Maricopa (or a nearby location) available somewhere.  I will use this to build a weather model to base the regression on.

dlebauer, 2016-08-23:
Here is one for 1979-2010. Is that a long enough time series?

[maricopa_daily_met.csv.txt](https://github.com/terraref/reference-data/files/433424/maricopa_daily_met.csv.txt)

For (mostly my own) reference here is the R code to extract the data

```r
library(PEcAn.data.atmosphere)
library(PEcAn.BIOCRO)
metfile <- nc_open("/projects/arpae/met/narr/all.nc")
lat <- 33.075	
lon <- -111.975
x <- load.cfmet(metfile, lat, lon, start.date = ymd('1979-01-01'), end.date = ymd('2010-12-31'))
y <- data.table(cf2biocro(x))
z <- y[,list(tmin = min(Temp), tmax = max(Temp), precip = sum(precip)), by = 'year,doy']
```

rmgarnett, 2016-08-23:
Perfect for now, thank you!  Closing the issue; will open a new issue if something goes awry.

closed 2016-08-23 by rmgarnett

dlebauer, 2016-08-23:
PS Temperature in C and precipitation in mm / d

dlebauer, 2016-08-23:
PPS citation for these data:

Wei, Y., S. Liu, D.N. Huntzinger, A.M. Michalak, N. Viovy, W.M. Post, C.R. Schwalm, K. Schaefer, A.R. Jacobson, C. Lu, H. Tian, D.M. Ricciuto, R.B. Cook, J. Mao, and X. Shi. 2014. NACP MsTMIP: Global and North American Driver Data for Multi-Model Intercomparison. Data set. Available on-line [http://daac.ornl.gov] from Oak Ridge National Laboratory Distributed Active Archive Center, Oak Ridge, Tennessee, USA. http://dx.doi.org/10.3334/ORNLDAAC/1220