terraref/reference-data #50: SWIR out for service

smarshall-bmr, 2016-09-01:
Friday 8/26 the SWIR was uninstalled because its sensor thermoregulation system is failing to maintain a constant temperature. We are currently waiting for an RMA tag from Headwall before we ship it out for repairs. LemnaTec is heading this process but I'll act as a liaison and keep everyone updated as the RMA process continues. If the RMA on the VNIR is anything to go by then we will be without the SWIR for at least a month.

smarshall-bmr, 2016-10-31:
We reinstalled the SWIR on 10/29. We're still working out some kinks with the frame grabber at the moment though.

smarshall-bmr, 2016-10-31:
Back up and running! Turns out an internal power switch had been flipped off.

smarshall-bmr, 2016-11-04:
I suppose I should reopen this while we are still having issues with y-axis triggered frame grabbing.

rachelshekar, 2016-11-30:
@smarshall-bmr - has the frame grabbing issue been corrected?

TinoDornbusch, 2016-12-02:
@rachelshekar  No the SWIR triggering is not working. Camera will be shipped back for repair again.

rachelshekar, 2016-12-02:
FYI @dlebauer @czender @FlyingWithJerome

smarshall-bmr, 2016-12-14:
The current plan is that the SWIR will be used in it's current state (running in untriggered mode) to capture emergence of the wheat crop before being uninstalled; likely sometime in the second or third week of January is my best guess.

dlebauer, 2016-12-14:
@smarshall-bmr @JeffWhiteAZ 'untriggered mode' may break the hyperspectral workflow that @czender et al developed. How will the data be different? Is this a mode that we want to support in the long term, or more a short term bug that we can handle with some one-off fixes to the code. 

If it is worthwhile to support this in our workflow can you please create a new issue in which we can address these issues, point to some example data so that we can test it out? Will running in untriggered mode allow it to go faster to help with https://github.com/terraref/reference-data/issues/67?

JeffWhiteAZ, 2016-12-14:
I would prefer to see the SWIR shipped as soon as possible. It is more useful for later stages of crop development, so with luck, we can get it back in time to use on the wheat. It would be especially useful just before harvest in order to look at moisture, cellulose and lignin.
On the un-triggered mode, my understanding is that we still have exact data on the X (north-south) position, but that the Y-location has to be inferred indirectly and may be inaccurate. That is what is pending to verify. Others should clarify.

smarshall-bmr, 2017-01-18:
SWIR has arrived at Headwall. Updates to follow.

smarshall-bmr, 2017-04-05:
SWIR is reinstalled and functioning as of 3/31.

closed 2017-04-05 by smarshall-bmr