terraref/reference-data #52: Halogen direct illumination spectrum

dlebauer, 2016-09-21:
Please provide a per-wavelength analysis of the light produced by the halogen lights in the sensor box. Columns would be wavelength and flux. 

Something like this (from http://www.cs.virginia.edu/~cab6fh/research/meetings/6_14_11/wikidown_6_13_11.html):

[image]

Presumably this should extend across the range of the VNIR and SWIR (~300-2500nm)

TinoDornbusch, 2016-09-26:
@dlebauer Sorry I just saw this today. I add it to my ToDo list and send it asap.

TinoDornbusch, 2016-09-28:
@dlebauer  We have no measured spectrum for the build-in halogen lamps Showtec 240V/75W. This information is also not available from the supplier.

We have a similar type measured Philips Twistline Halogen 230V 50W 18072. attached the spectrum.
[MeasurementPhilipsHalogenSpot.xlsx](https://github.com/terraref/reference-data/files/498197/MeasurementPhilipsHalogenSpot.xlsx)

closed 2016-11-30 by rachelshekar