terraref/reference-data #53: Add RSR curve data for hyper / multispectral imaging

dlebauer, 2016-09-21:
Relative Spectral Response curves for all of the spectrometers

First column in each table should be the wavelength. Then, for each detector there should be a column (e.g. three cols for NDVI, four for RGB). Presumably there is one for  
Please provide table for the following:

* [ ] VNIR
* [ ] SWIR
* [ ] Skye NDVI pointing up
* [ ] Skye NDVI pointing down
* [ ] Skye PRI 
* [ ] Crop Circle
* [ ] Quantum PAR sensor

Here are some examples from the documentation provided 

### SKYE NDVI

[image]
[image]

### SKYE PRI

[image]

### Ocean Optics

[image]

TinoDornbusch, 2016-09-26:
@dlebauer Do you want to get the response curves as numeric values? Please check with @czender on the requirement of spectral sensitivity of Hyperspectral cameras. To my understanding the spectral sensitivity of the chip is elimated by referencing to the Spectralon White target.

TinoDornbusch, 2016-09-26:
@dlebauer @czender  Here is what I got from Labsphere regarding the Spectralon Target:

The target is not specifically calibrated. It is part of a production chain. Attached the relevant documents.

czender, 2016-09-27:
Please provide a specific hyperlink to the documentation you refer to. The calibration procedure requires numeric values for the absolute reflectance at each wavelength. Right now we assume 95% at all wavelengths. @TinoDornbusch we're looking for tabular values for VNIR and SWIR wavelengths.

serbinsh, 2016-09-27:
If you received a reflectance calibration standard from Labsphere you should also have received a PDF and a small CD with the calibration coefficients to convert relative reflectance to absolute reflectance.  And example PDF is attached here for one of my standards

S
------------------------------------------------------------------------------
Shawn P. Serbin
Environmental & Climate Sciences Department
Brookhaven National Laboratory
Upton, NY 11973-5000

Phone: +1 631-344-3165
Fax: +1 631-344-2060
Email: sserbin@bnl.gov<mailto:sserbin@bnl.gov>
Bio:  http://www.bnl.gov/envsci/bio/serbin-shawn.php
TEST Group: http://www.bnl.gov/TEST/
ORCiD: http://orcid.org/0000-0003-4136-8971
Twitter: @doctorjackpine, @test_bnl
------------------------------------------------------------------------------

99AA02-0316-4214	
250	0.9618
251	0.9622
252	0.9633
253	0.9638
254	0.9645
255	0.9641
256	0.9647
257	0.9644
258	0.9650
259	0.9646
260	0.9648
261	0.9655
262	0.9659
263	0.9668
264	0.9661
265	0.9669
266	0.9672
267	0.9665
268	0.9671
269	0.9678
270	0.9672
271	0.9659
272	0.9660
273	0.9673
274	0.9682
275	0.9676
276	0.9690
277	0.9695
278	0.9705
279	0.9705
280	0.9693
281	0.9702
282	0.9703
283	0.9707
284	0.9721
285	0.9728
286	0.9738
287	0.9729
288	0.9748
289	0.9754
290	0.9765
291	0.9759
292	0.9755
293	0.9769
294	0.9778
295	0.9779
296	0.9783
297	0.9781
298	0.9802
299	0.9796
300	0.9796
[... 2175 lines cut ...]
2476	0.9504
2477	0.9499
2478	0.9298
2479	0.9505
2480	0.9621
2481	0.9390
2482	0.9501
2483	0.9375
2484	0.9217
2485	0.9336
2486	0.9376
2487	0.9423
2488	0.9462
2489	0.9288
2490	0.9303
2491	0.9339
2492	0.9527
2493	0.9371
2494	0.9524
2495	0.9458
2496	0.9163
2497	0.9445
2498	0.9340
2499	0.9393
2500	0.9365

LTBen, 2016-09-28:
We did, the CD with the reflectance calibration is in Maricopa with the target, in a red envelope.
I’m not sure if Bob has it already or if Stuart has access to it, either way we will upload the content of the CD soon.

Would it help for setting up the workflow to get the values from a different target in the meantime?

Best regards
Ben

Ben Niehaus

LemnaTec GmbH
Pascalstraße 59
52076 Aachen, Germany

Phone: +49 2408 9383-301
Fax: +49 2408 9383 300

E-Mail: ben.niehaus@lemnatec.de
Internet: http://www.lemnatec.com

Geschäftsführung: Stefan Frey (CEO), Ralf Radermacher (CFO)
Registergericht: Aachen Registernummer: HRB 7295
Umsatzsteuer-Identifikationsnummer gemäß §27 a Umsatzsteuergesetz: DE 812467325

TinoDornbusch, 2016-09-29:
Attached the ceriticate for Spectralon white target
[SphereOptics Panel.pdf](https://github.com/terraref/reference-data/files/500521/SphereOptics.Panel.pdf)

TinoDornbusch, 2016-09-29:
Attached the spectral response of VNIR (G4-384) and SWIR (G4-383) we have got from headwall

[G4-383 & 384 Pixel to wavelength.xlsx](https://github.com/terraref/reference-data/files/500526/G4-383.384.Pixel.to.wavelength.xlsx)

TinoDornbusch, 2016-09-29:
Spectral characteristics of RGB camera not available according to AVT support:

Sehr geehrter Herr Dr. Dornbusch,

Das Quanteneffizienz-Diagram bekommen wir direkt von dem Hersteller, in diesem Fall OnSemi. Onsemi veröffentlicht keine numerische Werten für ihre Sensoren. Auf Seite 9 und 10 des entsprechenden Onsemi Manuals der Sensor, befindet sich  (Link: http://www.onsemi.com/pub_link/Collateral/KAI-08050-D.PDF). Bitte darauf achten, dass ihre GT ein Sensor der 1. gen besitzt.

Mit freundlichen Grüßen,  

i.A. Bernardo Luck // Applications Engineer

czender, 2016-09-29:
@TinoDornbusch thank you for posting the spreadsheet and certificate. It appears on my MacOS application (Numbers) to show pixel vs. wavelength for both sensors. I thought there would also be a spreadsheet with the calibrated reflectance vs. wavelength at 1 nm resolution (based on file posted by @serbinsh). Instead, for reflectance, the only information I see is a poor quality photo of a PDF that shows a plot and has printed numeric values every 50 nm. We would like a file with all the data used to plot those curves. Am I missing something? please check the disk for the tabular spectral reflectances at 1 nm resolution. tagging @solmazhajmohammadi too.

dlebauer, 2016-09-29:
@craig-willis please see e6a540a6ac40bff4201d46c43bfd0f7a787ca81e for how the spectralon data are used in the computing pipeline, and where being able to query it on the fly would be of value.

@TinoDornbusch fig 6 provides the RSR curves for the RGB, no? Is AVI willing to provide the figure but not the raw data?

solmazhajmohammadi, 2016-09-29:
@dlebauer Here is the response I got from Allied Vision:

=================================================================================

_Hello Solmaz, 
 I regret it's not available in spreadsheet format.  I guess you could create such a table "by eye and hand", by reading from the graph.  Remember, it's just a sample measurement meant to be representative, so modest variations according to how pure your light source is etc. would all factor in anyway.
Regards,
Scott_

===================================================================================

And as I mentioned in today's talk Crop Circle is not providing this information. Proprietary ....

solmazhajmohammadi, 2016-09-29:
Spectral Response Curve for PAR sensor : 
[SQ-110 and SQ-500  Spectral response.xlsx](https://github.com/terraref/reference-data/files/501829/SQ-110.and.SQ-500.Spectral.response.xlsx)

TinoDornbusch, 2016-09-30:
@czender  This is all we got from headwall. Further we have no tabular data of the Spectralon target. 

@dlebauer AVT gets these response curves from their chip supplier and they told me that they do not have the raw data.

dlebauer, 2016-09-30:
This looks like it
https://github.com/terraref/reference-data/blob/master/Zenith%20LiteT%20Diffuse%20Reflectance%20Target%20-%2095%25R.csv

czender, 2016-09-30:
@dlebauer Yes! looks like @serbinsh received 50 nm more calibration, out to 2500 nm,  but we'll take what we can get.

solmazhajmohammadi, 2016-10-12:
RSR Curve for NDVI and PRI sensors 

[45951-45954.xlsx](https://github.com/terraref/reference-data/files/524953/45951-45954.xlsx)

craig-willis, 2016-10-26:
@TinoDornbusch 

I'm new to RSR curves and have been asked to convert the provided files into a common matrix format . Most of the provided spreadsheets report the response at a given wavelength, except for VNIR/SWIR. "G4-383 & 384 Pixel to wavelength.xlsx" has two columns Pixel and Wave. It's unclear to me how this relates to the RSR curve -- is this something you can help clarify for me?

solmazhajmohammadi, 2016-10-26:
@craig-willis please disregard the VNIR/SWIR file. That's not a RSR curve

dlebauer, 2016-10-26:
@solmazhajmohammadi what is it? Are the RSR curves for these sensors available?

solmazhajmohammadi, 2016-10-26:
@dlebauer That's wavelength to pixel calibration data which is shown in the headwall camera's certificate. Headwall does not have the RSR curve for SWIR (not delivered from their suppliers), and for the VNIR, LemnaTec will sign a non-disclosure agreement to get the file.

craig-willis, 2016-10-26:
@dlebauer
I've put together an example Clowder dataset for the NDVI sensors with the RSR matrices embedded in the attached JSON.

In this case, there are two NDVI sensors (serial numbers 45951 top/up and 45952 down). 

I've created a dataset in clowder-dev for the NDVI sensor with attached PDFs, etc.

https://terraref.ncsa.illinois.edu/clowder-dev/datasets/5811034ce4b0d51359b959e8

The attached metadata is available via the metadata.jsonld endpoint. For example, loading the JSON data via R:

`
d <- fromJSON("https://terraref.ncsa.illinois.edu/clowder-dev/api/datasets/5811034ce4b0d51359b959e8/metadata.jsonld")
d$content$sensors[[1]]$serial_number
d$content$sensors[[1]]$rsr$data
`

Feedback welcome.

dlebauer, 2016-10-26:
@craig-willis looks great! @ashiklom I think that the following would be identical to the rsr curves that you have: 

```r
rsr <- d$content$sensors[[1]]$rsr$data[[1]]
colnames(rsr) <- d$content$sensors[[1]]$rsr$dimnames[[1]]
head(rsr)
     wavelength CH1 CH4
[1,]        350   0   0
[2,]        351   0   0
[3,]        352   0   0
[4,]        353   0   0
[5,]        354   0   0
[6,]        355   0   0
```

serbinsh, 2016-10-26:
I agree this organization looks great and is comparable to what we work with for other sensors, nice job!  I really like how easy it is to pull the data from JSON in R

craig-willis, 2016-11-01:
Hello @czender 

One of my tasks has been to define a way to store reflectance data so that it's accessible programmatically and I could use your input on how I'm handling the SphereOptics Zenith target. 

Similar to the approach taken for the sensor RSR data above, I've created a dataset to represent the SphereOptics Zenith Polymer target in Clowder that contains a product datasheet and the calibration certificate:

https://terraref.ncsa.illinois.edu/clowder/datasets/5818bb544f0c63f2a8d20963

I've added metadata  to the "dataset" via the Clowder API in JSON-LD format:
https://terraref.ncsa.illinois.edu/clowder/api/datasets/5818bb544f0c63f2a8d20963/metadata.jsonld

The basic JSON-LD structure is required by Clowder, but I've defined the "content" object contents. This includes basic metadata (manufacturer, serial number, etc) as well as a "reflectance" object with fields "wavelength" and "R" with values taken from 

https://github.com/terraref/reference-data/blob/master/Zenith%20LiteT%20Diffuse%20Reflectance%20Target%20-%2095%25R.csv

I think this matches the matrix in your hyperspectral extractor.

At this point, I'm just looking for general feedback: Would the JSON API approach be useful to you? Does the structure make sense?

czender, 2016-11-03:
Hi Craig,

Thanks for systematizing the Spectralon data. 
The JSON-LD structure is "metadata-poor", yet should be serviceable for some programs. If the goal is to make this site the one-true repository of the calibration data, then please add all the relevant metadata for the wavelength and reflectance fields that you find in the hyperspectral_calibration.nco file. Also, perhaps we should adopt the same names for the variables (i.e., factory_calibrated_reflectance).
The JSON API is not yet useful to us to read-in the data, although it may become useful with more work on our end. This is the first I've seen about the "content" metadata. Perhaps we can make more use of that ourselves when we dump JSON.

Charlie

craig-willis, 2016-11-07:
@czender Thanks for the feedback.  I'll make the changes as suggested.

rachelshekar, 2017-01-12:
@craig-willis - can you please update this issue?  Can a new issue be made?

craig-willis, 2017-01-12:
@rachelshekar As far as I'm concerned, yes (I created #76 for actions I'm aware of). There are some unchecked boxes at the top and I'm not sure whether we're still waiting for data for any of the devices.

rachelshekar, 2017-01-13:
@dlebauer - are we waiting for data from VNIR,  SWIR, and crop circle?

craig-willis, 2017-01-13:
@dlebauer 

Per Solmaz's follow-up comments, those are not RSR curves for the VNIR and SWIR. I missed her second comment that there is nor RSR curve for SWIR and that Lemnatec will not sign the NDA for VNIR.

So, no the file attached to that comment are not captured in the metadata. It wasn't clear to me whether the pixel-to-wavelength data was useful.   It sounds like we will not have RSR for Crop-Circle, VNIR, or SWIR.  If this is acceptable, and I don't need to do anything with the pixel-to-wavelength data, I'll close this.

dlebauer, 2017-01-13:
@craig-willis I think it is acceptable not to have RSR curves for VNIR and SWIR. Presumably this is corrected in the calibration. If there is a demand / use case we can revisit this. We can likely generate RSR curves for these instruments with acceptable precision using the a radiometer and Spectralon target, or some other method. I am closing this issue. 

However, I see @remotesensinglab hasn't been on this thread ... so I am pinging him and @Mamatemenrs FYI.

closed 2017-01-13 by dlebauer