terraref/reference-data #54: Format and method for publishing authoritative sensor metadata

craig-willis, 2016-09-21:
The JSON data produced by Lemnatec includes a "sensor_fixed_metadata" section with the sensor name, description, manufacturer, and serial number.  We also have copies of the sensor manuals and data sheets with calibration data, sometimes as PDF or CSV.

For example:
```
    "sensor_fixed_metadata": {
      "sensor manufacturer": "Headwall Scientific",
      "sensor product name": "VNIR",
      "sensor serial number": "PaS6-114",
      "sensor description": "hyperspectral camera to measure visible near infrared (VNIR) radiation",
      "sensor purpose": "measures spectral reflectance from 380nm to 1000nm"
    },
```

The goal is to publish this authoritative sensor metadata in a way that can be easily consumed programmatically.  Ideally, the sensor itself would be described using some standard format and linked to the datasheet/manual/calibration data.

[SensorML](http://www.opengeospatial.org/standards/sensorml) seems to be targeted more at the sensor manufacturer and much more complex than needed.  The OGC [SensorThings API ](https://en.wikipedia.org/wiki/SensorThings_API )defines a model that includes a simple Sensor object (Name, Description) that can be linked to a SensorML or PDF data sheet.  There is also the newer [Sensor Network Ontology](https://www.w3.org/TR/vocab-ssn/).

Initial options appear to be to:
* SensorML description of each sensor
* Use the SensorThings API model, which doesn't include all of the attribute (e.g., Serial number, purpose, etc).
* Use the Sensor Network Ontology (along with other ontologies) to produce an RDF/JSON-LD description of the sensors.

JeffWhiteAZ, 2016-09-21:
The JSON example from LemnaTec would be enough information to allow a researcher to write a sentence in a methods section for a paper, but it wouldn't help someone troubleshooting data problems. Would we have information on sensor maintenance and routine calibration linked via the serial number and contained in a separate sensor log?

craig-willis, 2016-09-21:
@JeffWhiteAZ Are we already collecting this information?  I hadn't considered it, but linking from the sensor description to a maintenance log shouldn't be too difficult.

JeffWhiteAZ, 2016-09-21:
A more less binary flag is set for each sensor, every time the operator logs the system status.
You can see this at:
https://docs.google.com/spreadsheets/d/1eQSeVMPfrWS9Li4XlJf3qs2F8txmddbwZhjOfMGAvt8/edit#gid=1394972273
I don't believe that more detailed information on calibrations or maintenance is being captured, in part because we are still learning from LemnaTec as to what these involve. Note also that the current log references generic sensors and not specific serial numbers. If an instrument were swapped (e.g., for upgrade or replace a defective system), we would rely on the operator to provide additional notes.

dlebauer, 2016-09-22:
Notes on scope of content: 

* info from sensor docs
* reference / calibration objects
  * what they are
  * where they are
* calibiration protocols

from @ashiklom and @serbinsh 
developing a mini relational database (SQLite) with tables:
* "spectra" stored by band number and spectral ID (one per spectrum), 
* "specInfo" that links each spectral ID to an instrument, collection date, species, location, etc., an 
* "instruments" table that provides information about each instrument
  * model name, probably something about the way the measurement is taken (e.g. leaf clip vs. integrating sphere), what kind of measurement it is (directional vs. integrated over the hemisphere), if possible, something about the radiometric accuracy.
  * most important info is RSR, FWHM, camera info (like optics, geometry, etc), band numbers/centers, calibration date and wavelength calibrations, keeping up-tp-date calibration spectra showing the spec response for different spec targets like Mylar, and different lamps. Temperatures of detectors during collection, offsets/gains, radiometric info to convert from DN to radiance to reflectance. Formats will vary but metadata vocab should be consistent, that is optical imaging vocab
  * spectrometer characteristics such as spectral channels, sampling, FWHM, radiometric info, etc
* "RSR" table that has columns "instrumentID", "band_number", "wavelength", and "spectral response".

craig-willis, 2016-09-22:
@dlebauer Please clarify the above comment -- when we spoke yesterday we discussed the fixed sensor metadata from Lemnatec, PDF manuals, and PDF/CSV calibration data. 

From your comment, it sounds like you want this stored in a database? From my understanding, most of the information was in a PDF file -- so are you expecting that we'd extract data from the PDF to put in the relational database.

dlebauer, 2016-09-22:
@craig-willis on one hand, those are quick notes from a conversation with colleagues on what is useful, and much of it is already within the scope (e.g. RSR, camera info) + calibration details. once we have a format for storing the relevant sensor metadata, we should extract relevant information from the PDFs and make them accessible. This can all be in json - I only mentioned SQLite because that is what @ashiklom is working with. It doesn't matter so much how the data is stored but if there is already a database model that we can use it might be worth adopting it / helping build it.

craig-willis, 2016-09-22:
Thanks, @dlebauer -- I understand now.  If available, pointers to the project, database schema, and any associated documentation would be helpful.  

Can you remind me where I can find the PDFs for the various sensors?

dlebauer, 2016-09-23:
They are on google drive here: https://drive.google.com/open?id=0B7JeWs8SrKqJSlpjYjktQ1BFdjA

And here, for convenience:

[Vaisala.pdf](https://github.com/terraref/reference-data/files/490068/Vaisala.pdf)
[Ocean Optics.pdf](https://github.com/terraref/reference-data/files/490070/Ocean.Optics.pdf)
[FLIR.PDF](https://github.com/terraref/reference-data/files/490071/FLIR.PDF)
[G4-384 VNIR Wavelength_Calibration.pdf](https://github.com/terraref/reference-data/files/490067/G4-384.VNIR.Wavelength_Calibration.pdf)
[G4-383 SWIR Wavelength_Calibration.pdf](https://github.com/terraref/reference-data/files/490069/G4-383.SWIR.Wavelength_Calibration.pdf)
[Skye_NDVI_PRI_Protokolle.pdf](https://github.com/terraref/reference-data/files/490072/Skye_NDVI_PRI_Protokolle.pdf)
[09001-SensorDocumentation.pdf](https://github.com/terraref/reference-data/files/490073/09001-SensorDocumentation.pdf)

dlebauer, 2016-09-23:
@craig-willis there is already a spreadsheet on google drive that summarizes a lot of the general information about these sensors: https://docs.google.com/spreadsheets/d/1xkEHQd_5x3Yzv3f61bfns3ZFoP7xXq8AUNK4fmS3kXg/edit#gid=194274492

(note that on google drive these are in the folder terraref/terraref-danforth-team/field scanner operations)

craig-willis, 2016-09-26:
@dlebauer  I've been comparing the "sensor_fixed_metadata" in the latest *metadata.json files to the calibration certificates in Google drive and noticed the following:

* FLIR: sensor_fixed_metadata product name (A645) and serial number (13070601) do not match calibration certificate (A615, 55003076)
* VNIR/SWIR: sensor_fixed_metadata serial number (PaS6-114/PaS6-115) does not match calibration certificate (G4-384/G4-383). 

I expect this is known, but there are no calibration certificates for the Stereo-VIS, PSII camera, 3-D scanner.  Is there another way to verify the model/serial number?

dlebauer, 2016-09-26:
@craig-willis I've put the issues with inconsistent and missing sensor data sheets in #56 and have assigned this to @TinoDornbusch

craig-willis, 2016-09-26:
@dlebauer 

Discussed two options for storing sensor data with @max-zilla and @robkooper today:

1. Store in Clowder.  Create a Collection "Sensor Information" with per-sensor dataset. The dataset would contain the sensor metadata, PDF, calibration matrices, photos, etc.  Each sensor would therefore have a linkable unique ID in Clowder.  With the postgres plugin enabled, we can also add the sensor as a geostream/stream, linking the data from the sensor to the sensor dataset.

2. Store outside of Clowder.  Host the sensor description information (metadata, manuals, etc) in external service (could be a simple webserver). This would serve up both human-readable and machine-readable (json) content.  The URI for each sensor would serve as an ID (e.g., https://terraref.org/sensors/VNIR).  This could be added to the geostream/stream metadata.

The plan is to try out no 1 this week for a single sensor and test it out, with a first pass by the end of this week.  Let me know what you think.

dlebauer, 2016-09-26:
Both of approaches sound sensible and it makes sense to start with the first.

craig-willis, 2016-09-27:
Thanks, @dlebauer 

I've started a collection on the dev instance: 

https://terraref.ncsa.illinois.edu/clowder-dev/collection/57ea83a4e4b0581365b856c0

Also -- do we have more LemnaTec specifications similar to  the [stereo camera](https://github.com/terraref/reference-data/files/490082/Stereocamera_Specification_V06.pdf). This is a good source of contextual information that isn't necessarily in the individual sensor datasheets.  If there's an overall specification for the Scanalyzer, that might be helpful too.

dlebauer, 2016-09-27:
@craig-willis you can feel free to curate the collection on the production instance if it would be easier.

@TinoDornbusch will know if there are specification docs from Lemnatec similar to the one from the stereo camera. 

There is a specification for the entire Gantry 
[7100019_System_Specification_V10_General_Version.pdf](https://github.com/terraref/reference-data/files/496697/7100019_System_Specification_V10_General_Version.pdf).

craig-willis, 2016-09-27:
@dlebauer I'm using the dev instance because it has the postgres plugin enabled, which allows me to create the associated geostreams/sensor records, and I'm learning a few lessons along the way.  I will certainly move to the production instance as soon as it makes sense.

craig-willis, 2016-10-26:
@dlebauer I'm looking for the sensor_fixed_metada for the Quantum PAR and Ocean Optics spectrometer, but don't see folders for either in Globus. Is this expected?  I can create the fixed metadata from the Lemnatec sensor documentation.

dlebauer, 2016-10-26:
These are both in the EnvironmentalLogger meta-data

* quantum PAR sensor:

```json
  "environment_sensor_fixed_infos": {
    "par_sensor": {
      "manufacturer": "www.apogeeinstruments.com",
      "model": "SQ214",
      "location in gantry system": "top of gantry"
    },
```

The Ocean Optics spectrometer:

```json
    "spectrometer": {
      "manufacturer": "www.oceanoptics.com",
      "model": "STS-VIS",
      "location in gantry system": "top of gantry"
    }
```

craig-willis, 2016-10-28:
@dlebauer it looks like the SensorML group has created a set of vocabularies (sensorml.com/orr/) that will be useful here without requiring us to conform to the SensorML spec.  

Question:  The PDF 09001-SensorDocumentation_V4_oK.PDF from Lemnatec has a ton of device/sensor information, but it would be cumbersome to extract (copy/paste) values from PDF.  Who would I contact to find out whether the information is available in another format?

dlebauer, 2016-10-28:
@TinoDornbusch @LTBen @markus-radermacher-lemnatec : do you have the sensordocumentation in a different format, e.g. word with tables etc?

@craig-willis I would consider the sensor metadata that we already have in spreadsheets and text files (like metadata.json and google sheets) to be the first priority. If it is onerous to extract data from the pdf, we can extract additional parameters can be added based on need. You could try to import the pdf into word; I've had good luck with a software called 'table extractor' http://pdf.zanran.com/index.php#section-06 and https://pdftables.com/ (which I used to generate 
[09001-SensorDocumentation_V4_oK.xlsx](https://github.com/terraref/reference-data/files/559590/09001-SensorDocumentation_V4_oK.xlsx) ... does that help?

craig-willis, 2016-10-28:
Thanks @dlebauer -- yes, this is helpful and I do understand the priorities.  However, some of this information is duplicated across all three sources (JSON, Sheets, PDF).  For the calibration, certification and test-specimen data, which should I consider to be the more reliable source of information: JSON, Google sheets or PDF?  In some ways it seems that the PDF is actually more up-to-date.

dlebauer, 2016-10-28:
I'd consider reliability json > PDF > google sheets. If you see any differences between json and PDF we can use the json value but ask Lemnatec to resolve them.

craig-willis, 2016-10-31:
@dlebauer 
Focusing just on the calibration information, the JSON data is both incomplete and inconsistent.  The PDF reflects what's actually in the calibration certificates.  For example, the sensor_fixed_metadata only has calibration dates for 3 sensors, we have certificates for 7 (same in PDF); the sensor_fixed_metadata has three different fields "calibration available", "Calibration available", "is calibrated" -- set to false for 3 sensors and not present for the rest.

My inclination would be to use the PDF as the source of truth. If the goal is to use the sensor_fixed_metadata as the source of truth, then I guess I'll need to audit and compare to the PDF for accuracy and file a ticket with corrections.  Who should I contact for more information about the sensor_fixed_metadata?  I'd like to get documentation from Lemnatec about each of the fields.

dlebauer, 2016-10-31:
@craig-willis ok, makes sense to use PDF as source of truth but to keep track of inconsistencies and let Lemnatec know

craig-willis, 2016-10-31:
@dlebauer Looking back at https://github.com/terraref/reference-data/issues/42,  this seems to be a duplicate.  Feel free to assign to me if you agree.

I've created and populated a collection on the production Clowder instance with the various PDFs and datasheets and with the dataset metadata set to the original raw LemnaTec fixed sensor data. 

https://terraref.ncsa.illinois.edu/clowder/collection/58035fa34f0c4a438cbb53dc

Because of all of the work already done on the LemnaTec fixed sensor data (particularly [151](https://github.com/terraref/computing-pipeline/issues/151), [170](https://github.com/terraref/computing-pipeline/issues/170)), I'm concerned about me arbitrarily changing the structure of the JSON data for this ticket.

I've created a separate Github repository with the fixed sensor metadata to allow us to use the PR review/approval process for changes to the JSON structure and content. 

https://github.com/terraref/sensor-metadata/

I will tag this before adding new data (e.g., RSR data, updated calibration information ,etc). 

Let me know if you have any concerns.

craig-willis, 2016-12-15:
I think this issue can be closed after we have the sensor metadata review. Additional issues can be opened to address feedback or recommended changes.

craig-willis, 2017-01-12:
Closing this issue following the Sensor Metadata review meeting on 1/9.  Will open separate tickets for specific tasks going forward.

closed 2017-01-12 by craig-willis