terraref/reference-data #55: ICASA RDFS/OWL ontology

craig-willis, 2016-09-24:
As a result of the 9/7/2016 meeting "[TERRAref RDF model for ICASA](https://docs.google.com/document/d/1rPZTQ1AmVs4C0PVrsu_5o0oGXgckfW19vGPeKnzZdIo/edit)", we agreed to pursue a best-effort rendering of the [ICASA Master Variable List](https://docs.google.com/spreadsheets/d/1MYx1ukUsCAM1pcixbVQSu49NU-LfXg-Dtt-ncLBzGAM/pub?output=csv) as a linked-data vocabulary or ontology. The goal is to be able to use the AgMIP [JSON Data Objects](http://research.agmip.org/display/dev/JSON+Data+Objects) as a starting point for JSON-LD or similar serialization of TERRA-REF management and trait data.

This ticket is complete when we have:
* An RDFS or OWL implementation of the ICASA master variable list hosted on an authoritative vocabulary server.  This includes management as well as measured data.
* Associated JSON schemas and JSON-LD contexts to demonstrate how to use the ICASA ontology for the AgMIP JDO and ultimately TERRA-REF.

craig-willis, 2016-09-24:
@chporter @JeffWhiteAZ 

This is in progress, but I do have a preliminary OWL implementation for the ICASA Management_Info:

https://github.com/craig-willis/icasa

A few questions/issues I could use guidance on:
* Person/Institution/Document models appear to be duplicated for Experiment, Soil, Weather, and Suite. I think these can be represented using a single class (e.g., Person instead of SoilPerson).  Is there a reason to model them separately?
* Some of the variables/datasets/subsets/groups are not included in the V 2.0 documentation. For example Suite, DOME Simulation, Simulation Control and AgMIP group variables. Should I ignore or include these?  If include, would it be possible to get definitions?
* The AgMIP JDO appear to generally use the Code_Display values for field names. I noticed that some codes in the JDO examples do not exist in the spreadsheet (people, tr_name, icrzno, icbl, elev). I assume this just means AgMIP has added variables that aren't in the ICASA tables. Also, the JDO examples in one instance use the variable_name instead of the code (crop_model_version instead of model_ver).
* On a related node, some codes contain % or # in the name.  This is unusual -- I'd be inclined to change to PCT and NO for the OWL ontology -- but can stay true to the spreadsheet if best.

chporter, 2016-09-26:
@craig-willis @JeffWhiteAZ 
Craig - this looks like a great start. I think we should schedule another call to discuss, but here are some short answers.

1st question - Yes, I think that some of these variables could be collapsed. But the people collecting soils and weather and experimental data are very likely to be different. As long as we can capture that, we are OK.
2nd question - We added some AgMIP-specific variables, including DOME references and AgMIP ratings, etc. Maybe needs some discussion as to whether these need to become part of the ICASA standards. I think the format allows for adoption of user-specific variables that are not part of the standard.
3rd question - All variables used by AgMIP should be in the ICASA DD. I will have to investigate further, but my guess is that the documentation is incorrect.
4th question - the 'code_query' eliminates the '%' and '#'. In our JSON files, though the code_display is used.

craig-willis, 2016-09-26:
Thanks for the feedback, @chporter.  I'll setup a follow-up call, but would like to finalize a couple of JSON-LD examples that cover both the AgMIP examples and PlantCV.

Your responses all make sense (and I just noticed the "Explanatory_Notes" sheet in the ICASA spreadsheet).  Regarding question 3, these are likely simple typos (tr_name = trt_name; icrzno=icrz#).  

I've just checked in a couple of changes:
* Added examples/agmip_jdo.jsonld as a very rough example of how we can support the existing "json-translation-samples."  I ran this through http://rdf-translator.appspot.com/ to generate triples for testing.
* Added Management, Event, MeasuredData, SummaryData, and TimeSeriesData classes to icasa-mgmt-info.owl.
* Changed the python script to use Code_Query instead of Code_Display. 
* Added icasa-units.csv to start collecting definitions for the units.  I think this might be helpful, since some of these units might not be in existing vocabularies.

max-zilla, 2017-03-08:
@craig-willis and I are discussing Phenome 2017 outcomes re: BRAPI, MIAPPE, other equivalency tables and how this still fits in. Meeting is March 9 (tomorrow)

rachelshekar, 2017-03-08:
Next step: update from Phenome meeting from Max.  Meet with Max and David to determine future direction and next steps.

closed 2018-06-13 by craig-willis