terraref/reference-data #56: Missing calibration data sheets

dlebauer, 2016-09-26:
From https://github.com/terraref/reference-data/issues/54#issuecomment-249656855

@craig-willis noticed

1. comparing the "sensor_fixed_metadata" in the latest *metadata.json files to the [calibration certificates in Google drive](https://drive.google.com/open?id=0B7JeWs8SrKqJSlpjYjktQ1BFdjA) and noticed the following:
  * FLIR: sensor_fixed_metadata product name (A645) and serial number (13070601) do not match calibration certificate (A615, 55003076)
  * VNIR/SWIR: sensor_fixed_metadata serial number (PaS6-114/PaS6-115) does not match calibration certificate (G4-384/G4-383).
2. There are no calibration certificates for 
  * Stereo-VIS, 
  * PSII camera, 
  * 3-D scanner

TinoDornbusch, 2016-09-28:
1. This is an error in the FLIR firmware of the camera. Product name (A645) and serial number (13070601) are encoded in the camera firmware. We change this our code.

Regarding the Hyperspecs serials. I personally noted the serial numbers down on site. Will double check asap.

2. These sensors are not calibrated.  Cameras as build in stereo vis and PS2 are never calibrated. 
3D scanner is calibrated by Fraunhofer internal routines and tested based on the test target.

craig-willis, 2016-09-28:
@dlebauer  @TinoDornbusch 
Regarding the FLIR camera, doesn't this imply that we have the wrong calibration certificate? How is it possible that the certificate, which includes the serial number, wouldn't match the physical device?

TinoDornbusch, 2016-09-29:
@craig-willis 
I do not know how this happened. You might want to address this issue to FLIR. Either they have shipped a wrong camera or they put the wrong firmware on it.

TinoDornbusch, 2016-09-29:
@craig-willis 
I double checked with our logistics, the delivered camera is Model A615, serial 55003076. I update the meta data

TinoDornbusch, 2016-09-29:
Serial number of VNIR and SWIR has been changed according to certificate.

rachelshekar, 2016-10-11:
@craig-willis - we are still in need of the calibration certificates for
Stereo-VIS,
PSII camera,
3-D scanner

Correct?

craig-willis, 2016-10-11:
@rachelshekar 
Based on my records, that correct.  I'm not clear on whether we should expect them, but I don't have callibration information for Crop Circle, PAR, the Theis Clima either.

dlebauer, 2016-10-26:
@TinoDornbusch any luck finding calibration documents for the remaining sensors?

rjstrand, 2016-10-27:
@dlebauer 
I just talked to Holland Scientific.  They set the zero and span for the Crop Circle at the production facility and then ship them.  There is no documentation.

dlebauer, 2016-11-17:
@craig-willis At this point, I'd say gather+organize what we have and close the issue. We can re-open / address as needed based on need / end-user demand.

craig-willis, 2016-12-01:
Closing this.  All available calibration data sheets have been added to Clowder.

closed 2016-12-01 by craig-willis