terraref/reference-data #57: Add Agronomic management data to BETYdb

dlebauer, 2016-09-27:
Add information on fertilization, date of planting, any tillage, and irrigation to BETYdb.

This can be done via the BETYdb web interface (https://terraref.ncsa.illinois.edu/bety). See documentation for details: https://pecan.gitbooks.io/betydbdoc-dataentry/content/adding_managements.html

NewcombMaria, 2016-09-27:
I've been including agronomic management activities from the current planting in the Field Book (https://docs.google.com/spreadsheets/d/1eQSeVMPfrWS9Li4XlJf3qs2F8txmddbwZhjOfMGAvt8/edit#gid=665425213).  Let me know if it would be helpful to collate the agronomic management activities into a separate document.  (Maria)

dlebauer, 2016-09-27:
@NewcombMaria if you could please either format the managements in this table: https://docs.google.com/spreadsheets/d/1s-kQc3K0h4-hfwMzKOU_zpRXN5loByIpKW8Rvw76mh0/edit#gid=522503440 or add them using the web interface (https://terraref.ncsa.illinois.edu/bety/managements) that would be great. 

To use the web interface, you will need to create an account; then I can give you the permissions that you need and also answer any questions. (this is the same account you will need to upload field measurements).

NewcombMaria, 2016-11-08:
I added management activities for sorghum season1 and sorghum season2 at https://docs.google.com/spreadsheets/d/1s-kQc3K0h4-hfwMzKOU_zpRXN5loByIpKW8Rvw76mh0/edit#gid=522503440

rachelshekar, 2016-11-30:
@dlebauer - is this info in the database now?

NewcombMaria, 2016-12-01:
@rachelshekar the agronomic management data was included in Issue #61 (List of field measurements to import into BETYdb).  Scott Rohde inserted managements from the spreadsheet into the database, but more work needs to be done on the units and levels.  Next week (after harvest) I can work on collecting the missing information.  I'll need to know if I should update the spreadsheet or otherwise how to get the missing info ready for database edits.

closed 2016-12-15 by dlebauer