terraref/reference-data #59: Missing Enviromental data in globus

solmazhajmohammadi, 2016-09-30:
I have noticed that some of the data from EnviromentalLogger is missing. It seems that gantry's weather station is working almost 24hours but still some of the data has not been transferred yet. 

Here is some examples:

**Data from 2016-06-06:**
- Gantry server: 07-09-31: 23-09-34
- Globus: 07-09-31: 15-09-33

**Data from 2016-06-07:**
- Gantry server: Recorded 24hours
- Globus: 03-09-35 : 23-09-35

**Data from 2016-08-26:**
- Gantry: Recorded 24 hours
- Globus: 11-25-32: 23:25:33

rachelshekar, 2016-11-02:
FYI @jdmaloney

rachelshekar, 2016-11-30:
@jdmaloney ???

jdmaloney, 2016-11-30:
@rachelshekar This is included in @max-zilla and I's work on the pipeline, hope to have it all completed by week end.  We're moving at the speed of our disk cache at Arizona & it's Internet connection.

jdmaloney, 2017-01-12:
This was corrected early in December, closing issue.

closed 2017-01-12 by jdmaloney