terraref/reference-data #61: List of field measurements to import into BETYdb

dlebauer, 2016-10-28:
@NewcombMaria please provide a list of the field measurements that are ready to be added to BETYdb (have a date column and are arranged in 'long' format with one measurement per row.

You can edit this top post and list them, e.g. 

* [ ] height measurements for season 2 [here](https://docs.google.com/spreadsheets/d/1B30WB5vD-JAU2Pgluhdu1kIxwttdOcQphSIin7BByKg/edit#gid=2059697035)
* [ ] radiation interception 
* [ ] ...

We should include also the agronomic metadata and the plot --> genotype mappings.

NewcombMaria, 2016-10-30:
Working on it.  Will update field measurements list and agronomic metadata as soon as possible.  Not certain what is meant by plot to genotype mappings.  Unique genotype id??

dlebauer, 2016-10-30:
by plot to genotype mappings, I meant, what genotype was planted in each plot?

NewcombMaria, 2016-11-08:
For genotype names, please use the genotype in the column 'name_corrected' in the spreadsheet https://docs.google.com/spreadsheets/d/107aijRIvO5i-HHwVDQTXn0-9sE02cXO_MdZk3DERzQ0/edit#gid=1166373347
@dlebauer @rachelshekar Let me know if you have questions.  Thanks.

NewcombMaria, 2016-11-08:
The following are formatted and ready for importing into BETYdb:

Season1 plant heights: see 'Sorghum_ManualData_Maricopa2016_HtsAddedJuly22_LongFormatJuly25' the last sheet (Heights_Full_field_LongFormat) in https://docs.google.com/spreadsheets/d/1FQ8_ualom7ZUtr5iKiEn80vHlh9zIYpDxawpaek20Og/edit#gid=2067779457

Season1 emergence stand counts 'F13_2016_sorghum_whole_field.csv' in https://docs.google.com/spreadsheets/d/1r_1aBeIlgROsWWhbH9B_Rc_3vDmqUej1TuhaRGuSwBQ/edit#gid=1350157961 

Season1 emergence vigor 'F13_2016_sorghum_emergence.csv' in https://docs.google.com/spreadsheets/d/1zhyhAGOFJ-KXd_-kF_6kOeqh4NtppKbRwVHRk1cZzAk/edit#gid=23726065 

@dlebauer @rachelshekar Let me know if you have questions.  Thanks.

NewcombMaria, 2016-11-08:
I added 2 new files of Sorghum season2 data in the Data for uploading google docs directory https://drive.google.com/drive/folders/0B4JvLJOau3GmaFZrZ2Yzc2xSbUk

Sorghum2_emergence_SubsetPlots_plusFinalCounts_Nov7.xlsx, see https://docs.google.com/spreadsheets/d/1xIRwroYObD125I0TDCDp3YPIGPi2Qzeh-vlUCg_b85Q/edit#gid=1432184448 .  This has emergence vigor and emergence final stand counts - needs to be formatted.

Sorghum2_NodeTillerCounts_SubsetPlots.xlsx, see https://docs.google.com/spreadsheets/d/10eEcZST1D0_kogUVvK_iG0cFE_I70OTJM6HitYztRBU/edit#gid=707942000 . This has node and tiller counts on marked plants, and  leaf length and width on marked plants, one date - needs to be formatted. 

@dlebauer @rachelshekar   Let me know if you have questions.  Thanks.

NewcombMaria, 2016-11-08:
The following is formatted and ready for importing into BETYdb:

Season1 leaf desiccation ratings (see 'Sorghum_ManualData_Maricopa_Leaf_Desiccation_final' at https://docs.google.com/spreadsheets/d/1CDV3dJSoV5F_mFgZI6cYEt2fUpa4BHgv-adbP704bSQ/edit#gid=2112860580  ).

@dlebauer @rachelshekar  The ratings include an incidence and a severity of leaf desiccation.  Let me know if you have questions.  Thanks

NewcombMaria, 2016-11-08:
I will ask Nadia if someone at Danforth can work on reformatting the following for importing into BETYdb:

Season1 phenology growth stage data, (see 'Sorghum_ManualData_Maricopa_Phenology.xlsx' at https://docs.google.com/spreadsheets/d/1BQeFBlyk9y_eE4wiDKYCwQ2wxMqMx1t8rUrplGSqMhI/edit#gid=194121296 ).

Season1 radiation interception, (see 'Sorghum_ManualData_RadiationInterception' at https://docs.google.com/spreadsheets/d/1QnO2yu_njVTdPueQNrI4O2PY_MBqz9iyN8WZWfVvc1g/edit#gid=482347767 )

@dlebauer  I'll do my best to work on reformatting the other data sets and adding management info.

NewcombMaria, 2016-11-08:
I added management activities for sorghum season1 and sorghum season2 at https://docs.google.com/spreadsheets/d/1s-kQc3K0h4-hfwMzKOU_zpRXN5loByIpKW8Rvw76mh0/edit#gid=522503440 

@dlebauer   It may be incomplete, but it fills in the info that I have available to me at this time.  Thanks

NewcombMaria, 2016-11-08:
I reformatted sorghum season2 plant heights, and added the long format as the last sheet in the following file 'DRAFT_Sorghum2_PlantHeights_AugustPlanting.xlsx',  (https://docs.google.com/spreadsheets/d/1B30WB5vD-JAU2Pgluhdu1kIxwttdOcQphSIin7BByKg/edit#gid=2076100922 ).

@dlebauer @rachelshekar  Please let me know if this needs additional formatting changes.  Thanks

NewcombMaria, 2016-11-08:
I reformatted sorghum season2 emergence vigor, and the emergence vigor and emergence final stand counts are both now in the long format in the spreadsheet 'Sorghum2_emergence_SubsetPlots_plusFinalCounts_Nov7.xlsx' (https://docs.google.com/spreadsheets/d/1xIRwroYObD125I0TDCDp3YPIGPi2Qzeh-vlUCg_b85Q/edit#gid=732810075).

NewcombMaria, 2016-11-09:
Cesar and Nadia (Danforth) reformatted the sorghum season1 phenology data.  The long format data are in the csv file 'Sorghum_ManualData_Maricopa_Phenology_Fixed.csv', see https://docs.google.com/spreadsheets/d/1AFwjkUz5lRzR47xc3R2OwjKd-mRJLz-ongAIzD2Hnvs/edit#gid=1209684786 .  Please find the definitions in the 2nd sheet/tab in the spreadsheet 'Sorghum_ManualData_Maricopa_Phenology.xlsx', in https://docs.google.com/spreadsheets/d/1BQeFBlyk9y_eE4wiDKYCwQ2wxMqMx1t8rUrplGSqMhI/edit#gid=622354195 .  

@dlebauer @rachelshekar  Please let me know if you have questions.

NewcombMaria, 2016-11-09:
Cesar and Nadia (Danforth) also reformatted the sorghum season1 radiation interception data, and I added interpolated incident radiation for each measurement.  The long format data set is in the csv file 'Sorghum_ManualData_RadiationInterception_BAP_Fixed_V2.csv' (https://docs.google.com/spreadsheets/d/1aW3NJ98GxF-XIRibvUwBcuqG6CQVqVqFBW1-_SRfLJ0/edit#gid=1280560409 ).  Please find the definitions in the 2nd sheet/tab in the spreadsheet 'Sorghum_ManualData_RadiationInterception' (https://docs.google.com/spreadsheets/d/1QnO2yu_njVTdPueQNrI4O2PY_MBqz9iyN8WZWfVvc1g/edit#gid=1238504444).  

@dlebauer @rachelshekar   Please let me know if you have questions.  Thanks

NewcombMaria, 2016-11-09:
I reformatted the sorghum season1 leaf node and tiller count data on marked plants.  See the last sheet (long format) in the spreadsheet 'Sorghum_ManualData_NodeCounts_BAP_RepeatMeasures.xlsx' (https://docs.google.com/spreadsheets/d/1hGeZq2CWd-LakcuHZhGXiONFZrdPwWUTVJeDYxDP7ZY/edit#gid=680357011)
Definitions are on the first sheet/tab of the worksheet.

@dlebauer @rachelshekar   Please let me know if you have questions.  Thank you

NewcombMaria, 2016-11-09:
It's still incomplete and will need additional info, but the preliminary sorghum season1 harvest info provided by Mike Ottman is in the spreadsheet 'Sorghum1_harvest_data_July2016_INCOMPLETE_SAS - Copy.xlsx' (https://docs.google.com/spreadsheets/d/1iHmSyFeCO7np_lxYyvdiXQQcAwBJpIKpeAOyMXFd-OY/edit#gid=901711246) .

dlebauer, 2016-11-09:
@gsrohde could you work on importing these spreadsheets into the bety6 database?

I've updated the cultivars (https://gist.github.com/dlebauer/e42dd56cc7823ff1cf3f3852218a6886)

dlebauer, 2016-11-09:
@NewcombMaria is this emergence vigor or emergence count? ['F13_2016_sorghum_emergence.csv'](https://docs.google.com/spreadsheets/d/1zhyhAGOFJ-KXd_-kF_6kOeqh4NtppKbRwVHRk1cZzAk/edit#gid=23726065),

NewcombMaria, 2016-11-09:
Regarding:  is this emergence vigor or emergence count? 'F13_2016_sorghum_emergence.csv':
     It's emergence count over time, or an emergence rate, which some prefer to call emergence vigor.  A repeat measure of emergence over time, which indicates emergence vigor

gsrohde, 2016-11-09:
@dlebauer Before I begin work on importing these, I need to know _exactly_ which spreadsheets you mean.  For example, do you mean _every_ CSV or spreadsheet file linked to in this issue?  Or only those with unchecked checkboxes next to them?  

Then I will probably have questions about each one, since they are not yet in the format required for bulk upload.  And perhaps some don't even pertain to trait data, so I will have to devise some other method of inserting the data.

dlebauer, 2016-11-10:
@gsrohde the links in https://github.com/terraref/reference-data/issues/61#issuecomment-259049732 go to the relevant worksheets. The first set of 8 should be ready or close enough. Did you see these? All are traits except the managements that can be entered via the web interface or insert statements.

gsrohde, 2016-11-10:
@dlebauer Many or most of these files contain several sheets.  It's not clear whether I need only pay attention to the one that opens upon clicking the link.

So I'll start with items 2 and 3, which each consist of only a single CSV file.  Here are my questions and comments on those:

First of all, I’m assuming all new trait rows connected with this data should refer to citation "Newcomb, Maria, 2016, _Field measurements from Maricopa Agricultural Center_” and species “Sorghum bicolor”.

Can I assume that date values are the date _at the site_?

Should `access_level` be 2 as for the other data?

I’m assuming no treatment, method, or entity unless you tell me otherwise.

File-specific notes:

Item 2 (F13_2016_sorghum_whole_field): I removed every row that didn’t have a date and a value for `emergence_count`.  I’m assuming no specific cultivar for this data.  I’m ignoring the `Entry` column; I’m also ignoring `plot_id` and `Row`, these being encapsulated in the site name.

Item 3 (F13_2016_sorghum_emergence): Again, I’m ignoring the `Entry`, `plot_id`, and `Row` columns.

dlebauer, 2016-11-10:
Yes, data are in the worksheet that opens from the link. Many I've highlighted columns to upload in green

Yes all of your assumptions are correct. We will insert  cultivars and methods separately / later

gsrohde, 2016-11-10:
OK, items 2 and 3 are done.  Here are the id numbers of the new traits:

F13_2016_sorghum_whole_field: 6000018709–6000019908
F13_2016_sorghum_emergence: 6000019909–6000021252

I'll take a look at the other files next and see if I have further questions.

NewcombMaria, 2016-11-10:
@gsrohde If you have questions for me on the other files as you move forward, I'll do my best to answer promptly.  Thanks!

gsrohde, 2016-11-10:
I'm about to upload the first spreadsheet ('Sorghum_ManualData_Maricopa2016_HtsAddedJuly22_LongFormatJuly25').  I'm making the same assumptions as before.  I'm ignoring columns, DataEvent, Experiment, Plot,	Range, Column, Rep, Entry, and Row_subplot.  I removed all rows having neither a `canopy_height` nor a `spike_height` value.  I partitioned the remaining rows into two separate files: One for those having a `spike_height` value and one for those that don't.  This is because neither of the bulk upload methods currently allows leaving a trait value blank.

gsrohde, 2016-11-10:
'Sorghum_ManualData_Maricopa2016_HtsAddedJuly22_LongFormatJuly25' data is uploaded now.  The id numbers are 6000021253–6000028399.

gsrohde, 2016-11-10:
@NewcombMaria, @dlebauer I'm looking now at the fourth sheet, 'managements for sorghum season1 and sorghum season2'.  Most of this seems pretty straight-forward: the column headings date, mgmttype, level, units, and notes correspond exactly with like-named columns in the managements table of the database, and I assume the `citation_id` should point to the citation "Newcomb, Maria, 2016, Field measurements from Maricopa Agricultural Center” and that the date level-of-confidence value should be 5 (to the day).  

I am confused, however, by the data in columns I through L of rows 5 through 17.  Not only are the headings (in columns I through L of row 5) at variance with the database table column names, but there is no level specified, even though most of the units are for numeric values.

gsrohde, 2016-11-10:
@dlebauer The worksheet that opens from the _5th_ link is just a Readme sheet.  It's not clear what to do with this.

gsrohde, 2016-11-10:
@NewcombMaria, @dlebauer I'm about to upload the 6th item ("sorghum season2 emergence vigor 'Sorghum2_emergence_SubsetPlots_plusFinalCounts_Nov7.xlsx'").

I'm making the following (additional) two assumptions: (1) the word "Field" should be omitted from site names.  For example, "MAC Field Scanner Season 2 Field Range 3 Pass 3" should actually be just "MAC Field Scanner Season 2 Range 3 Pass 3"; (2) the "Name_corrected" column is to be matched to a cultivar name in determining the cultivar id.

Also, I am again ignoring several columns—namely, columns A through K.

gsrohde, 2016-11-10:
I've upload data from the 6th item.  Id numbers are 6000028400–6000028903.

gsrohde, 2016-11-11:
@NewcombMaria (and @dlebauer): I'm looking at the **_7th_** item, "season 2 emergence final stand counts" (file 'Sorghum2_emergence_SubsetPlots_plusFinalCounts_Nov7.xlsx - StandCounts_Final_AllPlots_long').

The only column heading that looks like it might be a trait variable name is "Final_stand_count", but there is no variable by this name.  Also, the dates are all in the wrong format.  I'm again assuming "Name_corrected" should be used for the cultivar name (= the genotype).  And I don't see any site names in this file.  I'm ignoring columns A–P, R, and S, and I'm assuming the Comments column should map to the traits.notes column.

gsrohde, 2016-11-11:
@NewcombMaria OK, I'm now looking at the last tab of 'DRAFT_Sorghum2_PlantHeights_AugustPlanting.xlsx' (the 5th item).  As with the 7th item, the dates are all in the wrong format, I'm assuming "Name_corrected" should be used for the cultivar name, and I don't see any site names in this file.  I assume "Canopy_ht_cm" should map to the trait variable "canopy_height" (which has "cm" for units).  There is no "panicle_height" variable, however, so I'm assuming either the "Panicle_ht_cm" column should be ignored or a "panicle_height" variable should be added (which?).  I'm assuming only columns N, R, S, U, and maybe T have any data that should be recorded.

gsrohde, 2016-11-11:
@dlebauer In https://docs.google.com/spreadsheets/d/1hGeZq2CWd-LakcuHZhGXiONFZrdPwWUTVJeDYxDP7ZY/edit#gid=680357011 (item 8), "Common_name" isn't one of the columns you highlighted, but the values in this column all match an existing cultivar.  Shall I use this column to set `cultivar_id` or leave it NULL?

gsrohde, 2016-11-11:
@dlebauer, @NewcombMaria: I've inserted the managements from spreadsheet 4 into the database.  I used a dateloc value of 5 (date accurate to the day).  There was only one management row ("planting (plants / m2)") in the table before I started, so the 33 others are the new ones.  Many of the level and units column values are blank.  It's possible that in some cases these can be populated from information in the notes column, but I didn't attempt this.

rachelshekar, 2016-11-30:
@dlebauer - what is the next step?

NewcombMaria, 2016-12-01:
@dlebauer @rachelshekar some spreadsheets need additional work.  I can work on this next week if it's clarified what the next steps are on my end (for example which spreadsheets need work and what revisions are needed).

dlebauer, 2016-12-01:
@NewcombMaria and @gsrohde I think that some of the confusion is that in editing existing comments it isn't clear who made what comment and email notifications aren't sent out. (I originally started doing this thinking it would be more clear to update an existing comment ... but obviously this is a catch)

IN any case, to answer a few questions:

* date is the one variable with a clear standard: YYYY-MM-DD
* @gsrohde so far your assumptions have been correct. I appreciate your careful attention to detail!
* I added the variable [panicle_height](https://terraref.ncsa.illinois.edu/bety/variables/6000000017); @gsrohde please feel free to do so in the future, based on definitions in the [definitions worksheet](https://docs.google.com/spreadsheets/d/1B30WB5vD-JAU2Pgluhdu1kIxwttdOcQphSIin7BByKg/edit#gid=2063141745)
 * [ ] @NewcombMaria please confirm that this is okay. Should the definition include the clause "When the panicle extends notably above canopy height"? This suggests that panicle height is only measured on this condition - is that the case? This and the other information specific to our field (e.g. center of four rows _could_ be recorded as a 'method'. But for the definition of the variable we should try to make it generic
* I've updated the remaining datasets from group 1 in https://github.com/terraref/reference-data/issues/61#issuecomment-259049732

Lets discuss offline.

dlebauer, 2016-12-02:
TODO

* [ ] run queries from col S here: https://docs.google.com/spreadsheets/d/1xIRwroYObD125I0TDCDp3YPIGPi2Qzeh-vlUCg_b85Q/edit#gid=879963447

gsrohde, 2016-12-02:
@dlebauer Regarding the "Multiple matching referents" error for cultivar "TX7000".  The problem is that in the Bulk Upload Wizard, matching is done case-insensitively, and there is also a cultivar whose name was entered as "Tx7000".  On the other hand, the database constraint that checks for duplicate cultivar names _is_ case sensitive.  Perhaps this is a sort of bug, and either the constraint should check for duplicates case insensitively or the wizard should require an exact match, including case.

In any case, we should remove one of these cultivar entries and reassign any references to the other.  I can do this if you tell me which one to remove.

By the way, you can find these cultivars easily from the Cultivar list page by using a little-known search feature: If you enter "name=TX7000" (without the quotes), you will get only the cultivar whose name is exactly TX7000; you won't get all the cultivars whose names contain this as a proper substring.  This feature is available for all table searches (but not "advanced search"), but this is the first time I have actually found it useful.  Since the search value must match the entire database column value _exactly_, it is best used for columns having values that are short strings.

dlebauer, 2016-12-02:
@gsrohde TX7000 is preferred ...

gsrohde, 2016-12-03:
OK, I deleted the Tx7000 cultivar and and reassigned all the traits referring to it to TX7000.

rachelshekar, 2017-01-25:
@NewcombMaria - can you please give us an update on this?  Are you waiting for anything from NCSA?

NewcombMaria, 2017-01-25:
Thanks for checking in on this.  It's simply an issue of finding/making time on my side.  I'll put it as high priority on task list.

gsrohde, 2017-03-15:
@NewcombMaria Do you have whatever information you need to upload remaining data?

gsrohde, 2017-04-03:
@NewcombMaria It is not yet the case that the cultivar is automatically deduced from the site information.  This is a planned feature but it has not yet been implemented.  (It is also planned to update existing data lacking cultivar information by using the well-defined sitename-->cultivar mapping; but again, this hasn't yet been done.)  If you want to expedite having cultivar information in the database, then yes, include cultivar names.  If you want to save yourself some work and it isn't so important to have the cultivar information in the database right away, then you can omit that information from the upload files.

Regarding issue 99, see my latest comment there.

dlebauer, 2017-04-03:
Scott do you have an ETA on automatic cultivar mapping?

gsrohde, 2017-04-04:
@dlebauer I'll try to do a little work on this tomorrow to see if I can get a sense of how involved this will be.

dlebauer, 2017-05-03:
@NewcombMaria could you provide a status update?

NewcombMaria, 2017-05-04:
When I've had time, I've worked on standardizing variables and cultivars and date formats.  Will prioritize making the May 2017 milestone.  Thanks

NewcombMaria, 2017-05-30:
@dlebauer and @gsrohde I just agreed to a hard deadline with @nshakoor  for uploading 2016 field measurement data sets by the end of next week (by June 11).  Between now and then I may have a few questions along the way.  I'll do my best to make rapid progress.

gsrohde, 2017-05-31:
@NewcombMaria Please try to get any questions that arise to me as soon as you can.  My availability is very unpredictable right now, so it's best if I have a larger window of time in which to respond.

rachelshekar, 2017-06-01:
see https://github.com/terraref/reference-data/issues/125 for yield data question

NewcombMaria, 2017-06-11:
@dlebauer I'm trying to use BETYdb Bulk Upload for the Sorghum_season2_nodetillercounts.csv (see https://drive.google.com/drive/folders/0B1OR5-zeXagnWjJaaWRlSENsVms ).  Have been getting Time-out '504 Gateway Time-out'. 
I've defined the variables and am using sites and names that we used earlier.  Any suggestions?

gsrohde, 2017-06-11:
@NewcombMaria I've found that I sometimes have had to break up a bulk-upload file into batches.  I was able to process  a 4900-row file by breaking it up into files with <= 1000 data rows in addition to the header line.  Your file isn't that much bigger (1431 data rows), but maybe that's enough to make a difference.  I suggest dividing the file into three files of <= 500 lines each.  You will have to repeat the header line in each file.

Does the time-out happen at the first step where validation happens?

NewcombMaria, 2017-06-11:
@gsrohde thank you!!  Your suggestion of dividing into 3 smaller files worked.  I had a few errors to work out after that but in the end was able to successfully upload the data in 3 batches.  I defined the access level as 'Internal EBI and collaborators'.  Is that correct?

NewcombMaria, 2017-06-11:
@gsrohde 'leaf_node_count' is defined as a variable, but it looks to me that there is not be any data associated with the variable yet.  Is that correct?  Is there a way to confirm if there is data already entered under a defined variable?  I'm going to try to upload Sorghum season1 data under the variable 'stem_elongated_internodes_number' because this is a term in the Sorghum Crop Ontology.  I don't think I'm duplicating season1 data but it would be good to be able to doublecheck that I'm not duplicating before I upload a new dataset.

NewcombMaria, 2017-06-11:
@gsrohde or @dlebauer how do I see a list of accepted cultivar names for Sorghum season1 for the whole field?

NewcombMaria, 2017-06-12:
@gsrohde or @dlebauer another question.  How do I add a different method to an existing variable?  I uploaded Sorghum season 2 yield data [aboveground_fresh_biomass (kg/ha), aboveground_dry_biomass (kg/ha), and aboveground_biomass_moisture (%)] and would like to add information on methods.  I'm not sure how to do that.

dlebauer, 2017-06-12:
List of cultivars names is in the cultivars table and methods are in the
methods table

NewcombMaria, 2017-06-12:
@dlebauer or @gsrohde in the cultivars table in the database, I'm finding it difficult to determine which ones are for Maricopa Sorghum season1.  Is there anything for Season1 similar to this spreadsheet for Season2?  https://docs.google.com/spreadsheets/d/1RZKk7zZ5uqGcKsJ6XRMEJEyG7hlA389PKi2sEDMLy74/edit#gid=222361549

To create a method in the methods table, the 'New Method' requires a Name and Description.  Is it ok to have two different Descriptions for the same Name?

NewcombMaria, 2017-06-12:
I was able to upload 6 data sets today from 2016 sorghum seasons 1 and 2.  The csv files are in this google docs directory:  https://drive.google.com/drive/folders/0B1OR5-zeXagnWjJaaWRlSENsVms .  There are still more 2016 data sets to upload, from spreadsheets that need a bit more more work.  I'll continue to work on it.

[41 more replies]

closed 2017-08-24 by rachelshekar