terraref/reference-data #63: Revise entity names

gsrohde, 2016-12-02:
This issue pertains to item one in this comment: https://github.com/terraref/reference-data/issues/61#issuecomment-264520078

gsrohde, 2016-12-05:
@dlebauer In https://docs.google.com/spreadsheets/d/1xIRwroYObD125I0TDCDp3YPIGPi2Qzeh-vlUCg_b85Q/edit#gid=879963447, it looks like the new entity names not only substitute W2, W, E, and E2 for a, b, c, and d (respectively), as I remember from our phone meeting with Maria; it looks like the Range and Pass numbers change as well.  Is this intended?  I just thought I'd check since I didn't remember discussion of this.

If the update statements really are what you intend, I'll just download the whole list and run them as a batch.

dlebauer, 2016-12-12:
@gsrohde I think the range and pass numbers will also change. I think that on a few of the uploads all of the range / pass specifications were identical, as a result of incorrectly specifynig the formula in google drive that I filled down. This will likely require re-uploading a few of the datasets since it will not be possible to differentiate among the sites when they all had the same name.

The rows / datasets in error can be identified by counting records by sitename and looking for an overabundance of sitenames with hundreds or thousands of records (probably names with 'Range 1 Pass 1 season 2" in them )

gsrohde, 2016-12-12:
@dlebauer After your first sentence, I'm not sure I follow.  I tried the following query of traits for sites in Arizona:
```
select sitename, count(*) from traits t join sites s on s.id=t.site_id where state = 'Arizona' group by sitename order by count(*) desc;
```
No site has more than 21 traits associated with it.

dlebauer, 2016-12-12:
Sorry I remembered incorrectly. The error was that I used column numbers
instead of pass numbers. The old names might have 16 columns. The new ones
should have 7 passes (the border rows on each side account for the two
additional columns)

gsrohde, 2016-12-12:
@dlebauer I guess the crucial question is whether you want me to check your proposed update statements based on some well-defined rule, or whether I should just take it on faith that you wrote them correctly.  If the former, perhaps looking at a plot map would help get a more solid picture of what the updates should be.

dlebauer, 2016-12-12:
@gsrohde The updates should be 
* update the column used to define pass number from col F to E
* Add the E / W designation for the subplots.
* All of Maria's spreadsheets should have the same set of plotnames. 

the plots are defined in the [Sorghum_2ndCrop_fieldmap](https://docs.google.com/spreadsheets/d/1xIRwroYObD125I0TDCDp3YPIGPi2Qzeh-vlUCg_b85Q/edit#gid=1432184448) worksheet of the same spreadsheet you referenced in https://github.com/terraref/reference-data/issues/63#issuecomment-264970968.

gsrohde, 2016-12-16:
@dlebauer Run this on the terraref database and see if the name changes look correct:
```
SELECT
    name,
    'Season 2 ' || (CASE substring(name FROM 'subplot (a|b|c|d)$')
         WHEN 'a' THEN
              regexp_replace(name, 'subplot a$', 'subplot W2')
         WHEN 'b' THEN
              regexp_replace(name, 'subplot b$', 'subplot W')
         WHEN 'c' THEN
              regexp_replace(name, 'subplot c$', 'subplot E')
         WHEN 'd' THEN
              regexp_replace(name, 'subplot d$', 'subplot E2')
         END) AS "new name"
FROM
    entities
WHERE
    name ~ 'subplot (a|b|c|d)$';
```
I'm assuming sitenames containing 'subplot a', 'subplot b', etc. should change even if they don't happen to be mentioned in the spreadsheet https://docs.google.com/spreadsheets/d/1xIRwroYObD125I0TDCDp3YPIGPi2Qzeh-vlUCg_b85Q/edit#gid=879963447.

If these name changes look correct, I'll run the following update:
```
UPDATE entities
    SET name =  ('Season 2 ' || 
        (CASE substring(name FROM 'subplot (a|b|c|d)$')
         WHEN 'a' THEN
              regexp_replace(name, 'subplot a$', 'subplot W2')
         WHEN 'b' THEN
              regexp_replace(name, 'subplot b$', 'subplot W')
         WHEN 'c' THEN
              regexp_replace(name, 'subplot c$', 'subplot E')
         WHEN 'd' THEN
              regexp_replace(name, 'subplot d$', 'subplot E2')
        END)
WHERE
    name ~ 'subplot (a|b|c|d)$';
```

dlebauer, 2017-01-12:
The changes look correct ... but the second query threw error 

> [Err] ERROR:  syntax error at or near "WHERE"
LINE 13: WHERE

gsrohde, 2017-01-12:
done

closed 2017-01-12 by gsrohde