terraref/reference-data #64: Early sorghum-season 2 metadata corrections until about Aug 23

NewcombMaria, 2016-12-04:
Metadata for the first scans after the August 3rd planting needs to be checked up until approximately August 23rd, for 'sorghum season 2' entry and 'August 3' planting date entry, and for the correct name for the Stereo-Vis scans (sorghum2).

dlebauer, 2017-01-12:
@NewcombMaria can you please clarify? What exactly do we need to check?

smarshall-bmr, 2017-01-13:
@dlebauer The "user_given_metadata" field was not updated until 20 days into the sorghum 2 experiment so it contains values from the sorghum 1 before that.

craig-willis, 2017-02-01:
There are two separate issues here.  First, we need to get the XML files under SCM so we can track and review changes going forward.  Second, we need to define an approach similar to how we're handling the sensor-fixed metadata and find the best place to store the experiment information that's in the user_given_metadata.

Digging through the files on Roger, there were several fields changed on different dates.

Original values:
* "campaign": "testing",
* "date of sowing": "2016-04-19"
* "date of emergence": "2016-04-25"
* "experiment title": "Sorghum field experiment"
* "mission or scan": "3d_scan_4m"

Updated values (date changed)
* "campaign": "seedling emergence and early vigor" (~8/8/2016)
* "date of sowing": "2016-08-03" (~8/25/2016)
* "date of emergence": "2016-08-08"  (~8/25/2016)
* "experiment title": "Sorghum field experiment 2" (~8/25/2016)
*  "mission or scan": "StereoVIS_IR_Sensors_emergence full plot full Field 0.33m"  (~8/8/2016)

dlebauer, 2017-02-01:
Within BETYdb, we can store this information thus:

* in the managements table: 
  * date of sowing
* in the traits table
  * date of emergence (though for scientific inference we need to know days observed before emergence)
* experiment title: experiments table

This information, as well as the mission or scan could be inserted into the geostreams database as well. The 'sensor' would be the entire field / field scanner. @max-zilla or @robkooper do you have suggestions for how to structure this?

craig-willis, 2017-02-01:
Discussing with @max-zilla two points came up:
* We might still want to link the LemnaTec metadata to this information if stored elsewhere.  This could be by URI or identifier.
* Is there a need for any of this to be available for search? For instance, will users want to search by experiment or date of sowing, etc?

rachelshekar, 2017-02-23:
@craig-willis - please update

craig-willis, 2017-03-31:
Update on this ticket:
* We have identified the incorrect values and dates in the LemnaTec metadata, but have decided to leave the raw metadata as-is.
* Experiment metadata including dates, missions/scans, etc should all be tracked in BETYdb. The LemnaTec metadata will not be authoritative for this information.
* We will be working on a metadata extractor that generates curated metadata fields used by extractors and search.

closed 2017-05-17 by rachelshekar