terraref/reference-data #65: Unique identifier for germplasm

dlebauer, 2016-12-07:
### Issue

We need to uniquely identify germplasm. Currently these are stored in the 'cultivars.name' field of BETYdb, but this is not sufficient and not consistently used within the TERRA program.

This is something that DivSeek and AgBioData groups are working on.

### Requirements

* [ ] Get input from Noah Fahlgren, Geoff Morris, Mike Gore, Carolyn Lawrence-Dill, and others
* [ ] Identify one or more unique identifiers for germplasm that we can adopt right now
* [ ] Determine how to link data in BETYdb and a breeding database (like BMS or other that supports BRAPI) with these unique identifiers. In BETYdb, the unique identifier can be used in the cultivars.name field
* [ ] Create plan for providing a canonical reference to genotypes for all TERRA teams
* [ ] Create new issues related to implementation within the TERRA program

### See also 

Summary of discussions at Plant and Animal Genome meeting from [DivSeek news](http://www.divseek.org/news/2016/2/12/divseek-hosts-roundtable-discussion-in-san-diego)

> Enabling gene banks to use genomic and phenotypic data for better conservation and use of germplasm is central to DivSeek, and having permanent unique identifiers (PUIs) to identify germplasm from which data has been derived would be a major step forward. The workshop discussed why PUIs are essential and considered metadata that need to be associated with PUIs, so that genebank managers can track accessions within and across genebanks and researchers can be assured they are working with and using the same material (presentation available here). 

Comments on need for genome PUID by Ethalinda Cannon on the AgBioData thread:

> DivSeek has been looking at permanent unique identifiers (PUID) to identify germplasm from which data has been derived, so that genebank managers can track  accessions within and across genebanks and researchers can be assured they are working with and using the same material. 

> Although there are a number of solutions out there including GRIN, not everyone uses the same one and in some cases there no ID is used at all.  Therefore DivSeek is interested in a PUID that could be used by all. I should stress this PUID would not replace any system that a genebank or institution already uses, the PUID would be in addition to this. The International Treaty on Plant Genetic Resources for Food and Agriculture (ITPGRFA) undertook a consultation in 2015 on this topic and this concluded that DOIs would be the best form of PUIDs. So there is now a test project in this area funded by the ITPGRFA and some of the discussions relating to PUIDs and this project where presented at the DivSsek meeting in January. The presentation that was given on PUIDs and an associated news item can be found on the DivSeek website http://www.divseek.org/news/  

* MOWG used by AgBioData https://github.com/MOWG/Metadata
* MOWG MiXS comparison https://docs.google.com/spreadsheets/d/1KgXi0w5_nZeobMHp-prj-VjV775y9eEMwxdJdSD1xCs/edit#gid=1300889300

rachelshekar, 2016-12-15:
@nshakoor to share with groups and ensure that the correct identifiers are used.

dlebauer, 2016-12-15:
@nshakoor please review this list: https://terraref.ncsa.illinois.edu/bety/api/beta/cultivars?key=

NewcombMaria, 2017-01-31:
The 16PR looks very much like a seed source code for the year and location where the seed was harvested.  16 = 2016 and PR is probably Puerto Rico, with a few 15WF which would be 2015 and a different location.  I think that the Pedigree would be more informative as part of the name, and also the generation of the RILs (F4 I believe, for season 1??).  Seed source can be an important variable perhaps, but not as a name.

NewcombMaria, 2017-02-22:
@dlebauer what are the cultivar identity names that are being used for genotypic data for Sorghum season 1?  I am currently doing QC on the yield data spreadsheet and associated definitions and protocols and need to know the correct cultivar id.  We should be consistent and use the identity names for phenotypic and genotypic data.  I noted above that the betydb cultivar names for Sorghum season 1 are incorrect - they are seed source codes and not F4 RIL identities.

dlebauer, 2017-02-22:
@NewcombMaria you are correct that the names for Season 1 are in the 'name' field here: https://docs.google.com/spreadsheets/d/1s-kQc3K0h4-hfwMzKOU_zpRXN5loByIpKW8Rvw76mh0/edit#gid=796817704, which derives from the 'source' field.

To be clear, are you pointing out two different issues, 1) that we should not use source as name and 2) the source values are incorrect, or are these the same issues (if we combine parents into a name that would be sufficient).

What would you propose for a unique identifier?

rickw-ward, 2017-02-22:
@dlebauer and @NewcombMaria - sorry, I accidentally eliminated assignees, but believe I restored it.     I am visiting Geoff Morris (@gpmorrisksu) tomorrow at KSU and will discuss this matter and report back.

NewcombMaria, 2017-02-22:
@dlebauer the problem is that the source should not be used as a identifier name.  Rick and Geoff will be the best experts to propose a unique identifier.  One option is to concatenate 'Pedigree' and 'Pedigree2', for example RIL-CS1_(Tx2910/(Macia/R07007)-CS44)-CSF1-PRF2-CS.  This would include the RIL# and the cross and would be a unique identity.  Rick and Geoff should provide their suggestions and I'll agree with their perspectives.

rickw-ward, 2017-02-27:
if we are adopting the Integrated Breeding Platform at KSU and UA, we should look there for guidance.    More to come...

dlebauer, 2017-03-06:
@rickw-ward any updates?

dlebauer, 2017-03-06:
@NewcombMaria I updated the names using the sql statements here: https://docs.google.com/spreadsheets/d/1s-kQc3K0h4-hfwMzKOU_zpRXN5loByIpKW8Rvw76mh0/edit#gid=796817704

dlebauer, 2017-03-06:
@rickw-ward and @gpmorrisksu please review the genotypes here: https://terraref.ncsa.illinois.edu/bety/cultivars ([in csv form](https://terraref.ncsa.illinois.edu/bety/cultivars)) and confirm that the names are sensible and confirm:

1. will allow us to uniquely identify the lines
2. will enable cross-referencing of genomic data

Any additional suggestions appreciated

rickw-ward, 2017-03-07:
@dlebauer   and @NewcombMaria your format works, backfitting to include unique IDS (GID) and separating the "RIL" part as Designation, from parents etc. as Cross, will be in compliance with the Integrated Breding Platform https://www.integratedbreeding.net/ .  This file contains suggestions and some background:  [germplasm naming.xlsx](https://github.com/terraref/reference-data/files/823031/germplasm.naming.xlsx)

"List" is a set of records of germplasm accessions that exist in storage, or were used in an experiment, were genotyped, etc.  Each list has what seems to be metadata regarding owner, purpose, etc.  The key element for cross purpose usage is the GID.  Materials from breeding programs are named on the basis of the GIDs of their parents, and the breeding methods that link the new GID and the parents.

rickw-ward, 2017-03-07:
eventual back-fitting; that's what i meant.   what you have is good.

dlebauer, 2017-08-11:
@NewcombMaria or @gpmorrisksu here is a case where we have three records that appear to be from the same cultivar: 

[image]

https://terraref.ncsa.illinois.edu/bety/cultivars?utf8=%E2%9C%93&DataTables_Table_0_length=25&search=ATLAS&direction=&sort=

Currently we only have trait data for the first named PI586537. Is this the appropriate approach? Interestingly, the Ecotype and notes all differ ... is this sweet, forage, or BAP?

zhenbinHU, 2017-08-11:
Same problem for cultivar: LEOTI
[image]

dlebauer, 2017-08-11:
I've merged these and deleted the records that don't use PI as the name.

NewcombMaria, 2017-08-14:
@dlebauer ATLAS (common name) planted in Maricopa in 2016 and 2017 refers to different PI identities for each year.  Sorghum season4 (2017) 'ATLAS' is PI641807.   See Issue #126 comment from May 26: 

[The only changes were LEOTI (old name) to PI_641825 (new name) and ATLAS (old name) to PI_641807 (new name). I changed both entries to BAP in the ecotype column. We also had 'ATLAS' in season1 but that ATLAS was PI_586537 (season4 ATLAS is PI_641807). I found this confusing so I checked in with the Clemson group who provided the seed and received this response: "Hello again Maria: PI_586537 and PI_641807 both represent Atlas. They are just different seed lots that represent Atlas."]

dlebauer, 2020-02-13:
germplasm have been linked to ncbitaxon and genesys id; information is stored in the terraref.org/bety 'attributes' table and available at the BrAPI endpoint

 ... see https://terraref.org/brapi/v1/germplasm?commonCropName=sorghum

closed 2020-02-13 by dlebauer