terraref/reference-data #66: Create form where users can apply to the early access program

dlebauer, 2016-12-07:
Collect information about users who are interested, 

### Fields: 

* name 
* affiliation
* role
* preferred user name
* use cases /, applications
  * what data are you most interested in?
  * by sensor (hyperspectral, 3D, RGB) 
  * traits
  * 
* background / expertise (click one or more)
  * breeding
  * crop science
  * plant physiology
  * genomics
  * computer vision
  * machine learning
  * robotics
  * other (please provide)
* preferred software (check all that apply)
  * microsoft excel
  * R
  * python
  * C/C++
  * Java
  * matlab
  * other (please provide)
* comment box
* level of interest in data standardization   
* what else? propose additional fields below or just put in the draft

### Information and links

* overview of project
* expectations
 * 
* links to terraref.org
* summary of data use policy and link to official policy

## completion criteria

* [ ] determine information to collect
* [ ] create google form 
  * [ ] create draft with above fields
  * [ ] get feedback on content from group
* [ ] announce via mailing list, website, and twitter

dlebauer, 2016-12-07:
Here is some text based on a response that I recently sent in response to an inquiry:

> It is possible for you to access our data. The terms of use are here: https://terraref.gitbooks.io/terraref-documentation/content/user/data_release_policy.html Please direct any legal questions to Erica Fischel.

> There is no commitment from alpha release users beyond what is defined in our data release policy (briefly, attribute the TERRA REF team and don't scoop us on publication). However, the goal of the early release is to get feedback from end users - this includes all aspects of the computing pipeline, what works / doesn't work with respect to data access, how we might improve / extend our catalog of data products, any errors or inconsistencies that are found. We will try to make it easy for people to provide feedback as well as send out surveys. This will be optional but highly encouraged.

> We are also looking for commercial partners to engage in a variety of ways, for example: 
> * easiest: if possible allowing us to brag about the fact that X is using our data in proposals, publications, and / or public outreach
> * providing public access to data via a user-friendly UI and / or robust API 
> * contributions and extensions to our open source computing infrastructure
>  * also input and support for emerging standards and conventions to support interoperability
>   * other web services that we can leverage
> * deploying algorithms on our infrastructure (these can be proprietary if the derived phenotypes can be added to the pipeline
> * other contributions 
> * anything else you can think of

rachelshekar, 2016-12-09:
@dlebauer First draft:
https://docs.google.com/forms/d/1YCKF9JUn7Yur3UzoqbtZ9wqLs1LywvXvS4jhlZKG5yQ/edit?usp=sharing

dlebauer, 2016-12-13:
@rachelshekar I added a lot of questions. It is way too long and I'll probably revise again, and want to get feedback from others on the team, but feel free to take a look.

rachelshekar, 2016-12-15:
@dlebauer 
This questionnaire is user friendly.  
Not sure about difference between interface vs. access questions.
Lets get feedback from the team on most important questions.  We could do follow-up questionnaires for the least important.

rachelshekar, 2016-12-15:
Moved alpha tester questions to new form https://docs.google.com/forms/d/1KG65Ai0WiS9yWAK56ChHFAf3bYgdNwObkUb2JPSr_QM/edit

rachelshekar, 2016-12-15:
Under
Check any of the following interfaces that you would be likely to use to get data?
What does "just the data" mean?

dlebauer, 2016-12-15:
I see 'just the files' ... that is what we currently offer.

rachelshekar, 2017-01-10:
Who is this survey for?  Potential users?  If so, I think you should leave out questions 11 and 12.  Asking them what they think of the interfaces TERRA offers and if they will likely use them (and why/why not) could be a follow-up questionnaire.  ie those who answer yes to "Are you interested in cloud development environments?"

dlebauer, 2017-01-10:
Survey is for people to express interest in the data / apply for early
access.

Good point. I'll clean it up.

dlebauer, 2017-02-02:
I think this is complete. The form is here: https://goo.gl/forms/PdnV3K16oXqb3iBI2

closed 2017-02-02 by dlebauer