terraref/reference-data #68: How to share genomic data and pipeline?

dlebauer, 2016-12-15:
Currently the genomic data is on Roger. 

It needs appropriate metadata and registration in Clowder
* [ ] needs to link to names in cultivars table https://terraref.ncsa.illinois.edu/bety/api/beta/cultivars?key=
* [ ] link to code used in the pipeline - can we put this in a repository under github.com/terraref?
* [ ] update documentation https://terraref.gitbooks.io/terraref-documentation/content/products/genomics-data.html so that it is reproducible (and links to code)
* [ ] what more do we need with respect to a genomics database?

phloz, 2017-01-26:
A HTcondor Dag workflow was created at the Danforth Center to run the genomics pipeline that was outlined here: https://terraref.gitbooks.io/terraref-documentation/content/genomic-data-product.html the Dag is unique to HTcondor and the paths within the Dag and the underlying jobfiles are unique to our file system so I'm not sure they useful to include.  Using the software and following the instructions listed with allow someone to reproduce the processed data (vcfs) listed on  https://terraref.gitbooks.io/terraref-documentation/content/products/genomics-data.html 
Wouldn't a link to the genomic-data-product.html page on the genomics-data.html be sufficient?

Noah and I will be in contact with CoGe to move forward with integrating their tools and the genomics pipeline.

dlebauer, 2017-01-26:
> Wouldn't a link to the genomic-data-product.html page on the genomics-data.html be sufficient?

CoGe provides an API as well as a nice interface for tracing provenance & enabling reproducibility, evaluating and comparing genomes, storing and sharing data, and more.

nfahlgren, 2017-01-26:
@dlebauer as we have discussed though, CoGe doesn't do all the things we need.

dlebauer, 2017-01-26:
From Todds email

> David and I have both received commitments from Eric Lyons and others at CyVerse that they will continue to improve CoGe and their infrastructure to enable TERRA-REF’s needs. 

Last week @mbomhoff said that he can implement the pipeline described in the documentation.

Even then, there can be multiple parallel pipelines. The one you use for a particular analysis does not have to be the same one that we provide as part of the public pipeline.

phloz, 2017-01-26:
@dlebauer I was addressing issues based what's currently available.  As I mentioned at the end of my earlier comment, @nfahlgren and I are going to work with CoGe.  I was able to get the CoGe API working, using their available tools with help from @mbomhoff.  The plan is to work with CoGe to implement a pipeline similar to what's already in use at DDPSC and obtain an outline of their resources so we don't overload their system when using the API.

nfahlgren, 2017-02-05:
@dlebauer @phloz @craig-willis all of the raw sequencing data has been uploaded to CyVerse:

```
/iplant/home/shared/terraref/sites/hudson-alpha/raw_data/year1
/iplant/home/shared/terraref/sites/hudson-alpha/raw_data/year2
```

I need to clean up a few things and put our data release policy in the same directories. Then I think we can make it public.

dlebauer, 2017-06-08:
@nfahlgren is waiting for output data from Mockler's group

@dlebauer needs to update cultivars table to remove space following `PI_` names

nfahlgren, 2017-06-08:
@dlebauer I confirmed with @phloz that the genotype IDs in the processed genomics data will be named `PIxxxx`

dlebauer, 2017-06-13:
@NewcombMaria and @nfahlgren et al ... I've updated the cultivar names from `PI_*` --> `PI*`

dlebauer, 2020-02-13:
The pipeline is well documented at docs.terraref.org and also available in COGE

closed 2020-02-13 by dlebauer