terraref/reference-data #69: Review alpha user sign up form

dlebauer, 2017-01-05:
Please review 
Please fill out the “alpha user” signup and provide feedback https://goo.gl/forms/7dpyUVtF4ZSl6iwy2

craig-willis, 2017-01-10:
A few comments:
* Consider linking to the data release policy on that first page
* Initial description could be clearer (set expectations for what's to come in the form -- I wasn't expecting the questionnaire).
* Better instructions for each question: why do you need my github username? Why do you need the data access information?
* Link to data products documentation, I might not know what I plan to use.
* A few minor typos (Whereever, "How do you want access our data?")

On a larger note, I think it might make sense to separate sign-up from survey/questionnaire.  Only include questions in the sign-up that will be used to approve.  Send a follow on questionnaire for information that would help improve the system. That said, maybe you'll get a higher response rate if you force the questions during sign up (or maybe fewer alpha users).

rachelshekar, 2017-01-11:
I agree with Craig.  I can make updates tomorrow after we meet, @dlebauer

closed 2017-01-25 by rachelshekar