terraref/reference-data #70: Top 3D Scanner Misalignment

solmazhajmohammadi, 2017-01-05:
Physical position of the scanner has been changed (due to temperature change or wheel replacement). this  caused the misalignment between the heading_west and heading east pointclouds. 
Applying following transformation to the "Top-heading-west" pointcloud will fix it.

New offset:[-13  +50  0] 

I will write a script to compensate the offset.

dlebauer, 2017-01-05:
@craig-willis is the sensor offset available in the meta-data, and if so, can we (should we) programmatically access it with related extractors? Can we assign dates for which this information is valid? What would be an appropriate protocol for updating metadata?

@solmazhajmohammadi can you work with Craig to generalize this solution?

solmazhajmohammadi, 2017-01-05:
Sure,  
It has not been added to metadata yet. I'll ask Markus to add it to the metadata.
I am looking at the data to see when the misalignment occurred.

dlebauer, 2017-01-05:
@solmazhajmohammadi I am not sure if the offset _needs_ to be in the file metadata. Even though the "sensor fixed metadata" is written with each image capture, I believe it should be stored somewhere like https://github.com/terraref/sensor-metadata/blob/master/sensors/scanner3D/sensor_fixed_metadata.json or in the geostreams sensor database; @craig-willis please advise

craig-willis, 2017-01-06:
@solmazhajmohammadi @dlebauer 
I was under the assumption that the "sensor_fixed_metadata" written by LemnaTec would no longer be updated and that https://github.com/terraref/sensor-metadata repository will become the authoritative location for changes (which will be POSTed to the Clowder API). 

Maintaining this information in two places will be complicated.

solmazhajmohammadi, 2017-01-10:
@craig-willis We haven’t added the offset to the metadata. you can update the sensor-metadata repository. 
@max-zilla I was not sure that if the code should go under the reference-data or computing pipeline, I uploaded the code here for now:
https://github.com/solmazhajmohammadi/3D_Misalignment

craig-willis, 2017-01-12:
@solmazhajmohammadi  Sorry for the delay.  I've created ticket #74 to track my part of this. I'll let you know when complete.

craig-willis, 2017-01-13:
@solmazhajmohammadi  Do you know the effective date of the change so I can add to the metadata?

solmazhajmohammadi, 2017-01-20:
@craig-willis The effective date of the misalignment is November 22, 2016
Please update the offset to the following transformation, it fits better for the full row scan
[X Y Z]=[ -10  +42  0]

craig-willis, 2017-01-20:
Thank you, @solmazhajmohammadi. Because "offset" is a little ambiguous, can you think of a more descriptive label for this -- something like "position correction offset"?

solmazhajmohammadi, 2017-01-20:
"position correction offset" is OK. Here is some options:
- alignment correction offset
- post-calibration transformation

rachelshekar, 2017-02-23:
@solmazhajmohammadi - can this issue be closed?

solmazhajmohammadi, 2017-02-27:
No, we still need to apply the transformation to the dataset. 
The offset is hardcoded now, @craig-willis What is the label for this offset?

rachelshekar, 2017-04-04:
@craig-willis and @solmazhajmohammadi - has this been completed?

solmazhajmohammadi, 2017-04-04:
Final step depends on [Issue 44](https://github.com/terraref/reference-data/issues/44)

rachelshekar, 2017-06-01:
scans being done next week to determine offset dependence on temperature; moving milestone

solmazhajmohammadi, 2017-07-13:
The result is posted in #74

closed 2017-07-13 by solmazhajmohammadi