terraref/reference-data #72: Add hyperspectral indices to BETYdb variables table

dlebauer, 2017-01-06:
There is a list of indices we are computing here https://github.com/terraref/documentation/blob/master/spectral_indices.md

These need to be added to the variables table in BETYdb at betydb.org and copied to terraref.ncsa.illinois.edu/bety

dlebauer, 2017-01-06:
Any thoughts or suggestions for defining indices @ashiklom, @serbinsh, @remotesensinglab ?

ashiklom, 2017-01-06:
Personally, I'd rather see the raw data, and then have SQL and/or R functions for calculating the spectral indices from that. But I'd like to hear what others think.

dlebauer, 2017-01-06:
@ashiklom to be clear, we are planning to provide 1) the calibrated hyperspectral data cubes as well as a 2) data cube with the same dimensions and pre-computed indices and 3) plot level means within (the [terraref instance of](https://terraref.ncsa.illinois.edu/bety) BETYdb. The reason we are doing this is to provide the data that diverse users want (e.g. the plot level means are the primary input for genomic analysis, who primarily request 'trait vectors' for each plot or genotype).

dlebauer, 2017-01-06:
@ashiklom in addition, as with other parts of the PEcAn workflow, we are using the variables table to define variables in the traits table as well as variables found in external files (e.g. met and flux tower). In this case, the naming would be consistent for the high resolution raster data in netCDF files and the plot level means in the traits table.

ashiklom, 2017-01-06:
Fair enough. I do think we should distinguish between raw and derived data, and have a system for linking derived data like spectral indices (or eventually our RTM inversions, and the outputs of the other PEcAn modules like allometry and photosynthesis) to their original data sources to make sure they aren't treated as independent data points.

Incidentally, I'm excited to hear NetCDF used in the context of hyperspectral data -- AFAIK, this isn't common practice, even though I think the format lends itself exceptionally well to hyperspectral imagery.  I was planning on using NetCDF for storing Landsat/MODIS data once I start working on it for my dissertation, so we should try to coordinate so we're using similar conventions to maximize interoperability.

dlebauer, 2017-01-06:
Thanks for your feedback @ashiklom 

> we should distinguish between raw and derived data, and have a system for linking derived data ... to their original data

We've discussed this extensively within the TERRA REF team as well as some within the PEcAn team. We could probably do queries that link data based on location, time, date, method, etc. or we could extend the data model to explicitly track links. From the clowder side, the database is sufficiently flexible that we could probably easily track which records in BETYdb were generated from a particular dataset. BETYdb could also point to datasets (and algorithms) used to generate each record, but that might require modifying the schema of BETYdb. @robkooper and / or @max-zilla may have thoughts on this.

> Incidentally, I'm excited to hear NetCDF used in the context of hyperspectral data 

me too ...

>  we should try to coordinate so we're using similar conventions to maximize interoperability.

Yes, this is the plan (and why I keep pinging you and Shawn) ... to coordinate development of shared data standards across different communities (starting with PEcAn and TERRA REF) using hyperspectral and other data.

rachelshekar, 2017-01-25:
@dlebauer - are you adding these to BETY or is Scott?

dlebauer, 2017-02-02:
@gsrohde could you please add these to the variables table?

gsrohde, 2017-02-03:
It's not clear to me how the columns in https://github.com/terraref/documentation/blob/master/spectral_indices.md should map to the database table columns.  I'm guessing something like
```
variables.name <--- Reflectance index
variables.description <--- Reflectance index
variables.notes <--- Acronym + Equation + References
variables.units <--- [empty string] (?)
variables.min <--- NULL (or 0?)
variables.max <--- NULL (?)
variables.standard_name <--- ???
variables.standard_units <--- ???
variables.label <--- ???
variables.type <--- ???
```
Possibly Acronym could be used for the name, but keep in mind that we had talked about making names unique (though we don't yet enforce this).

Also, I haven't yet implemented https://github.com/PecanProject/bety/issues/459, so if these are to be used for trait inserts, I will either have to implement that first or (temporarily?) add the new variables to the `trait_covariate_associations` table.

dlebauer, 2017-02-06:
* Variable name --> Acronym
* standard_name -->  should match the long name from the script written by @hmb1 here https://github.com/terraref/computing-pipeline/blob/master/scripts/hyperspectral/hyperspectral_indices_make.nco except 
  * all lowercase
  * convert spaces to underscore: https://github.com/terraref/computing-pipeline/issues/183#issuecomment-265512121
  * where 
* variables.description <--- Reflectance index + Equation + References
* variables.notes <--- Reflectance Index + Acronym
* variables.units <--- if the equation represents a ratio, use 'ratio', otherwise, use empty string or similar
* variables.min <--- if ratio, min = 0 else NULL 
* variables.max <--- NULL (?)
* variables.label <--- 'Reflectance Index'
* variables.type <--- 'trait'

dlebauer, 2017-02-06:
@gsrohde please match these indices to this database: http://www.indexdatabase.de/db/i.php and where there are differences, use the name / abbrev. from the database

gsrohde, 2017-02-06:
@dlebauer Do you mean: Any discrepancies between _Reflectance Index_ and _Acronym_ in https://github.com/terraref/documentation/blob/master/spectral_indices.md and _Name_ and _Abbrev._ (respectively) in  http://www.indexdatabase.de/db/i.php should be resolved in favor of the latter?

By the way, I _did_ check Reflectance index names from the table against the long names in https://github.com/terraref/computing-pipeline/blob/master/scripts/hyperspectral/hyperspectral_indices_make.nco and I think there are typos in lines 149 and 168 of the script.

dlebauer, 2017-02-06:
Yes please these fields in our database consistent with the fields in the index database

gsrohde, 2017-02-07:
@dlebauer It's a bit confusing to me in many cases how these match up.  It seems that looking at the formulas is often the best way to determine the best match.  Does the formula
```
NDVI=(R900-R680)/(R900+R680)
```
in the markdown table correspond with the formula
```
900nm - 680nm
-----------------
900nm + 680nm
```
in the Index Database table?  If so, that implies `NDVI` should be renamed to `ND900/680`.

Also, note that abbreviations in the Index Database table are not unique.  For example, `CI` is used both for _Coloration Index_ and for _Curvature Index_.  This is only a problem I suppose if we are planning to use in _our_ database two different variables that happen to have the same abbreviation.

dlebauer, 2017-02-07:
Yes, you are correct about the formulas being equal ... the numbers indicate the reflectance at wavelength x (R = reflectance, nm = units in which the wavelength is measured). So as long as the formulae are equivalent the meaning is the same

In the indexdatabase.de, other than indices with abbreviation = empty string (all of these appear to be minerals like Silica, Dolomite, etc). The remaining nine non-unique abbreviations are:   

|name                                                                                                              |abbreviation |formula                                                        |variables                       |
|:-----------------------------------------------------------------------------------------------------------------|:------------|:--------------------------------------------------------------|:-------------------------------|
|Cellulose Absorption Index                                                                                        |CAI          |100&InvisibleTimes;(0.5&InvisibleTimes;(2030nm+2210nm)-2100nm) |                                |
|Cellulose absorption index 2                                                                                      |CAI          |0,5&InvisibleTimes;(2020nm+2220nm)-2100nm                      |                                |
|Coloration Index                                                                                                  |CI           |RED-BLUERED                                                    |                                |
|Curvative Index                                                                                                   |CI           |675nm&InvisibleTimes;690nm683nm2                               |                                |
|Enhanced Vegetation Index 2                                                                                       |EVI2         |2.4&InvisibleTimes;NIR-REDNIR+RED+1                            |                                |
|Enhanced Vegetation Index 2 -2                                                                                    |EVI2         |2.5&InvisibleTimes;NIR<U+0096>REDNIR+2.4&InvisibleTimes;RED+1  |                                |
|Green Normalized Difference Vegetation Index                                                                      |GNDVI        |NIR-[540:570]NIR+[540:570]                                     |                                |
|Normalized Difference NIR/Green Green NDVI                                                                        |GNDVI        |NIR-GREENNIR+GREEN                                             |                                |
|Normalized Difference 819/1600 NDII                                                                               |NDII         |819nm-1600nm819nm+1600nm                                       |                                |
|Normalized Difference 850/1650 Normalized Difference Infrared Index                                               |NDII         |850nm-1650nm850nm+1650nm                                       |                                |
|Normalized Difference 790/720 Normalized difference red edge index                                                |NDRE         |790nm-720nm790nm+720nm                                         |                                |
|Normalized Difference NIR/Rededge Normalized Difference Red-Edge                                                  |NDRE         |NIR-rededgeNIR+rededge                                         |                                |
|Normalized Difference MIR/NIR Normalized Difference Vegetation Index (in case of strong atmospheric disturbances) |NDVI         |MIR-NIRMIR+NIR                                                 |MIR=[1300:3000],NIR=[800;10;10] |
|Normalized Difference NIR/Red Normalized Difference Vegetation Index, Calibrated NDVI - CDVI                      |NDVI         |NIR-REDNIR+RED                                                 |RED=[670;50;30],NIR=[800;10;10] |
|RDVI                                                                                                              |RDVI         |800nm-670nm(800nm+670nm)0.5                                    |                                |
|Renormalized Difference Vegetation Index                                                                          |RDVI         |800nm-670nm800nm+670nm                                         |                                |
|Red-Edge Stress Vegetation Index                                                                                  |RVSI         |718nm+748nm2-733nm                                             |                                |
|RVSI                                                                                                              |RVSI         |714nm+752nm2-733nm                                             |                                |

In all of the cases where we are using the hyperspectral imager, we should use indices that have numeric values. For normalized difference indices like NDVI using ND[wavelength][wavelength] seems like a reasonable convention. The two variables with abbreviation 'NDVI' in this table are defined with large parts of the light spectrum (and the first shuld probably be called 'MIR/NIR'. By contrast, RVSI has two distinct definitions. And NDRE has one that is defined with wavelengths and another defined generally (NIR, rededge don't correspond to wavelengths). 

After a quick glance, I think NDRE is the only ambiguous case. In this case, lets call it `NDRE` and ask the database admins what they recommend for a longer term solution (ideally they would provide unique ids other than the numeric column).

dlebauer, 2017-02-07:
feedback sent http://www.indexdatabase.de/info/feedback/

gsrohde, 2017-02-08:
@dlebauer Here is the original table (from https://github.com/terraref/documentation/blob/master/spectral_indices.md) followed by a corrected version.  See if my corrections look OK.  Or, if you don't have time now, I'll just use this now to do the database insertions and we can go back later and make corrections if needed.

I added a Units column, which either has "ratio" or blank.  The line numbers in the Notes column refer to the table at http://www.indexdatabase.de/db/i.php in default sort order (by name).

Also, I didn't change the Equation column, even when the acronym, which appears to the left of the equals sign, was changed.  I thought changing
```
NDVI=(R900-R680)/(R900+R680)
```
to
```
ND900/680=(R900-R680)/(R900+R680)
```
might be confusing, since the left-hand side of this looks like a division.  Perhaps something like
```
ND900/680 := (R900-R680)/(R900+R680)
```
or
```
Formula: (R900-R680)/(R900+R680)
```
would be acceptable.  In general, the spacing in the equations should probably be cleaned up a bit to make it consistent but I haven't yet done this.

Also, please verify that the `0.5` in the equation for RDVI is supposed to be an exponent (see line 256 of the Index DataBase table).

```
| Reflectance index                                        | Acronym     | Equation                                          | References                       |
|----------------------------------------------------------|-------------|---------------------------------------------------|----------------------------------|
| Normalized Difference Vegetation Index                   | NDVI        | NDVI=(R900-R680)/(R900+R680)                      | Rouse et al. (1973)              |
| Simple ratio                                             | SR          | SR=R900/R680                                      | Rouse et al. (1973)              |
| Optimized Soil-Adjusted Vegetation index                 | OSAVI       | OSAVI=(1+0.16)*(R800-R670)/(R800+R670+0.16)       | Rondeaux et al. (1996)           |
| Water index                                              | WI          | WI=R900-R970                                      | Penuelas. et al. (1993)          |
| Chlorophyll index                                        | Chl index   | Chl index= R750/R550                              | Gitelson and Merzlyak (1994)     |
| Modified simple ratio 705                                | mSR705      | mSR705=(R750-R445)/(R705-R445)                    | Sims and Gamon (2002)            |
| Transformed chlorophyll absorption in  reflectance index | TCARI       | TCARI=3*((R700-R670)-0.2*(R700-R550)*(R700/R670)) | Haboudane et al. (2002)          |
|                                                          | TCARI/OSAVI | TCARI/OSAVI                                       | Haboudane et al. (2002)          |
| Carotenoid index (Chappelle)                             | CarChap     | CARChap =R760/R500                                | Chappelle et al. (1992)          |
| Carotenoid index  (Blackburn)                            | Car1Black   | CAR1Black=R800/R470                               | Blackburn (1998)                 |
| Carotenoid index 2 (Blackburn)                            | Car2Black   | CAR2Black=(R800 - R470)/(R800+R470)               | Blackburn (1998)                 |
| Photochemical reflectance index (570)                    | PRI570      | PRI=(R531-R570)/(R531+R570)                       | Gamon et al. (1992)              |
| Structure intensive pigment index                        | SIPI        | SIPI=(R800-R450)/(R800+R650)                      | Penuelas. et al. (1995)          |
| Anthocyanin (Gamon)                                      | AntGamon    | AntGamon =R650/R550                               | Gamon and Surfus (1999)          |
| Anthocyanin (Gitelson)                                   | AntGitelson | AntGitelson =(1/R550-1/R700)*R780                 | Gitelson et al.(2003,2006)       |
| Chlorophyll content                                      | ChlDela     | ChlDela=(R540- R590)/(R540+R590)                  | Delaieux et al. (2014)           |
| Chlorophyll index                                        | CI          | CI=(R750-R705)/(R750+R705)                        | Gitelson and Merzlyak (1994)     |
| Photochemical reflectance index (586)                    | PRI586      | PRI=(R531-R586)/(R531+R586)                       | Panigada et al. (2014)           |
| Photochemical reflectance index (512)                    | PRI512      | PRI512=(R531− R512)/(R531+R512)                   | Hernández-Clemente et al. (2011) |
| Fluorescence ratio index1                                | FRI1        | FRI1= R690/R600                                   | Dobrowski et al. (2005)          |
| Fluorescence ratio indices 2                             | FRI2        | FRI2= R740/R800                                   | Dobrowski et al. (2005)          |
| Normalized Difference Vegetation Index1                  | NDVI1       | NDVI1=(R800-R670)/( R800+R670)                    | Rouse et al. (1973)              |
| Renormalized Difference Vegetation Index                 | RDVI        | RDVI = (R800 − R670)/(R800 + R670)0.5             | Rougean and Breon (1995)         |
| Red edge ratio index                                     | RERI        | RERI= R700/R670                                   | Part of TCARI index              |
[... cut ...]
| Red-Edge Inflection Point 1                                           | REIP1       | REP=700+40*{[(R670+R780)/2-R700] /(R740-R700)}    | Guyot and Baret, 1988            |       | #line 246
| Normalized Difference 790/720 Normalized difference red edge index    | NDRE        | NDRE=(R790-R720)/(R790+R720)                      | Barnes et al. (2000)             | ratio | #line 172
| Triangular Vegetation Index                                           | TVI         | TVI=0.5*(120*(R750-R550)-200*(R670-R550))         | Haboudaneet al. (2004)           |       | #line 510
```

dlebauer, 2017-02-08:
Looks good to me. I would say go ahead and upload it. We can change the equations and/or acronyms later if ensuing discussion requires it.

With respect to the formula, I would suggest either:

* omit the left hand side (as in your final suggestion `Formula: (R900-R680)/(R900+R680)`) 
* come up with acronyms without the `/` symbol. For the most part, the `/` doesn't really correspond to a division. So maybe ND_900_680 makes more sense (downside of this is that we loose the ability to cross reference the original tables ... but perhaps we can keep the Reflectance Index intact as the foreign key

Any thoughts @remotesensinglab?

@ashiklom, @serbinsh these are going into BETYdb ... does the mapping in https://github.com/terraref/reference-data/issues/72#issuecomment-277795515 make sense?

While I have you three on the line, does the use of ND900/680 to refer to a specific version of NDVI make sense? Or would it be more clear as NDVI900/680 or similar?

gsrohde, 2017-02-08:
@dlebauer Not having heard any comment, I went ahead and inserted these.  You can see a list conveniently by going to the variable list page and searching on "formula" (for now at least, since no other variables yet contain "formula").

I did use the "Formula: ..." designation.  In retrospect, it might have been better to put the formulas in square brackets to set them off from the other parts of the description, e.g. "Anthocyanin (Gamon) [Formula: R650/R550] Gamon and Surfus (1999)".  This could still be added, albeit with a bit more difficulty (or I could just delete these and re-add them rather easily).

gsrohde, 2017-02-08:
@dlebauer Also, I _did_ assume the 0.5 in the RDVI formula was an exponent.

Please close this if it's done.

closed 2017-02-08 by dlebauer

hmb1, 2017-02-09:
Hi David,
 just like to check in that the following  index has all the required fields with the correct syntax.
 
NDVI = ( R900 -R680) / (R900+R680 );
NDVI@long_name="Normalized Difference Vegetation Index";
NDVI@standard_name="normalized_difference_vegetation_Index";
NDVI@description="Normalized Difference Vegetation Index |  ( R900 -R680) / (R900+R680 ) | Rouse et al. (1973)"
NDVI@notes="Normalized Difference Vegetation Index | NDVI"
NDV@units="ratio"
NDVI@min=0.0;
NDVI@max="NULL";
NDV@label="Reflectance Index"
NDVI@type="trait"

Thanks
  ...Henry

remotesensinglab, 2017-02-09:
NDVI_900_680 makes more sense than ND900/ND680 for the short name. The ratio is confusing on the short name.

NDVI@max="NULL"; Shouldnt the NDVI max =1? If data calibration is correct, then this value should be within 0-1 for soil and vegetation. If there is water or snow in the field the value can be minus but this is not the case for us.

remotesensinglab, 2017-03-16:
The metada above does not spell out NDVI (it says Normalized Difference 900/680 Formula:). Do we have NDVI on the file name or or somewhere? As we all get used to the name NDVI rather than exact wavelengths, it is difficult to know if it is NDVI or not.

dlebauer, 2017-03-31:
@remotesensinglab could you please review variable names in #105, or feel free to directly edit them here https://terraref.ncsa.illinois.edu/bety/variables?search=reflectance+index. That is where the canonical definitions will be stored in BETYdb.