terraref/reference-data #74: Add 3D Scanner offset to sensor metadata

craig-willis, 2017-01-12:
Subtask related to https://github.com/terraref/reference-data/issues/70. 

Completion criteria:
* [ ] Offset information is stored in 3D scanner metadata
* [ ] Reviewed by relevant stakeholder

rachelshekar, 2017-01-12:
The offset needs to be stored with the date.

max-zilla, 2017-01-12:
3 values for offset - XYZ. No rotation or anything yet.

ZongyangLi, 2017-01-12:
Another value needed if that's possible: unit for voxel in point cloud coordinate system.

craig-willis, 2017-01-13:
@ZongyangLi Just to confirm, you're asking for a "unit" field with value "voxel" to be added to the 3d scanner metadata? How is this used?

dlebauer, 2017-01-13:
Also, @ZongyangLi I thought that 'voxel' refers to the cubes that make up a regular array, and that this sensor generates a point cloud in which points have no volume and are not on a regular grid. Am I missing something?

ZongyangLi, 2017-01-13:
@craig-willis @dlebauer 
Sorry for the confusion I have generated. When I say 'unit for voxel', I would like to know when point coordinate move from (0,0,0) to (1,1,1), how long is it in each direction in the field(real world). 

May be I am wrong, but to my understanding, a 3d scanner offset means an offset from point cloud coordinate system to gantry coordinate system.  When we have the 3d scanner offset and this 'unit transfer value', we are able to map every point to the field.

dlebauer, 2017-01-13:
@ZongyangLi Now I understand ... I think the documentation section on ['geospatial information'](https://terraref.gitbooks.io/terraref-documentation/content/user/geospatial-information.html) has the transformations that you are looking for (if not, we can ping Yan).

craig-willis, 2017-02-01:
Pull request:
https://github.com/terraref/sensor-metadata/pull/5

I've posted this to the clowder-dev instance for review:
https://terraref.ncsa.illinois.edu/clowder-dev/api/datasets/58920d9ae4b007e405e03301/metadata.jsonld

After PR is merged, I'll post to the primary clowder instance.

craig-willis, 2017-02-01:
Assigning to Solmaz for review.

solmazhajmohammadi, 2017-02-02:
@craig-willis Thanks for the update. The offset need to be applied to only one of the scanners. Current offset is for  "Top-heading-west" scanner. Can you please specify it in the metadata. Or add the offset for the "Top-heading-east" scanner as well. It should be  [0 , 0 ,0 ]

rachelshekar, 2017-02-23:
@craig-willis - can this be closed?

craig-willis, 2017-03-07:
Sorry for the delay, lost track of this ticket.  I've added a sensor identifier to the alignment_correction_offset and update the PR.

rachelshekar, 2017-03-16:
@ZongyangLi - do you have what you need now?  If so, please close this issue

ZongyangLi, 2017-03-16:
@rachelshekar I do not have what I need now.

rachelshekar, 2017-03-30:
dependent upon https://github.com/terraref/reference-data/issues/44

rachelshekar, 2017-04-04:
Changing this to April milestone since dependent on terraref/computing-pipeline#265 and https://github.com/terraref/reference-data/issues/44

rachelshekar, 2017-05-25:
Changing to June; #44 still open

rachelshekar, 2017-05-25:
https://github.com/terraref/sensor-metadata/pull/5
delayed because offset is temperature dependent.  @solmazhajmohammadi will update with additional data.

rachelshekar, 2017-07-12:
@solmazhajmohammadi please provide an update on the temperature measurements

solmazhajmohammadi, 2017-07-13:
I uploaded the report here:
[3D_Alignment(2).pdf](https://github.com/terraref/reference-data/files/1145923/3D_Alignment.2.pdf)

rachelshekar, 2017-07-20:
@solmazhajmohammadi - is this for air or box temperature?

rachelshekar, 2017-07-20:
- box was not moved when the calibration data was collected
- temperature is air - weather sensor on top of gantry
Does this offset apply for all sensors?
Is the gantry expanding? - eg Is the temperature response different at different ends of the field?

rachelshekar, 2017-07-20:
Original offset supplied for PR was not correct because temperature plays an important role in the offset.

rachelshekar, 2017-07-20:
@solmazhajmohammadi - please create new issue for temperature correction and close this issue

solmazhajmohammadi, 2017-07-20:
@rachelshekar Issue [#63](https://github.com/terraref/reference-data/issues/161)

closed 2018-05-31 by sbrad77