terraref/reference-data #75: Add SWIR and VNIR properties to sensor metadata

craig-willis, 2017-01-12:
Task related to https://github.com/terraref/reference-data/issues/67:

Add the following properties to the sensor metadata for SWIR/VNIR:
* Lens/aperture size
* Max Frame Rate

Completion criteria:
* [ ] Properties have been added 
* [ ] Review by relevant stakeholder

smarshall-bmr, 2017-01-12:
@craig-willis I just noticed that a typo in the VNIR metadata is keeping the entirety of the metadata from being written. I'm fixing it now.

craig-willis, 2017-01-12:
@smarshall-bmr Are you aware that we've been working to centralize the fixed sensor metadata to a github repository? https://github.com/terraref/sensor-metadata

I'm not clear who is involved in updating the fixed metadata and could use input from anyone concerned.

smarshall-bmr, 2017-01-12:
@craig-willis @dlebauer The VNIR metadata file had the aperture setting labeled as f/1.4 when it is actually set at f/2. This part of the metadata wasn't being written due to the aforementioned typo but we might have to look out for this field being incorrectly reported in other places.

I somehow missed the memo on the centralization effort. I do have access to the metadata files so I can modify what is being written as necessary.

craig-willis, 2017-01-19:
@smarshall-bmr Would it be possible for me to get a copy of the current JSON Generator program?

smarshall-bmr, 2017-01-19:
@craig-willis at some point I'll upload the entire metadata folder to github but I can give an example of a .txt that the metadata program pulls from. This one is from the VNIR. The metadata generator pulls from these text files as appropriate.
`{
  "lemnatec_measurement_metadata": {
    "user_given_metadata": {
      "experiment title": "Wheat field experiment 2",
      "experiment responsible": "to be named",
      "project": "TERRA-REF",
      "instrument": "gantry at Maricopa phenotyping facility",
      "location": "Maricopa phenotyping facility",
      "date of sowing": "2016-12-20",
      "date of emergence": "2016-12-30",
      "campaign": "seedling emergence and early vigor",
      "mission or scan": ""
    },
    "gantry_system_fixed_metadata": {
      "system manufacturer": "LemnaTec Corp., 4240 Duncan Ave, Saint Louis, MO 63110",
      "LemnaTec ProjNo": "7100019",
      "LemnaTecs field scanalzyer no": "2",
      "project responsible": "ben.niehaus@lemnatec.de",
      "date of installation": "april 2016",
      "system scan area n-s [m]": "211",
      "system scan area e-w [m]": "23.125",
      "system scan area height [m]": "8",
      "funded by": "ARPA-E",
      "trivials": "biggest agriculture robot in the world",
      "date of handover": "15-12-2016"
    },
    "gantry_system_variable_metadata": {
      "time": "01/13/2017 01:00:08",
      "position x [m]": "97.451",
      "position y [m]": "22.135",
      "position z [m]": "1.615",
      "speed x [m/s]": "0",
      "speed y [m/s]": "0",
      "speed z [m/s]": "0",
      "camera box light 1 is on": "False",
      "camera box light 2 is on": "False",
      "camera box light 3 is on": "False",
      "camera box light 4 is on": "False",
      "Script path on local disk": "C:\\LemnaTec\\StoredScripts\\3D_Scan_Field_033MperS.cs",
      "Script copy path on FTP server": "ftp://10.160.21.2//gantry_data/LemnaTec/ScriptBackup/3D_Scan_Field_033MperS_a3923dba-a7a9-4ef3-ba68-0313414e6da0.cs",
      "scanIsInPositiveDirection": "False",
      "scanDistance [m]": "21.8",
      "scanSpeed [m/s]": "0.33",
      "scanMode": "Triggered",
      "sensor setting file path": "c:\\LemnaTec\\StoredSensorSettings\\3dTop.xml"
    },
    "sensor_fixed_metadata": {
      "sensor manufacturer": "Fraunhofer-Entwicklungszentrum R?ntgentechnik EZRT, Ber?hrungslose Mess- und Pr?fsysteme, www.iis.fraunhofer.de",
      "sensor product name": "Custom made 3D Scanner",
      "sensor east serial number": "S215-1030",
      "sensor west serial number": "S215-1036",
      "sensor description": "two laser / camera combinations",
      "sensor purpose": "measure 3D surface topology",
      "laser emission wavelenght [nm]": "810",
      "laser emission energy [W]": "2",
      "location in gantry system": "camera box, facing ground",
      "scanner east location in camera box x [m]": "2.070",
      "scanner east location in camera box y [m]": "0.306",
      "scanner east location in camera box z [m]": "1.135",
      "scanner west location in camera box x [m]": "2.070",
      "scanner west location in camera box y [m]": "2.726",
      "scanner west location in camera box z [m]": "1.135",
      "field of view y [m]": "0.800",
      "Calibration available": "true",
      "output data format": ".ply open file format",
      "sensor id": "3d camera box"
    },
    "sensor_variable_metadata": {
      "current setting Exposure [microS]": "70",
      "current setting Calculate 3D files": "0",
      "current setting Laser detection threshold": "512",
      "current setting Scanlines per output file": "100000",
      "current setting Scan direction (automatically set at runtime)": "0",
      "current setting Scan distance (automatically set at runtime) [mm]": "21800",
      "current setting Scan speed (automatically set at runtime) [microMeter/s]": "100000"
    }
  }
}`

craig-willis, 2017-01-23:
Thanks, @smarshall-bmr. I want to confirm a few things:
- The "location in camera box" x, y, and z values are identical to the SWIR?
- "optics focal length [mm]" is 22.5 (not 23mm, which is listed in #67) and is different than the value of 17 reported in the LemnaTec documentation.
-

smarshall-bmr, 2017-01-23:
@craig-willis I thought that the offset being the same for both cameras had been corrected already, I'll take a look. Should I correct the values if they are wrong? 

I don't completely trust all the documentation that we have so I pulled the model number from the lens and looked up the specs for it. It is marketed as a 23mm lens but specs show it is actually 22.5.

craig-willis, 2017-01-23:
@smarshall-bmr Thanks for the clarification.  Let me know what the correct values are and where you get the information.  I'm not sure what the source of the location information is, if not the LemnaTec spec.

craig-willis, 2017-02-01:
@smarshall-bmr  I've created a pull-request with these changes. Please review and merge if acceptable:

https://github.com/terraref/sensor-metadata/pull/4/files

craig-willis, 2017-02-01:
Assigning to @smarshall-bmr  for review/feedback.

smarshall-bmr, 2017-02-01:
@craig-willis are we absolutely sure of this y axis field of view? 1.000m seems rather convenient, it might be a placeholder.

craig-willis, 2017-02-01:
@smarshall-bmr  Thanks for reviewing. The value came from the hyperspecVNIRCameraBox.txt you attached above. The documentation I have (https://terraref.ncsa.illinois.edu/clowder/datasets/5873a7bb4f0cad7d8131a0b7) doesn't seem to include this value.  Should I check with LemnaTec or maybe just remove it?

smarshall-bmr, 2017-02-01:
@craig-willis Hold that thought, I'm finding out now.

rachelshekar, 2017-02-23:
@smarshall-bmr - please update this issue

smarshall-bmr, 2017-02-23:
@craig-willis FOV for the SWIR is 44.5 degrees and the VNIR is 21 degrees.

craig-willis, 2017-03-08:
Updated FOV values and units in VNIR and SWIR in OR.

rachelshekar, 2017-04-13:
@craig-willis - can this be closed?

craig-willis, 2017-04-20:
@rachelshekar 
PR is still open.  I assume we leave this open until merged.
https://github.com/terraref/sensor-metadata/pull/4/files

closed 2017-06-01 by rachelshekar