terraref/reference-data #77: Add uncertainty estimate to traits inserted into BETYdb

dlebauer, 2017-01-12:
BETYdb traits have fields

* statname
* stat
* n (can be 'effective n')

Completion criteria

* [ ] define general approach to estimating uncertainty
* [ ] implement for height and canopy cover data extractors

rmgarnett, 2017-01-13:
I was thinking about this some more. It's a little unclear to me what the uncertainty of a statistic such as

α [nth height percentile] + β

is. Possible sources of uncertainty are:

* uncertainty in our estimates of α and β due to finite training data
* uncertainty in our estimate of the height percentile from the point cloud data

Both are effectively zero. Previously I had been interpreting "uncertainty" as "expected error in predicting a hand-measured height (in general, phenotype) using this statistic," but that is not a typical definition of uncertainty. Note also that a large component of that error could be inherent noise in the hand-measurement process.

@ZongyangLi, @pless: thoughts?

rachelshekar, 2017-03-02:
@ZongyangLi will talk to @pless and @rmgarnett to make sure that the uncertainty can be added to his data.  If so, he will upload it to BETYdb.
We want to capture reproducibility and variability within plot - RMSE?

dlebauer, 2017-03-02:
What about estimating RMSE as the bias between what is estimated by the algorithm and what we would expect if we took a measurement in the plot?

rmgarnett, 2017-03-02:
What about something like leaf angle distribution, where we can't easily make a hand measurement?  What about a future "purely digital" phenotype?

ZongyangLi, 2017-03-02:
My understanding of this issue: we are looking for the source of uncertainty in our estimates of α and β, and add this uncertainty to our further algorithm.

My idea of possible sources of uncertainty:

- we do not know the precise transfer data from Z direction in ply file to real world meters, that may affects α ?
- I happen to notice that the field ground is not a flat at all, distance between lowest and highest ground point is around 90+ point units even in a same plot area.
- as plant grows, leaves beside the target plot may insert into the target area that may affect the height percentile?

@dlebauer @rmgarnett Do I get the point of this issue?

rmgarnett, 2017-03-02:
I think the point of the issue is a bit nebulous.  How do we define "uncertainty" in general.  I am totally on board with David's suggestion of considering comparison with hand measurements, when they are available.  I'm just thinking that in the future, more and more phenotypes we're working with are not going to be hand measurable, and then I'm wondering what it means to assign an uncertainty.

dlebauer, 2017-03-03:
For data where hand measurements do not exist, maybe what we need is repeatability (and could this be estimated, with correction, from a time series, or by using part of the plot to estimate mean and the other part to estimate error?

For some background, what typically goes in this field is the standard error of the mean `SE = SD/sqrt(n)` or the sample standard deviation `SD_x = sqrt(var(x)/(n-1))` along with n, or any summary statistic reported in the literature (BETYdb is designed to integrate data from many sources and supports meta-analysis). (Notably SE --> 0 as n --> infinity whereas SD_x approaches the true value of SD). In the traditional context, plots are replicates and n = the number of plots. In this context, measuring within the plot, there are no true replicates. So SD_x and SE may not be appropriate. But we can store any statistic if we can name and define it. If we can estimate and record the uncertainty around the 'true' value we then it will allow us to propagate this uncertainty downstream.

Maybe this paper is a useful starting point 
[Comber_etal_2006.pdf](https://github.com/terraref/reference-data/files/815462/Comber_etal_2006.pdf)

dlebauer, 2017-03-03:
Also, remember that while we can store a narrowly defined concept of variance / uncertainty in BETYdb, we can also use the Clowder MongoDB that is much more flexible.

rmgarnett, 2017-03-03:
@dlebauer Yes, I think this scheme could work and would be useful especially for e.g. forecast values.  One effect of this particular choice, however, will be that some phenotypes will end up with zero (or near zero) uncertainty associated with them by this measure.  I personally think that's okay but we should all agree.

dlebauer, 2017-03-08:
from @ashiklom 

> I would probably calculate the mean spatial autocorrelation across the plot and then use that to compute an effective sample size, which you can then use to inflate the variance. It's a bit of a hack, but I use basically the same thing in 1 dimension for accounting for autocorrelation in leaf spectra.
One of my fellow PhD students had a way to quickly compute spatial autocorrelation across a 2D image using Fourier transforms, but I don't remember exactly how he did it.

dlebauer, 2017-05-18:
@rmgarnett can you suggest next steps? Should we have a conference call to discuss how to implement?

rachelshekar, 2017-06-01:
Meeting to discuss set for June 8
see #130

rachelshekar, 2017-07-12:
next steps: https://github.com/terraref/reference-data/issues/138

closed 2018-12-13 by dlebauer