terraref/reference-data #78: Data organization in clowder

dlebauer, 2017-01-20:
A few suggestions to make curating and navigating data in Clowder easier:

* Organize data by level (raw, level 1, 2) similar to how it is organized on ROGER. Currently, inputs and outputs are in the same dataset (e.g. https://terraref.ncsa.illinois.edu/clowder/datasets/587107f34f0cc129fb4fa083#) contains raw png, derived ply, and also .las. It is not clear which files are the raw versions and which are the data products. It is also not clear that the png, ply, and las files contain the exact same information but in different formats. 
* remove random character strings or _at least_ add information like date, time, sensor name to the beginning of the file. 
  * for example, [MODIS](https://terraref.ncsa.illinois.edu/clowder/datasets/587107f34f0cc129fb4fa083#) or the [protocol used by the Purdue team](https://docs.google.com/document/d/19svNB3FatbVXf6m3iFBdc_ivgxJouqPJpqBVe-ITqBg) have file naming conventions that make it easier for end users.  
  * the goal is to make it useful for quick reference by end users; sensorname and date may be sufficient.
* When data are downloaded or loaded into a docker container, remove the folder names that are random strings and have two levels of nesting.

These seem to be important useability issues.

### TODO

* [ ] determine potential solutions and approx. time to implement
* [ ] implement and / or create new issues.

max-zilla, 2017-01-20:
OK - let's discuss this briefly on Monday.

I believe the random string folder names in Docker are a collision-preventative measure in case someone has uploaded 2 files with the same name to a dataset, but I expect logic could be added so those only occur IF there is such a collision.

Re: the filenames, we don't rename any raw files from the sensor and I don't believe Clowder currently supports a 'display name' for a file - @robkooper correct me if I'm wrong. I won't say renaming files after a Globus transfer is out of the question, although that isn't appealing for a few reasons (difficult to compare to gantry source if needed, might need to back-convert old files, etc). However, implementing something like a display name in Clowder could be possible.

Part of what I hope to do with the pipeline revisions I'm in the middle of is add Level_1, 2... files to folders in the dataset with those names:
```
DATASET
- raw png 1
- raw png 2
- Level_1 folder
  - ply 1
  - ply 2
  - las 1
  - las 2
```
...etc to make that distinction clearer.

dlebauer, 2017-01-20:
We don't need to rename the raw data, but it seems reasonable that we could do the renaming when generating the Level 1 and 2 data.

Also, with respect to meta-data we should make sure to distinguish the LT provided metadata from metadata that is converted to a 'standard' (@craig-willis please advise). Currently I only see LT metadata but not any information derived from the extractors.

dlebauer, 2017-01-20:
With respect to Docker - I thought we had discussed mounting the file system directly as read only rather than copying. Did we give up on this idea?

craig-willis, 2017-01-20:
@dlebauer 
Mounting the data read-only is still planned, it just hasn't happened yet.  I still think this is the best approach.

rachelshekar, 2017-01-26:
Create space for derived files

max-zilla, 2017-01-26:
@robkooper @craig-willis  fundamentally it's difficult to differentiate raw and Level1/2 data. would also be good to hide raw data and only show people derived.

- could store raw/Level1/Level2 in folders within dataset
- could create 2 datasets per sensor x timestamp - one in a raw data space, one in a derived data space (Rob's idea)
- could use tags or other property flags to make clear

rachelshekar, 2017-03-02:
@max-zilla - please make issues based on today's discussion

max-zilla, 2017-03-28:
Final decision that was made:
- a new Space was created that I've tentatively titled "MAC - 2017 Data Release": https://terraref.ncsa.illinois.edu/clowder/spaces/58da6b924f0c430e2baa823f
- the plot-clipped tiles and subsequent Level 1+ data products will go here, separate from all raw data currently in the MAC space.
- eventually we can purge the raw MAC space. in shorter term once we start populating 2017 Data Release we can remove access to raw data for most users.
- 2017 Data Release will be organized by Plot > Year/Month > Day with all plot-level imagery and data for each sensor in the same dataset.

Epic for achieving this new organization:
- https://github.com/terraref/computing-pipeline/issues/270

closed 2017-03-28 by max-zilla