terraref/reference-data #79: Exchange of supplemental halogen lamps

TinoDornbusch, 2017-01-24:
all,

halogen lamps need to be replaced. The original lamps are not to purchase anymore.

Questions:
1.) Is halogen supplemental light crucial and used for the SWIR/VNIR analysis?
2.) Which type of halogen lamps (220V) do you prefer if you could choose?

There are different light colors available. From next week on we can test different bulbs here in Aachen and make the spectrum available.

Tino

smarshall-bmr, 2017-01-24:
Its my understanding that the halogen lamps were intended for use with the hyperspectral cameras, but they don't have enough output to actually allow for nighttime scanning. Maybe someone else can weigh in, but my opinion is that we should just get a full set of new bulbs for now and worry about other problems later given that we aren't exactly sure what to do with the halogens in their current iteration. I don't necessarily think we should just abandon maintenance on the halogens though, we may find a good use for them yet!

rachelshekar, 2017-02-23:
@czender and @remotesensinglab - do you have any feedback for Tino?

czender, 2017-02-23:
I have no experience/advice with artificial lighting for hyperspectral measurements. the comment of @smarshall-bmr above seems reasonable.

remotesensinglab, 2017-02-23:
We used halogen lamps in lab environment. But do not have much to offer in this case as I have not looked either the lamps or the data collected using the lamps. I agree with @smarshall-bmr suggestion.

dlebauer, 2017-02-23:
I'll mark this resolved 

Answers to Tino's questions are 

1. No, not essential 
2. No preference on light source. But it will be necessary to have the spectra of light incident on the plant