terraref/reference-data #80: Storing protocols including software code in BETYdb

dlebauer, 2017-01-25:
(from anonymous user)

Yesterday in our local TERRA REF meeting the question arose as to how to document the calculations used to obtain plot-level yield data or any other data where intermediate calculations are made.

As you have seen, there are multiple ways to arrive at a value of yield per unit area, depending on what assumptions are made about effects of plot size and plant stand. In this case, Mike and I would prefer to provide the raw data and the SAS code. Other scientists might want to upload Excel macros or R scripts.
Should such information just be uploaded as part of the protocols or is there another way to link information on data processing with uploaded data?

Keep in mind that from our perspective, the phenotype (e.g., aboveground dry matter/m2) might remain the same over multiple seasons but that the calculations could vary. For example, if we continue to improve our planting and seedling establishment, we might drop corrections for stand and variable plot length.

dlebauer, 2017-01-25:
Here is the way we can do this:

1. When you upload data you can have a trait name like 'height' as well as a method like 'hand measurement'
2. the method has a field for description. Simple methods and code snippets can be placed in the description field. Each method also has a 'citation'. The citation can reference a person, a citation, or any other public resource (e.g. a github repository or the [terraref documentation](https://terraref.gitbooks.io/terraref-documentation/content/user/protocols-manual.html). Code, excel spreadsheets, macros, etc can all be stored on GitHub. We can discuss how best to organize these, for starters it might be easiest to create a protocols repository. For code that is in the pipeline we have approx. one repository per set of inputs / outputs.

As a reminder, we previously decided that we would use methods from Pérez-Harguindeguy 2013 where they fit the needs of our project. 

One thing I am not clear on is how you are correcting for yield. If you are correcting for establishment then perhaps we should have a 'stand biomass' with plot area as a covariate in addition to the yield adjusted for bare ground. Otherwise, two plots with very different establishment could look very different to sensors but have the same 'yield' reported. 

PS our server for BETYdb is undergoing maintenance right now. When it is back up I can point you to how we are already storing different methods for measuring height.

JeffWhiteAZ, 2017-01-25:
Thanks for the clarification. I would support developing a protocol repository that allows for more complex explanations than in a method description field. Looking at the height protocol examples would be good.

dlebauer, 2017-01-25:
@JeffWhiteAZ if you look at the protocols template it has room for more detail. And after compiling the documentation will render latex math. However, most of the field measurement protocols I was given last year fit into a spreadsheet cell.

Before I set up a repository, see if something like [protocols.scienceexchange.com](https://protocols.scienceexchange.com/) or [protocols.io](https://www.protocols.io/about) would be easier to use. If there is motivation for someone to write up proper protocols, One (or both) of these interfaces might be preferable. The advantage of these is that they make the protocols more easily discoverable and they also make it easier for other groups to share and adapt the protocols without changing the referenced version. I know that there is a lot of interest in a set of standard protocols, so these may provide an appealing and neutral interface.

dlebauer, 2017-01-28:
@nshakoor @terraref/developers 

from today's notes:

Publishing protocols

These will be developed with peer review. We have the standards committee, five teams within TERRA, and many interested industry partners. Discussions on GitHub can count as peer reviews and we can archive them as well. This is the new path.

* https://www.protocols.io/ has option to fork, version, and mint doi's w/ citations like 
> RACHELE CESARONI (2017). Euplotes crassus transformation using FuGene HD Transfection Reagent as vehicle. protocols.io
dx.doi.org/10.17504/protocols.io.g3ebyje 
* http://www.bio-protocol.org/e1275 (formal peer review. too slow.) 
 *Gilbert, K. B., Fahlgren, N., Kasschau, K. D., Chapman, E. J., Carrington, J. C. and Carbonell, A. (2014). Preparation of Multiplexed Small RNA Libraries From Plants. Bio-protocol 4(21): e1275. DOI: 10.21769/BioProtoc.1275; Full Text
 * Carbonell, A., Takeda, A., Fahlgren, N.,Johnson, S. C., Cuperus, J. T. and Carrington, J. C. (2014). Newgeneration of artificial MicroRNA and synthetic trans-acting small interferingRNA vectors for efficient gene silencing in Arabidopsis. Plant Physiol 165(1): 15-29.
* https://protocols.scienceexchange.com/
* http://www.nature.com/protocolexchange/ 

protocols.io has a simple user interface.

rachelshekar, 2017-03-09:
- [ ] share example protocol
- [ ] choose protocol website
- [ ] promote across team

dlebauer, 2017-05-18:
@nshakoor have you made any progress?

rachelshekar, 2017-09-07:
protocols.io being tested by @dlebauer - better use for field and benchwork.

rachelshekar, 2017-09-07:
@nshakoor - 
document the protocol for protocol documentation
please discuss with david about informing team of this requirement

dlebauer, 2017-10-26:
Will use protocols.io, documentation, readme files, and methods table in betydb

closed 2017-10-26 by dlebauer