terraref/reference-data #84: Put LemnaTec XML files under source control

craig-willis, 2017-02-01:
As noted in https://github.com/terraref/reference-data/issues/64, https://github.com/terraref/reference-data/issues/75 and https://github.com/terraref/computing-pipeline/issues/230, we have a problem with untracked changes to the XML files used to generated the user_given_metadata and sensor_fixed_metadata.

Completion criteria:
* [ ] All relevant LemnaTec XML files used to generate the fixed metadata fields are in a Github repository
* [ ] Process of updating the XML files is documented
* [ ] Review by relevant stakeholders

craig-willis, 2017-02-01:
@smarshall-bmr 
Please send me the current copies of the XML file(s) that store the user_given_metadata and sensor_fixed_metadata values.

smarshall-bmr, 2017-02-01:
@craig-willis metadata files are now available at https://drive.google.com/open?id=0BzWFmpKimq_6OGRVdWw5UTlNRTA

craig-willis, 2017-02-01:
@smarshall-bmr Thank you

@dlebauer 
I've created a new github repository with these files:
https://github.com/terraref/lemnatec-fixed-metadata/

This repository is configured to require review before changes can be merged to master.  My suggestion is that, going forward, changes to these files be made via this Git repository and are copied to the control PC via git from the master branch after changes are merged.  

Does this make sense?

dlebauer, 2017-02-01:
Makes sense to me. I just updated the README to clarify the scope of the repository https://github.com/terraref/lemnatec-fixed-metadata/pull/1

closed 2017-03-07 by craig-willis