terraref/reference-data #86: Approach to computing plot level mean phenotypes from high resolution data

dlebauer, 2017-02-02:
### Description

In https://github.com/terraref/computing-pipeline/issues/183#issuecomment-277045908, @czender raises the question of how to compute plot level means.

There will be a lot of sensors for which we generate plot level averages, so we should come to a general consensus on how we compute averages. At this stage the most important step is to provide both 1) a defensible plot level summary with assumptions described in the algorithm and protocol documentation and 2) the input datasets so that end users can explore alternative assumptions and algorithms.

Two options are 

* averages of per-pixel ratios: NDVI = mean(( R900 -R680) / (R900+R680 ))
    * useful for upscaling to simulate lower resolution sensors
    * Historically, indices such as NDVI are integrated on scales of 10-100 m (from satellites). Our sensor suite is fundamentally different because the mm resolution is << scale of the plant.
* ratio of spatial averages: NDVI = (mean(R900) -mean(R680) )/ (mean(R900)+mean(R680 ))
    * better approximation of the 'actual' phenotype (?)

Notably, 

Are there any cases where it would make sense to compute the indices on a per-pixel basis before averaging? 

I am familiar with this problem in the context of using gridded meteorological and soil datasets to run a crop or land surface model. If I had a high resolution meteorological dataset (like NARR on a ~30km grid) and I want to know mean yield for the state of Illinois, I would run the model on each pixel and then take the mean of the yields. Alternatively, I could take the mean light, temperature, precipitation for the whole state at each time point and then run the model once. I can see how these are different applications (esp. if the objective is to simulate lower resolution data), but if the objective is to get the best representation of a plant's traits, it seems that the latter would be more appropriate. 

### Masking

We don't yet have a clear way to mask data from the hyperspectral sensors. In the point clouds, I know that the ground is excluded based on height, and I understand that the RGB extractor creates a mask to compute canopy cover. We should be able to exclude pixels with soil based on NDVI or other ratios. 

Here again, if we want to simulate a lower resolution sensor we would keep the soil; if we want to estimate a property of the plant we would use the higher resolution data.

### Completion Criteria   

* [ ] determine a general approach to computing plot level summaries
* [ ] implement in the hyperspectral extractor
* [ ] update documentation

remotesensinglab, 2017-03-02:
**Suggestions:**
Calculate the indices using pixel reflectance values then average the indices over each plot/genotyp. Many indices, e.g., NDVI, are able to reduce the effects of soil and other noise factors. This is due to the fact that soil, shadow and other noise elements have equal or close to equal reflectance that can be balanced out by a ratio index.  See the figure below, which is generated using file 3bc4bd12-ff75-4cac-bb25-ac1f444c0423_.nc. Note: this image was one that collected with over exposure. 
[image]

rachelshekar, 2017-03-02:
Wasit will do more testing and give an update on what indices to use.
Charlie will make the default NDVI data the average of indices (not indices based on the averages)

czender, 2017-03-02:
@hmb1 this means that we should rename the variables so that the default names, e.g., NDVI, are for the variables computed with the pixel ratios done before averaging over the plots (the old _rba variables). The original primary method becomes the new "secondary" method, and those variables should be suffixed with _abr (average before ratio). @remotesensinglab Wasit will create an algorithm to mask shadow and soil, but that will come later. The renaming can be done now. Thanks!

remotesensinglab, 2017-03-02:
I think this issue can be closed. Removal of shadow and soil effects will be discussed on the issue below https://github.com/terraref/reference-data/issues/96

hmb1, 2017-03-02:
will Do !!

hmb1, 2017-03-06:
Have renamed vars as requested

remotesensinglab, 2017-03-12:
I am wondering if the <average before ratio> can be even dropped from production? Mixed pixel is a classic issue in land remote sensing. Often we try to eliminate information other than vegetation (especially for vegetation remote sensing). I tent to think that averaging before ratio 1) may confuse users with two different products, and 2) has very limited value for plant phenotyping or soil characterizing. I do not know the product size, but the size of the data can be reduced as well?

czender, 2017-03-13:
I agree it will be confusing to an end-user to have two sets of indices. We are in the development and feedback stage, and your suggestion is most welcome. I would like to maintain the code that computes indices both ways, but keep the "average before ratio" method from running by default. At some point I think we might learn something (and write a paper) that contrasts the ABR and RBA methods.

dlebauer, 2017-05-18:
Can this be documented and then closed?

craig-willis, 2018-06-13:
Stale issue. No comments in last 2 months. If necessary, create new issues based on this discussion.

closed 2018-06-13 by craig-willis